OpenCPN/OpenCPN #4030: Connection dialog

nocheinpeter, 2024-07-02:
**Describe the bug**
Cannot change the talker id of an existing connection.
Cannot add a new connection

**To Reproduce**
Steps to reproduce the behavior:

Open Connection dialog, select an existing TCP connection and change the talker id from EC to GP, comfirm and close the dialog.
Reopen again, but the talker id remains EC.

Adding a new connection is also not possible.

**Expected behavior**
Change of value is possible. Adding connection is possible.

**Video**
https://photos.app.goo.gl/7CD6AqBzwYx7X7QWA

**Desktop (please complete the following information if applicable):**
 - OS: Android 11, Android 14
 - Version 11, 14 

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Galaxy Tab A 2019, Galaxy Tab A9
 - Andorid version: 11, 14

leamas, 2025-07-20:
Ok, let's wait until until 5.12 eventually is out also for.android.

nocheinpeter, 2025-08-05:
Works now in Android with openCPN 5.12.2