OpenCPN/OpenCPN #4031: OpenCPN spoofs the User-Agent

MennoTammens, 2024-07-02:
**Describe the bug**
OpenCPN identifies itself as `Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:35.0) Gecko/20100101 Firefox/35.0`
OpenCPN uses a lot of free services (e.g. charts, GRIB files). The best practice for a HTTP client embedded in an application is to make itself known with the User-Agent header.

Spoofing the User-Agent is, in my opinion, disrespectful to the providers from which you use the free services.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a custom catalog to the chart downloader
2. Use a URL on a web server from which you can read logs
3. "Update" the catalog
4. Watch your logs

**Expected behavior**
`User-Agent: OpenCPN/5.9.0` or similar

**Desktop (please complete the following information if applicable):**
 - OS: Linux
 - Version 5.9.0 git

**Additional context**
See PR #4029

leamas, 2024-07-02:
OK, let's continue here in a better context.

I don't really see this as a bug, rather an enhancement. But let's avoid splitting hairs over this.

Are you aware that we currently are in a informal beta freeze state? That is, we are close to releasing 5.10 and reluctant to make changes which are not bugfixes if they carry risks.

leamas, 2024-07-08:
There will be no time to test this change before 5.10. Assigning to 5.12

leamas, 2025-05-03:
> User-Agent: OpenCPN/5.9.0 or similar

No. We should use the same main compatibility as almost everyone else: "Mozilla/5.0". _Everyone else_ includes Safari and  Edge, we should be able to use it as well.

New string in commit above is  something like ` Mozilla/5.0 (OpenCPN 5.11.0; debian-x86_64; 12) curl/8.9.1`, versions and ABI are as defined in various headers.

MennoTammens, 2025-05-04:
Non-browser tools use only their own name, e.g.:
```
$ curl http://httpbin.org/user-agent
{
  "user-agent": "curl/7.88.1"
}

$ wget -q -O - http://httpbin.org/user-agent
{
  "user-agent": "Wget/1.21.3"
}

$ http -b http://httpbin.org/user-agent
{
    "user-agent": "HTTPie/3.2.1"
}
```

For a Mozilla/5.0 type User-Agent, I think the `OpenCPN/<version>` part should be after the platform:
`Mozilla/5.0 (debian-x86_64; 12) OpenCPN/5.11.0 curl/8.9.1`

Compare with:
```
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36
Mozilla/5.0 (X11; Linux i686; rv:124.0) Gecko/20100101 Firefox/124.0
```

MennoTammens, 2025-05-26:
> There will be no time to test this change before 5.10. Assigning to 5.12

First beta for 5.11 is out. Can you merge this before the next beta?

commit b4d9756c (2025-07-18): downloader: Use ocpn-specific User-Agent - #4031

In order to make it possible for remoet peers to identify a call from
OpenCPN a new User-Agent field is crafted like

    Mozilla/5.0 (OpenCPN 5.11.0; debian-x86_64; 12) curl/8.9.1

This should hopefully stay compatible while still making it possible
to verify the client.

Closes: #4031
closed 2025-07-18 by leamas