OpenCPN/OpenCPN #4034: 5.9.2 Beta not recognizing charts as 5.8.4

svMigration, 2024-07-03:
**Describe the bug**
5.8.4 and 5.9.2 installed parallel. Open 5.8.4 and all charts are displayed in chart bar.
Turn chart quilting off.
Close 5.8.4.
Open 5.9.2.
Only some charts are displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Install various charts (MBtiles, CM93, BSB)
2. View Chart bar in 5.8.4 and 5.9.2. They are not the same

**Expected behavior** 
They should be the same.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS:Windows 11
 - Version 22

**Additional context**
This also sometimes crashes OpenCPN when switching through chart groups or turning quilting on and off.
In 5.9.2 turn quilting on and off with various chart groups or ALL charts group selected and watch how the chart bar or the chart does not update properly. Seems to be related to MBTiles.

Notice how many charts are available in 5.8.4
[image]

Close 5.8.4
Open 5.9.2 and it should be the same.

[image]

bcn2, 2024-07-03:
To avoid overcrowded chart bars, from a certain {n?} number of visible charts on, the charts are grouped by types in 5.9. Hover over the entries in the bar.
You can change that behavior at Options|Display|advanced
[image]

[image]

svMigration, 2024-07-03:
BCN2: Thanks for the explanation. However, this doesn't work with MBTiles.
[image]

1. Since the default for "Show extended chart bar info" is off, the chart bar just says "x charts combined". This will be VERY confusing for newbies (and I guess not-newbies since I was confused!).
2. MBTiles get grouped together even though they are completely different sources (ArcGIS, Google Satellite, etc.). One is unable to choose the chart one desires.

There needs to be a way to turn 'Chart Bar Grouping' off.

Hakansv, 2024-07-03:
I wouldn't call this issue a bug. A bug is when the program code is wrong, e.g. a typo, so the program don't work as intended.
It's more of a request or a discussion?
Personally I see this new function as feature avoiding a crowded chart bar disturbing most users.
Your 2. ----"One is unable to choose the chart one desires.", i.e. for a specific chart scale size, may be a enhancement if zooming  does not make it clear enough?

leamas, 2024-07-03:
No, this is indeed not a bug. In particular, what's stated under *Expected behaviour" is not true -- functionality has been changed, and the behaviour is thus expected to change. 

This is either a proposed enhancement or a Discussion. Relabeling as an enhancement for now

svMigration, 2024-07-03:
However, the behavior of OCPN has changed in this version so that it is no longer possible to select some of the charts that are loaded.  That is in essence a bug as the enhancement 'chart bar grouping' removes essential functionality.  Zooming does not allow one to use the various MBTiles that have been grouped.

**You are correct that it's not a coding bug. But it is a bug in the fact that a change in the code has created unexpectedly removed functionality that is crucial to the use of the program.**

Chart Bar Grouping combines MBTiles together when they should NEVER be grouped together because though they are the same 'type' of chart (MBTile), they are displaying information from a variety of chart sources.

bdbcat, 2024-07-03:
I think I understand the issue.  I'll take a look, may come back with further questions.
This is why we do formal Beta testing.
Thanks for the input.

svMigration, 2024-07-03:
Thanks, Dave.

Appreciate it.
Cheers,

Bruce

P.S. I will be underway and away from internet for about a month starting Saturday.

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

bdbcat, 2024-07-04:
Bruce...
Turned out to be fairly simple (read: safe) change.
If you would like to test and confirm before you go off grid, please install this one:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.9.0-0+1714.994e55d_setup.exe

soggygeek, 2024-07-04:
Actually, I think this grouping mbtiles is a positive change.  However, it does not seem to be working properly on my W10 laptop or W11 pro mini computer.  With 5.9.1-0+1622.b2cbf5a, and LOTS of mbtiles, I see 4 magenta piano keys.  Each one when hovered states it is a composite of X charts.  And I see charts (I think) properly highlighted when I hover over each piano key.  However, clicking that "composite" key does not turn on all of the charts represented by that key.  I wish it would.  
I have just now download and installed the new build posted above by bdbcat and test more thoroughly on W10 and W11 (both latest versions). 
-->Note that version numbering doesn't seem to be consistent--the original beta install was labeled opencpn_5.9.1-0+1622.b2cbf5a and the one you have posted is opencpn_5.9.0-0+1714.994e55d, which looks like an older version, not newer.
After a quick look at bdbcat's just-posted "fix".  I like the original change in 5.9.x, (partially because I do my mbtiles differently than Bruce on Jon Hacking do). But can live with what you have changed to.  Would be nice to see an option to include or exclude mbtiles in the grouping scheme, if it is simple (I know, it's never simple).
Bruce, if they fixed it so that clicking a "composite" mbtile turned on and off all the mbtiles in that composite properly would this suit you? (You can always zoom in to turn on and off individual mbtiles).
bdbcat & Hakan, thanks for all you do.

bdbcat, 2024-07-04:
soggy...
Version number difference:   This build is a quick fix to check the proposed solution before committing to the next Beta release.  Mainly done to get Bruce's input before he goes off grid for a while.

Hakansv, 2024-07-04:
A "silly" question:

> "....completely different sources (ArcGIS, Google Satellite, etc.). One is unable to choose the chart one desires."

If you've charts at the same scale covering the same area but different sources. Why not put each source in a chart group easily shifted between by the num keys?
>> Is it because you don't see there are more without shifting group?  
(Sorry for my ignorance working with a lot MBTiles)

svMigration, 2024-07-04:
Dave, I just installed it and looks good but I'll spend a bit more time to check it more thoroughly in the AM. Am I correct that MBTiles are now excluded from grouping? Thank you. While testing, still occasional crashes when panning with MBTiles. Will upload crash reports in AM.

Hakan, not a silly question. Absolutely right. Chart groups are the BEST way to use MBTiles and that's what I tell everyone to do. The number key/chart group features is one of the most awesome features of OCPN. There's a video all about it on TheChartLocker. That said, there are a lot of people who either don't understand Chart Groups, don't watch the video, don't want to learn, or....? So they just randomly click on the chart bar.  Also, there are times when having quilting off and switching between the buttons on the chart bar is easier... usually in an area where the only decent charts are MBTiles... especially true on a tablet when you have to long press for the chart group menu.

svMigration, 2024-07-04:
>Bruce, if they fixed it so that clicking a "composite" mbtile turned on and off all the mbtiles in that composite properly would this suit you? (You can always zoom in to turn on and off individual mbtiles).

Sherry, that doesn't work. What I was seeing on my system is all 4 MBTiles (ArcGIS, BingSat, GoogleSat, Navionics) grouped together for Fiji (for example). They all have the same zoom level (8-18) so zooming didn't change anything. There was no way to select just one MBTile from the chart bar (ArcGIS, for example). Of course I could hit '4' and use that Chart Group. But if one didn't have chart groups set up, one couldn't select a specific MBTiles. Cheers, Bruce

svMigration, 2024-07-06:
Dave, 
Should all charts except MBTiles be grouped regardless of quilting on or off?
Thanks,
B

svMigration, 2024-07-08:
Dave, we decided not to leave yet so should have internet for another week. Still trying to figure out how to reproduce crashes and how this chartbar grouping can be managed to not create confusion.
Cheers
Bruce

bdbcat, 2024-07-08:
Bruce...
I have been testing mbtiles on a W11 laptop, with hi density display.  I have not "succeeded" in crashing once.
I have loaded a bunch of your tiles, thanks very much.
But no luck yet.
Just so I'll know, can you tell me which tiles and general location you are most often crashing at?

svMigration, 2024-07-09:
Hi Dave,
Here are screen shots of what I have loaded and the chart groups I'm using.
Did any of the crash reports I uploaded help?

[image]

[image]

svMigration, 2024-07-09:
Forgot to mention that the crashes have occurred all over the map (ha ha). Over by Mexico, around Fiji, and around my current position in west French Polynesia.

svMigration, 2024-07-09:
Just created a crash. Will upload details and report to [https://github.com/OpenCPN/OpenCPN/issues/4043](url)