OpenCPN/OpenCPN #4035: API 18: ocpn_plugin.h "Plugin_Active_Leg_Info" - No data without a GNSS source

Rasbats, 2024-07-03:
**Describe the bug**
The plugin message_id "OCPN_RTE_ACTIVATED" works without a GPS source but "Plugin_Active_Leg_Info" does not carry any data. 

**To Reproduce**
This code:
```
void finSAR_edit_pi::SetPluginMessage(wxString &message_id,
                                      wxString &message_body) {
  
    if (message_id == "OCPN_RTE_ACTIVATED") {
      m_route_active = true;      
      SetActiveLegInfo(myleg_info);
    }
  
}

void finSAR_edit_pi::SetActiveLegInfo(Plugin_Active_Leg_Info &leg_info) {
  if (m_route_active) {
    wp_Dtw = wxString::Format("%f", myleg_info.Dtw);    
    wxMessageBox(wp_Dtw);
  }
}
```
No data given for the active leg even though the route shows as activated on the chart.

**Expected behavior**
Provides the ActiveLegInfo data.

**Desktop (please complete the following information if applicable):**
 - OS: Win 10

bdbcat, 2024-07-05:
We could sure fix this, but...
If no GNSS is active, then some of the fields in ActiveLegInfo are meaningless.  e.g. XTE.
I guess there is some value in knowing the name of the active point, and the range/bearing when the signal was lost.
It would help to understand the use-case of activating a route when no GNSS is present.

Rasbats, 2024-07-05:
@bdbcat ...

For the Finnish SAR project I needed to identify the leg of a route. ActiveLegInfo allowed me to identify the waypoint that defined a leg and also the previous waypoint.

A real mess but poc for an editor plugin is here:
[https://github.com/Rasbats/SARroute_pi](https://github.com/Rasbats/SARroute_pi)

The repo will be replaced shortly by this repo:
[https://github.com/opencpn-finSAR/finSAR_edit_pi](https://github.com/opencpn-finSAR/finSAR_edit_pi)

The editor repo is solely for editing purposes and not for navigation. That function is reserved for the ops plugin:
[https://github.com/opencpn-finSAR/finSAR_ops_pi](https://github.com/opencpn-finSAR/finSAR_ops_pi)

I was waiting for your workload to ease off before sending a github invitation for access to the _opencpn-finSAR_ organisation.

We are working with RTZ route files. Attached to each route will be an extensions xml file that holds parallel index, range and direction data for each leg of the route, which Finnish SAR needs for the navigation of fast rescue boats. 

[image]

Activating a route point was an intuitive way of identifying the leg of a route. With cross hairs to identify the navigation object for the index or range, the data can be produced, 

I can see from the OpenCPN base code that the leg info table is tightly bound to ActiveLegInfo. The display of the table in OpenCPN is a distraction but this might be acceptable as part of an editor.

Hubert (o-charts) is coordinating and liaising with the Finns. We don't know if this project can succeed but we are putting work into it, as time allows.

bdbcat, 2024-07-09:
Rasbats...
In your use-case, which fields of class Plugin_Active_Leg_Info are you likely to use in the Finn plugin?
From the example, I see Dtw.
Any more?
Reason I ask:  If core updates and provides this class/struct with no GNSS active, some fields will be (dangerously?) garbage.

Rasbats, 2024-07-10:
Dave ...
Thanks for looking at this when the beta cycle is in progress. This morning I have revisited the issue. I think I can make a context menu to replace 'Activate', including changing the waypoint icon to show the selected leg. There will be no distracting 'This Leg' information.

closed 2024-07-10 by Rasbats