OpenCPN/OpenCPN #4036: Windows 7 missing Shcore.lib creates a start up error

Hakansv, 2024-07-03:
**Describe the bug**
When starting O5.9.x on a Win7 an error about missing api-ms-win-shcore-scaling-l1-1-1.dll occurs. See shot.
It's possible to find the dll in question on a general dll-site and copy it to e.g. C:\Program Files\OpenCPN.
Others have seen the same. See a link herein.

**To Reproduce**
1. Run installer for O5.9.x on a Win7
2. Start OpenCPN
3. See the error

**Expected behavior**
Since we support Win7 it should be possible to run OpenCPN without copying a more or less questionable dll from external source.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Win7
 - Version O v5.9.x

**Additional context**
Other have had the same problem.
Here a link to a [patch example](https://github.com/flutter/engine/pull/30699) for a similar issue.

bdbcat, 2024-07-10:
As painful as this might be, we need to run a git bisect operation to determine when this problem first appeared after O584.

nohal, 2024-07-10:
Very likely from the multimonitor and display scaling support. Is there any problem simply including that library? I have no win machine with me so can't do much about it until back from Europe.

bdbcat, 2024-07-10:
I suppose the dll is safe enough.  That seems the simplest route.
As you can see, if we are serious about Win7, we will need some work on the ownship and AIS target rendering.  This will likely need to be done by detecting W7 at runtime, and branching to some workarounds for lack of true DPI-awareness on W7.

bdbcat, 2024-07-10:
I guess adding and committing the dll to OpenCPN/OCPNWindowsCoreBuildSupport is all that is required, magic takes over from there.
Agree?

nohal, 2024-07-10:
Something like that, don't know if the zip is created manually or from a tag release

bdbcat, 2024-07-13:
New tagged Release ready, "v0.5".  CI build script updated.
Ready to close.

closed 2024-07-13 by Hakansv