OpenCPN/OpenCPN #4040: gshhsCrossesLand not using high quality GSHHG files even though the full resolution GSHHG files are installed

sebastien-rosset, 2024-07-03:
**Is your feature request related to a problem? Please describe.**
OpenCPN loads and render full resolution GSHHG files, however the `gshhsCrossesLand` function only uses the crude quality GSHHG file. As a side effect, this means the WeatherRouting plugin uses crude quality GSHHG file for its computation, while OpenCPN displays the full resolution polygons.

**Describe the solution you'd like**
If OpenCPN can load and display high resolution GSHHG files, then the `gshhsCrossesLand()` function should use the full resolution files.

**Describe alternatives you've considered**
A workaround is to install the full resolution GSHHG files in the default install location.

**Additional context**
The default location contains crude resolution polygon files:
```
ls -al /usr/local/share/opencpn/gshhs/
total 5260
drwxr-xr-x  2 root root    4096 Jul  1 17:26 .
drwxr-xr-x 10 root root    4096 Jul  1 17:26 ..
-rw-r--r--  1 root root 3728000 Jul  1 16:41 poly-c-1.dat
-rw-r--r--  1 root root   85144 Jul  1 16:41 wdb_borders_c.b
-rw-r--r--  1 root root 1557000 Jul  1 16:41 wdb_rivers_c.b
```

The high quality GSHHG files are loaded in OpenCPN:
[image]

I'm not sure if this behavior is by design or not.

To troubleshoot, I've added some logs in gshhs.cpp to understand how `gshhsCrossesLandInit()` finds GSHHG files.
```
08:46:25.114 MESSAGE gshhs.cpp:1257 GSHHG: Checking if file /usr/local/share/opencpn/gshhs/poly-c-1.dat exists.
08:46:25.114 MESSAGE gshhs.cpp:1257 GSHHG: Checking if file /usr/local/share/opencpn/gshhs/poly-l-1.dat exists.
08:46:25.114 MESSAGE gshhs.cpp:1257 GSHHG: Checking if file /usr/local/share/opencpn/gshhs/poly-i-1.dat exists.
08:46:25.114 MESSAGE gshhs.cpp:1257 GSHHG: Checking if file /usr/local/share/opencpn/gshhs/poly-h-1.dat exists.
08:46:25.114 MESSAGE gshhs.cpp:1257 GSHHG: Checking if file /usr/local/share/opencpn/gshhs/poly-f-1.dat exists.
08:46:25.134 MESSAGE gshhs.cpp:1323 Loading World Chart Q=0 in 0 ms.
```

sebastien-rosset, 2024-07-03:
`gWorldMapLocation` is updated by `Update()` here: https://github.com/OpenCPN/OpenCPN/blob/bebefc8a9f10357b122957ce08b00469048eff36/gui/src/chartdbs.cpp#L1528-L1534

`gshhsCrossesLandInit` is called only once, so even if  `Update()` is called later, that won't have any impact on which GSHHG files are used for the land crossing calculation.

nohal, 2024-07-05:
The correct thing to do here is to stop using GSHHG now that we have the shapefile basemap, which is much more precise.

rgleason, 2024-07-05:
That suggestion leaves weatherrouting_pi high and dry.

IMHO It should be fixed.

sebastien-rosset, 2024-07-05:
> The correct thing to do here is to stop using GSHHG now that we have the shapefile basemap, which is much more precise.

As @rgleason pointed out, the GSHHG data is used by the weather routing plugin. I initially opened this issue as an enhancement, but now I think it's a bug. Using the shapefile basemap might be the right thing to do long term, but it seems this bug is worth fixing anyway. Using the basemap would be a bigger effort with possible performance implications on the weather routing computation time?

nohal, 2024-07-05:
The real bug here is that `PlugIn_GSHHS_CrossesLand` was not updated to use `ShapeBaseChart::CrossesLand` instead of `gshhsCrossesLand`,  which is to be completely removed together with all the other GSHHG related stuff.

sebastien-rosset, 2024-07-05:
> The real bug here is that PlugIn_GSHHS_CrossesLand was not updated to use ShapeBaseChart::CrossesLand instead of gshhsCrossesLand, which is to be completely removed together with all the other GSHHG related stuff.

Ok, sure. This is a bigger change and I will look into it. Is #4041 a reasonable fix until the weather plugin moves calculation to ShapeBaseChart?

rgleason, 2024-07-06:
I hope this release will please keep and improve gshhs high res handling, as per sebastien PR   until alternatives are proven to work effectively with WR.

Hakansv, 2024-07-06:
What I know GSHHS are history in 5.9. ->5.10. Replaced by Shape Base file(s). Pls read nohal's post above to redirect plugins to use the ShapeBaseChart::CrossesLand() function instead. I suppose just forgotten during implementation of the shape base?

nohal, 2024-07-06:
@rgleason what are you talking about here?

Zero changes are needed in WR, it does not give a damn crap about what data OpenCPN used to tell it if land was crossed or not.

bdbcat, 2024-07-06:
Let's defer the discussion of using shapefiles for land crossing detection as used by WR plugin..  Given our current point in the release cycle, this feature cannot reasonably be implemented and adequately tested for O5.10

So, sticking to the original point of this issue:
What is our general confidence level in the draft PR addressing the GSHHS high res data detection and subsequent use by WR_PI?
From my POV, the draft looks good.  But I have not functionally tested.
Thoughts?

rgleason, 2024-07-06:
Lets back up here, a little. I have always expected that the WR user must download the high res gshhs. 

I have always used Pavels chart downloader to do that. It is easy and simple. Then I always create a new chart group called MT which shows only the gshhs high res.

Then I run WR normally. I have always thought that I was then using high res gshhs for land avoidance. 

Is this still true or have I missed something. It has been awhile since I tested this.

bdbcat, 2024-07-06:
According to the OP of this issue, WR_PI always uses GSHHS "crude", no matter what quality of GSHHS is available and used for display.  And that is the point of this Issue, AFAICT.

sebastien-rosset, 2024-07-07:
To clarify, I'm not claiming the weather plugin uses crude resolution GSHHG files under *any* circumstances, for all users and for all operating systems.

A specific reproducible scenario is when you go to Settings -> Charts -> Chart Files, and for some reasons all the directories that had previously been configured are gone from the OpenCPN config, but the charts and GSHHG files are still present on the file system. Then click "Add Directory" to re-add the directories, then the problem occurs.

Here are more details. I was sailing when the issue occurred, so I wasn't focused on software troubleshooting.

1. Enable plugins, including  chart downloader, celestial navigation, dashboard, climatology, autopilot route, deviation, GRIB, javascript, Logbook, OCPN Draw, Plots, Polar, VDR, Weather Fax, WeatherRouting and VMM.
2. Use the chart downloader to download ENC charts, GSHHG base maps.
3. Download multiple GRIBs, including GFS
4. Create routes using Weather Routing
5. Go on the boat, connect OpenCPN to the NMEA bus.
6. Start sailing.
7. After a while, I noticed OpenCPN had crashed with a segmentation fault. I restarted a few times, and the crash occurred again. 
8. After a few OpenCPN restarts, I noticed the charts were no longer displayed. I clicked Settings -> Charts -> Chart Files,. At that point I was out of cell range, so I couldn't download the charts again. But that was ok since I could just clicke "Add directories" to re-add the charts and GSHHG files.

It's only when I came back home that I noticed the weather routing was using crude GSHHG resolution. I know some details are missing here, in particular I wish I could have specific reproducible steps to show how OpenCPN lost its chart config. I have a few core dump files that I need to investigate, and I have identified this logbook plugin bug: https://github.com/rgleason/LogbookKonni_pi/pull/67. For now I have disabled that plugin.

Thank you all for this great software, I really like that it's completely open.

bdbcat, 2024-07-07:
s-r...
In a more normal course of events, without crashing, but loading GSHHS with chartdownloader in the normal manner:
Are the higher definition GHSSH files found and used for WR_PI?

sebastien-rosset, 2024-07-07:
> s-r... In a more normal course of events, without crashing, but loading GSHHS with chartdownloader in the normal manner: Are the higher definition GHSSH files found and used for WR_PI?

I've created a new user and configured OpenCPN with the chart downloader. In that case, the weather routing plugin does use higher definition files.

I've noticed there are several other "normal" ways to install charts. For example, when OpenCPN runs for the first time, a wizard attempts to detect NMEA sources and it also asks the user to enter a directory that contains charts. Second, the user can also go to settings -> charts -> charts files -> add directory. There may be other normal ways to add charts that I am not aware of.

bdbcat, 2024-07-07:
Since under "normal" conditions, OCPN and WR_PI do the "right thing", let us defer this bug fix and potential PR for OpenCPN Version 5.12.
Thanks for all the analysis.

sebastien-rosset, 2024-08-12:
> The real bug here is that PlugIn_GSHHS_CrossesLand was not updated to use ShapeBaseChart::CrossesLand instead of gshhsCrossesLand, which is to be completely removed together with all the other GSHHG related stuff.

@nohal , now that 5.10 has been released, is a plugin allowed to invoke this function which is declared in `gui/include/gui/shapefile_basemap.h`? Or are you suggesting `ShapeBaseChart::CrossesLand` should somehow be exposed in `ocpn_plugin.h`?

nohal, 2024-08-13:
The plugin is only allowed to call `PlugIn_GSHHS_CrossesLand`, which internally should call `ShapeBaseChart::CrossesLand`

sebastien-rosset, 2024-08-13:
Unfortunately the name `PlugIn_GSHHS_CrossesLand` has the acronym `GSHHS`, whereas `ShapeBaseChart::CrossesLand` is more generic, e.g., it may be GSHHS, OSM or any other coastline shapefiles that may be added in the future. Do you want to overload that function? Or create a new `PlugIn_CrossesLand` function in `ocpn_plugin.h` that invokes `ShapeBaseChart::CrossesLand`?

nohal, 2024-08-13:
I have absolutely no intention to change the API because of this. The name was "wrong" for years since GSHHS was renamed to GSHHG and I did not notice any impact on it's functionality.

bdbcat, 2024-08-13:
No reason to change the name.  If we do, we get immediate breakage of weather-routing_pi.
Might be persuaded to add a new method:
`bool BackingChartCrossesLand( int back_chart_type = BACK_CHART_SHAPELIB);`
or so.
Some future plugin might like that....

sebastien-rosset, 2025-02-25:
I have modified `PlugIn_GSHHS_CrossesLand` to call `gShapeBasemap.CrossesLand`. This works, however, this causes the weather routing plugin to be extremely slow. 

```c++
bool PlugIn_GSHHS_CrossesLand(double lat1, double lon1, double lat2,
                              double lon2) {
  if (gShapeBasemap.IsUsable()) {
    return gShapeBasemap.CrossesLand(lat1, lon1, lat2, lon2);
  } else {
    // Fall back to the GSHHS data.
    static bool loaded = false;
    if (!loaded) {
      gshhsCrossesLandInit();
      loaded = true;
    }
    return gshhsCrossesLand(lat1, lon1, lat2, lon2);
  }
}
```

rgleason, 2025-02-26:
And not to mention the 600 mb of shapefiles that will be needed for smaller systems like RPI or Android.

nohal, 2025-02-26:
> And not to mention the 600 mb of shapefiles that will be needed for smaller systems like RPI or Android.

Why should that be needed?

rgleason, 2025-02-26:
Oh, I see, it is already downloaded and used by OpenCPN?   And there is room for this on Android and RPI?

nohal, 2025-02-26:
> Oh, I see, it is already downloaded and used by OpenCPN? And there is room for this on Android and RPI?

Of course the shapefiles are used by OpenCPN already.
Of course to use  the high resolution datasets you need to have space and computation power for them, regardless of platform.
Of course you do not need to have the full resolution dataset, especially if you do not have the storage or computation power to use it.
I would strongly suggest you actually have a look what the full resolution shapefile dataset contains and compare it to the outdated and by today's means terribly inaccurate GSHHG. It might perhaps save us from repeating this conversation over and over again.

rgleason, 2025-02-26:
Basemap_shp  is 15.2 mb
gshhs is  5.12 mb
charts/gshhg  215mb

Don't know yet how to download detailed shapefiles.  Sebastien says they are 600mb
Don't know or have any idea how detailed Basemap.shp is with 15.2mb, compared to charts/gshhg (detailed) 215mb.
Also don't know or have any idea how detailed high resolution shapefiles are comparte to charts/gshhg (detailed).

nohal, 2025-02-26:
An apple is not an orange. This is not a format conversion, the shapefiles are simply a different dataset, much more precise and much more detailed one than GSHHG. Not surprisingly the more detail also means it is bigger. Stop comparing megabytes, which is a completely useless metric here, and simply look at them.
In QGIS you can load both GSHHG (from https://www.soest.hawaii.edu/pwessel/gshhg/ also available in shapefile format) and the OSM shapefiles we replaced them with and immediately see the difference in quality. What more "information" do you need?

rgleason, 2025-02-26:
Thank you Pavel, I when I get some time I will travel that road. I will drop it now.

nohal, 2025-02-26:
For the next shapefile basemap update I plan extending the metadata with bounding box coordinates of each polygon, that alone should allow for a major speed up. Of course all the other proposed optimizations make sense as well. Tuning the dataset is a lengthy process as it takes days to generate the high and full resolution tiles even on a very beefy machine...

sebastien-rosset, 2025-02-26:
BTW, it looks like there is a missing function to check if the data has been loaded successfully (or maybe I don't know about it). When you start OpenCPN, the shapefile loading process is asynchronous and can take a minute or more. The user can start a weather routing simulation, which calls `ShapeBaseChart::CrossesLand()`. That function does not have a way to return errors.

rgleason, 2025-02-26:
I think wx route. should use gshhs low res and suggest using low res shapefile.

Then next step would be using high res gshhs, then high res shape

bdbcat, 2025-02-26:
Yes.
Very few users will be using the full resolution 600 MB shapefiles.  Just too heavy for everyday use.
So check the performance (speed and accuracy) of wxRoute using high res (15 MB) shapefiles.
IMHO, no real need to spend extraordinary effort the get acceptable wxRoute performance on full resolution shapefiles.

nohal, 2025-02-26:
> I think wx route. should use gshhs low res and suggest using low res shapefile.
> 
> Then next step would be using high res gshhs, then high res shape

GSHHG is miles off in places, there is absolutely no reason to use it if we have better data, ever.

sebastien-rosset, 2025-02-26:
> So check the performance (speed and accuracy) of wxRoute using high res (15 MB) shapefiles.

I'll check that in my local workspace, but unfortunately `PlugIn_GSHHS_CrossesLand` has no argument to specify the desired data quality.

> GSHHG is miles off in places, there is absolutely no reason to use it if we have better data, ever.

Agreed in principle, but on the other hand I would certainly not recommend merging my PR as is. It's just too slow. It's great that you are working on a better dataset, and I'll try to identify optimization opportunities.

bdbcat, 2025-02-26:
Sebastian..
You will need to temporarily move the full res shapefile aside, so that the highest found becomes "high" instead of "full".

sebastien-rosset, 2025-02-26:
> Sebastian.. You will need to temporarily move the full res shapefile aside, so that the highest found becomes "high" instead of "full".

Yes, I can definitely do that in my local workspace. Sorry I wasn't clear, I was trying to say even if I do that and performance is reasonable at that level, I don't think we should merge code that will cause hours of high CPU usage when the full shapefiles are installed. Either we should add optimizations, or there should be a new overloaded `PlugIn_GSHHS_CrossesLand` API which has more controls (such as data quality, and while we are at it, allows for vectorized operations and other improvements).

bdbcat, 2025-02-26:
"new overloaded PlugIn_GSHHS_CrossesLand API which has more controls (such as data quality,"
Which immediately breaks the wxRoute plugin.
Anyway, lets validate the performance first.  The we decide.

sebastien-rosset, 2025-02-26:
I removed the "full" resolution and tried with the "high" resolution. It's still very very slow. 

Instead of using highly detailed coastline data for route planning, we could create simplified "safety polygons" that represent coastlines plus a safety margin, which might be based on the coastline or depth.

Safety polygons can have far fewer points than detailed coastlines while serving the same purpose for navigation. One challenge is that it would require pre-processing to generate simplified coastlines.

sebastien-rosset, 2025-02-26:
It looks like`ShapeBaseChart::CrossesLand` is not only slow, it's also wrong.

One issue is the `latmax`, `lonmax` values seem to be wrong:

https://github.com/OpenCPN/OpenCPN/blob/060633f15922f520a81d872fd2b4a74a9216d65b/gui/src/shapefile_basemap.cpp#L528-L529

```diff
bool ShapeBaseChart::CrossesLand(double &lat1, double &lon1, double &lat2,
                                 double &lon2) {
  double latmin = std::min(lat1, lat2);
  double lonmin = std::min(lon1, lon2);
-  double latmax = std::min(lat1, lat2);
-  double lonmax = std::min(lon1, lon2);
+  double latmax = std::max(lat1, lat2);
+  double lonmax = std::max(lon1, lon2);
```

Even with this fix, when `_is_tiled` is `true`, I don't see `PolygonLineIntersect` ever returning `true`.

sebastien-rosset, 2025-02-26:
@nohal , if you have some insights for https://github.com/OpenCPN/OpenCPN/pull/4385#issuecomment-2686034948, that would be great. I just want to make sure my understanding is correct. It turns out I am facing 3 problems with the use of `ShapeBaseChart::CrossesLand()`:
1. Severe performance issue.
2. `CrossesLand()` never returns true even though clearly some paths in the WR simulation cross land boundaries.
3. Occasional crashes in `gui/src/shapefile_basemap.cpp`, not very predictable.

sebastien-rosset, 2025-02-27:
One correction: the full resolution OSMSHP file is 1.79GB, not 600MB. This is the basemap_full.shp file.
The high resolution is 318.3MB

sebastien-rosset, 2025-02-27:
@bdbcat , @nohal , I've listed some issues I've discovered in #4385 . The `ShapeBaseChart::CrossesLand()` does not work as intended because:
1. The polygons are not necessarily contained within the bounds of the tiles. This means it's not sufficient to loop through the polygon in the tile that contain the segment.
2. The presence of tiles in the index seems to depend on whether ENC charts are loaded or not.
3. It's slow.
4. The more I think about it, the less it makes sense to use the best quality OSMSHP, especially for weather routing. The weather routing plugin should have a means to control the resolution level it needs based on how far the routing segment is from the coastline.

nohal, 2025-02-27:
I'm especially interested in 1, are you really sure? I see no chance the polygons are not cut to the tiles in the workflow producing them.
And 2, the re is no relation between the shapefile basemap code and ENC, how could it relate?

sebastien-rosset, 2025-03-01:
> I'm especially interested in 1, are you really sure? I see no chance the polygons are not cut to the tiles in the workflow producing them.

I did more specific analysis by looking at the crud OSMSHP polygons. It turns out the **{lat, lon}** of the tiles represents the top-left corner of the tile, not the bottom-left corner. With that, I found out the OSMSHP file has:
1. 476 tiles
1. 21,683 polygons
1. 6 of the tiles have boundaries that exceed the tiles.

sebastien-rosset, 2025-03-01:
I now confirm the grid tiles are indexed by the top-left corner of the tiles. Previously it assumed the tiles were indexed by the bottom-left corner, which resulted in `CrossesLand()` function to always return `false`.

With these changes, the `CrossesLand()` function is now working as expected. Below is a weather routing simulation that correctly avoids land.

[image]

sebastien-rosset, 2025-03-01:
@bdbcat , I am proposing the following sequence of changes:
1. Merge #4385 - Fix several issues in the `ShapeBaseChart::CrossesLand` implementation.
2. Merge #4040 - Fix GSHHS problems. 
3. Follow-up PRs that I will work on:
   - Improve the performance of OSMSHP land crossing. There are specific performance optimizations in the GSHHS implementation that could be applied to OSMSHP.
   - Once the performance of OSMSHP is acceptable, change the `PlugIn_GSHHS_CrossesLand` implementation to use `ShapeBaseChart::CrossesLand()` as first priority if it's available, then GSHHS as a fallback. There could be further discussions about comparing the resolution of available files.
   - Depending on how the above goes, implement user controls, such as controls for the shapefile resolutions that are used to detect land crossing. I don't think it makes sense to use the highest resolution (e.g. OSMSHP full resolution) unless a very serious improvement to the algorithm is done.

closed 2025-03-15 by sebastien-rosset