OpenCPN/OpenCPN #4042: User Standard Objects does not save "Quality of data" selection

bc38, 2024-07-04:
I uncheck the "Quality of data" option and after rebooting the CPN the checkbox appears again!

The localization of the User Standard Objects list has also disappeared

[image]

Hakansv, 2024-07-05:
Since all user isn't comfortable changing the "User Standard" options the quality of data has got menu options saved to config overriding the "User standard" settings.
Please use the menu bar or the per canvas MUI-menu instead. 

[image]

rgleason, 2024-07-05:
What? 
I don't think that solves the issue.
Shouldn't these controls be working together in concert?

Otherwise what is a user to expect?

bc38, 2024-07-05:
Hakan, thanks, I solved the problem for myself!

Yes, I agree with rgleason, it should work forcibly according to the configured objects in the list, even though this is one object that does not behave normally!

Hakansv, 2024-07-05:
Could be.
There have been several patches to the Quality of data option the last years. ("The more cooks, the worse the soup" :) )
Jokes aside. This may be a future request?

bc38, 2024-07-05:
> Could be. There have been several patches to the Quality of data option the last years. ("The more cooks, the worse the soup" :) ) Jokes aside. This may be a future request?

Yes, it is quite! But let's wait for Dave's comment!
There is also a question about localizing the list of objects

bdbcat, 2024-07-05:
"Hakan, thanks, I solved the problem for myself!"
How did you do that?

Also, on the localized list of objects.
I actually do not think that these should be localized.  They text come from official IHO definitions for S52.
Did we localize these in previous OCPN versions?

bc38, 2024-07-05:
> How did you do that?
removed the selection from the map display options menu, as recommended by Hakan.
> Also, on the localized list of objects. I actually do not think that these should be localized. They text come from official IHO definitions for S52. Did we localize these in previous OCPN versions?
Yes, there was localization!
Of course, I don’t mind anything in English, as an ENC cartographer I understand everything. BUT what about other users!? Maybe I should ask on the forum?

bc38, 2024-07-05:
> "Hakan, thanks, I solved the problem for myself!" How did you do that?
> 
> Also, on the localized list of objects. I actually do not think that these should be localized. They text come from official IHO definitions for S52. Did we localize these in previous OCPN versions?

Removed the selection from the map display options menu, as recommended by Hakan.
There was localization!

bdbcat, 2024-07-05:
Where, what version, was localization?

bc38, 2024-07-06:
Indeed, in the latest versions there is no localization of object selection.  Perhaps I'm confused because...  Did translations in files describing objects and attributes, folder s57

bdbcat, 2024-07-10:
Please close as finished.

closed 2024-07-10 by bc38