OpenCPN/OpenCPN #4043: 5.9 Beta crashes when panning

svMigration, 2024-07-04:
**Describe the bug**
OCPN crashes when panning chart. Related to viewing MBTiles.
This is with the temp beta Dave created to test the chart grouping change. But it has happened with previous Betas (see reports in Beta thread on CruisersForum)

**To Reproduce**
Steps to reproduce the behavior:

Not reproducible on demand, unfortunately.

1. Have several MBTiles loaded.
2. Drag or pan chart.
3. Occasionally OCPN crashes.

**Screenshots**
Error
[image]
After saying OK
[image]
Then OCPN crashes

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version 22

[Crash Reports 2024-07-03.zip](https://github.com/user-attachments/files/16101858/Crash.Reports.2024-07-03.zip)

svMigration, 2024-07-05:
Was able to crash OCPN consistently -- 3 times in a row in less than 3 minutes by dragging chart around. Sometimes happened when displaying MBTiles but also with CM93. 
[Crash Reports 2024-07-05.zip](https://github.com/user-attachments/files/16113257/Crash.Reports.2024-07-05.zip)

This is with the interim Beta Dave made with the Chart Grouping change.

svMigration, 2024-07-09:
All charts shown (Chart Group 0). Quilting On. Panning near Mexico.
See [https://github.com/OpenCPN/OpenCPN/issues/4034#issuecomment-2216646981](url) for what charts are loaded.

Crash Report:
[Crash Report 2024-07-08.zip](https://github.com/user-attachments/files/16138797/Crash.Report.2024-07-08.zip)

[opencpn.log](https://github.com/user-attachments/files/16138810/opencpn.log)

svMigration, 2024-07-09:
Hi All,
I notice in the crashrpt logs there is this line:
[encoding_video]
Desktop video recording disabled; skipping.

How can this be enabled? Maybe it will help for debugging?
Thanks,
B

bdbcat, 2024-07-09:
About the crash reports:
They are useful, to a degree.  However, the actual fault locations are totally unrelated to MBtiles, or cm93.  This makes me think that the problem is some memory fault in the MBTiles code, writing outside of its allowed area.  That is my focus now.  I will try to add some more memory fault checking and protection logic, with some kind of logging.

svMigration, 2024-07-09:
Hi Dave,

Let me know if there is any specific debugging you want me to do.

B

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

bdbcat, 2024-07-09:
Bruce...
Please try this one:

https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.9.0-0+1731.2229ce1_setup.exe

It may not crash.  It may show some "blank" tiles.  It may log some errors like:"MbTiles: SE_Exception"
Maybe we will learn something.

Thanks
Dave

svMigration, 2024-07-10:
Hi Dave,
I installed and ran. I got a crash while displaying CM93 charts. Zoomed very far out and panning over by Mexico.
Crash Report: 
[Crash Report 2024-07-09a.zip](https://github.com/user-attachments/files/16155605/Crash.Report.2024-07-09a.zip)

So I thought maybe this is not about MBTiles but CM93. So I removed CM93 from the loaded charts. It took longer to crash but it was a different crash this time. Displaying ALL chart group. Got an error message. Screen shot below. After saying OK to error, I could still click on menus, toolbar, etc. But could not move the chart. The chart display was frozen and didn't change even if I changed chart group. When I chose EXIT from menu, did get a crash. Crash Report below.

[Crash Report 2024-07-09b.zip](https://github.com/user-attachments/files/16155612/Crash.Report.2024-07-09b.zip)
[image]

bdbcat, 2024-07-10:
Bruce...
Will you try a test with only cm93, and no MBTiles?
We need to isolate the problem more closely.  The way to do that is to simplify the config to the **simplest** possible setup that will still produce the error.
As you might imagine, no-one else is reporting stability problems with O5.9, especially on cm93.

Thanks
Dave

svMigration, 2024-07-10:
Hi Dave,
OK. More info. First of all, I should have mentioned that this may not be a 5.9 problem. I've had similar crashes in the past but ignored them. There are way more know because I'm trying to find the conditions that cause them so it seems more unstable.
Here's the news.
1. Tried to crash with no plug-ins: CRASHED
2. Tried to crash in safe mode: CRASHED
3. Removed all charts except CM93: **N0 Crash**
4. Added the Complete South Pacific Overview Charts. So now running with CM93 and 4 MBTiles.
[https://www.mediafire.com/file_premium/jldlsrftsasf0ab/Pacific_Overview_TCL10-23.zip/file](url) :
CRASHED
Crash Report
[2024-07-10a.zip](https://github.com/user-attachments/files/16169685/2024-07-10a.zip)

I create the crash by zooming in and out quickly with my mouse wheel -- all the way out and all the way in sometimes -- and dragging the chart around.

5. Removed CM93 and only had the 4 SP Overview Charts loaded: CRASHED
Crash Report:
[2024-07-10b.zip](https://github.com/user-attachments/files/16169693/2024-07-10b.zip)

So progress.
Let me know what you would like me to try next.
B

svMigration, 2024-07-11:
Hi Dave,
Just after I sent you the above, we moved to a new anchorage 100 meters away. I had OCPN running down below. When I came back down, it has crashed. I had added the MBTiles for the Societies. Here's the crash report.
B
[2024-07-10c.zip](https://github.com/user-attachments/files/16169942/2024-07-10c.zip)

bdbcat, 2024-07-11:
Bruce...
The link for the Pacific Overview MBTiles set is corrupted.  Please repost.

bdbcat, 2024-07-11:
Bruce...
All three crash dumps showed a crash in exactly the same location.  I added code to preclude a crash on this particular point, and it is building now.  Will have a new test article in 20 minutes or so.
To be honest, I cannot see how the crash is happening.  It relates to drawing of the lower right "hamburger" icon.  Never seen before.  It may be a red herring, caused by some other problem.  But for now, I am following the evidence.
Please stand by.

bdbcat, 2024-07-11:
Bruce...
This one is ready for test.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.9.0-0+1739.1bbf84c_setup.exe

Thanks
Dave

svMigration, 2024-07-11:
Pacific Overview files:
https://www.mediafire.com/file_premium/jldlsrftsasf0ab/Pacific_Overview_TCL10-23.zip/file
Will test new version now.
B

svMigration, 2024-07-11:
Hi Dave,
I rebooted my computer to make sure it everything was clean.
Started this new version and did my zooming/panning thing. After a bit I got this error:
[image]

I had to hit OK 4 times. Then OCPN was completely frozen and I had to end task in Task Manager.
Same charts were loaded (SP Overview and Societies).
No crash report was created.
Standing by....
B

svMigration, 2024-07-11:
wait... crash report coming...

svMigration, 2024-07-11:
[2024-07-10d.zip](https://github.com/user-attachments/files/16170690/2024-07-10d.zip)

bdbcat, 2024-07-11:
OK. got it.
Digging in

svMigration, 2024-07-11:
Reproducible! Dave, I was able to reproduce 5 times in a row by just holding down the + key and zooming in all the way and then holding down - and zooming out all the way. I think it only does this with Quilting on. I didn't pan or use the mouse. After the last crash, I restarted OCPN and this is what it happened immediately:
[image]
Hit OK 4 times and OCPN is frozen.  
Let me know if you want me to send anything. I'll leave it in this state for a while. Gonna take a break off the computer for a bit.

bdbcat, 2024-07-11:
The crash report again inconclusive.
I will try exactly your actions, with exactly your tiles, and see what happens.

svMigration, 2024-07-11:
Dave, will be off the boat all day so won't be able to test until this evening at the earliest. (UTC +10). 
B

bdbcat, 2024-07-11:
B
Following your minimal procedure exactly, I did finally reproduce the fault, with almost identical results.
This allowed me to discover the root cause of this issue.
The root cause is this:  The newer MBTiles threading model (introduced in O5.9, Feb. 2024) is very efficient at gobbling up and processing tiles from the sql database in the MBTiles set.  So efficient that the memory requirement while thread tile processing is occurring spikes to a very large number, for these tile in excess of 2 GB instantaneously.  This overwhelms the Windows memory manager, causing faults in memory allocation.  The result is memory corruption, at random spots in the OCPN data storage.
The adopted solution for O5.10 is to simply throttle the threads so that they do not attempt to process so many tiles at one time.  This keeps the instantaneous memory requirement down to a reasonable level, something less than 1 GB.
There is a small visual cost, however.  You may occasionally see a "patchwork" display of tiles when zoomed way out on a large tile set, like Pacific overview.  The patchwork get filled in as the tile threads "catch up" with the rendering request.  This only happens with high zoom-out, and generally only on startup of OCPN.
I will have new test code ready shortly.

bdbcat, 2024-07-11:
B
Here is the new test version.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.9.0-0+1740.3fc59cc_setup.exe

Good Luck
Dave

svMigration, 2024-07-12:
Hi Dave,
Glad you found that problem. Unfortunately, not solved on this end. I installed new version and immediately got 3 crashes by just zooming in and out with the keyboard.
B
Reports:
[2024-07-11.zip](https://github.com/user-attachments/files/16185747/2024-07-11.zip)

bdbcat, 2024-07-12:
From looking at the crash dumps, the problem is exactly the same as described.  We are estimating how much to throttle the tile thread.  Maybe more is needed.  What are the specs of your machine?
I'll sleep on that idea.
For fun, will you try some tests without those 4 super-huge Pacific Overview tilesets loaded?  Loading the full zoomed out content of those 4 nice tilesets kicks off parallel tile processing of about 13,000 tiles, and that is a little strenuous even for my 16 core, 32GB dev system.  Just sayin....

svMigration, 2024-07-12:
Hi Dave,
Specs:
Lenovo Slim 7 ProX 14ARH7
Processor	AMD Ryzen 7 6800HS Creator Edition   3.20 GHz
Installed RAM	32.0 GB (27.7 GB usable)
Device ID	B413E9EF-45A2-4298-AE97-EBAFA6A20645
Product ID	00355-60848-42740-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points
Edition	Windows 11 Pro
Version	22H2
Installed on	‎4/‎5/‎2023
OS build	22621.3880
Serial number	PF47W9ZA
Experience	Windows Feature Experience Pack 1000.22700.1020.0

Will try and report back.
B

svMigration, 2024-07-12:
Hi Dave,
OK. With just the Societies loaded it's _very_ difficult to get it to crash but I can eventually,. I have to go in and out a lot and then do some dragging/panning and fussing about. 

I'm fairly certain the system only crashes when Quilting is on and all charts are active on the chart bar. That makes sense as OCPN is trying to deal with a whole lot of tiles in that scenario. And that's why this doesn't present itself very often. There is no reason to have quilting on and all charts active. I try very hard to teach users of MBTiles to use chart groups.
B

bc38, 2024-07-12:
I also had this problem on my set of mbtiles maps.
The second time CPN froze
no more problems, maps load without problems

svMigration, 2024-07-13:
Hi Dave,
We have a wx window and must leave Sunday morning. We will have no internet for a month but will be able to receive emails if you have any questions. wdc6670  at  sailmail.  I have 5.9.2 as well as 5.9.0-0+1740.3fc59cc (the last one from above) on my computer.
B

bdbcat, 2024-07-14:
OK, thanks for your help.
Have a great cruise. :)

bdbcat, 2024-07-14:
Closing, pending further practical testing

closed 2024-07-14 by bdbcat