OpenCPN/OpenCPN #4045: ChartDownloader not always showing download progress

aviatorhh, 2024-07-05:
**Describe the bug**
When downloading some charts the download progress stays at zero (sometimes jumps [see video]).
This does happen to certain chart sources only.

**To Reproduce**
- Got to Options -> Charts Downloader
- Select catalog -> Add Catalog - choose ChartsCatalogs The Netherlands Inland ENC
- Click Update and thereafter go to Download Charts and select Waddenzee (or others)
- Download selected charts

**Expected behavior**
The amount of data downloaded shall sum up to the total amount expected

**Screenshots**

https://github.com/OpenCPN/OpenCPN/assets/13214417/09946b7c-d694-4952-9a8a-77880e3ecb0b

**Desktop (please complete the following information if applicable):**
 - Debian 12
 - on PC and RPi

**Additional context**
This behaviour is ongoing for some time now

leamas, 2024-07-05:
This is #3987 which supposedly is fixed. Which version are you on?

aviatorhh, 2024-07-05:
just pulled

leamas, 2024-07-05:
> just pulled

Commit?

aviatorhh, 2024-07-05:
dfca3d4d802f25aff3ee372b5644829eac449a5f
:)

leamas, 2024-07-05:
No time to dig now,  that's we you should always state the commit

aviatorhh, 2024-07-05:
I do know but days were long recently - my sincere apologies

nohal, 2024-07-05:
That Java miracle serving the NL IENC website, besides barely working, does not send the `Content-Length` header, so total size is impossible to obtain. Increasing the progress counter should be possible though.

aviatorhh, 2024-07-05:
You are right
`curl -I "https://vaarweginformatie.nl/fdd/main/wicket/resource/org.apache.wicket.Application/downloadfileResource?fileId=3979366550"`

```
HTTP/1.1 200 OK
Server: nginx/1.20.1
Date: Fri, 05 Jul 2024 19:42:25 GMT
Content-Type: application/x-zip-compressed
Connection: keep-alive
Set-Cookie: FISCookie=16aHJlcz0wLGxhbmd1YWdlPW5sLHZlcnNpb249MSxyZXNvbHV0aW9uPWhpZ2gsdnJlcz0w;Path=/fdd;Expires=Wed, 01-Jan-25 20:42:25 GMT;HttpOnly
Set-Cookie: JSESSIONID=node1nm1684s5aennphy68zhhytpp9412.node1; Path=/fdd; Secure
Expires: Fri, 05 Jul 2024 19:42:25 GMT
Cache-Control: must-revalidate
Pragma: no-cache
Content-Disposition: attachment; filename="20240704_U7Inland_Waddenzee_week 27_NL.zip"
Strict-Transport-Security: max-age=31536000
```

but as you said a running progress counter would be a good solution.

commit 1318d2ee (2025-05-03): chartdlr: handle missing total size - #4045

The GUI bug is caused by spurious events reporting 0 transferred bytes.
Ignore them in the GUI, the transferred bytes counter should never
decrease

Closes: #4045
closed 2025-05-03 by leamas