OpenCPN/OpenCPN #4046: Put "Name:" above "Chart File:" in Chart Key hover-over display

svMigration, 2024-07-06:
**Is your feature request related to a problem? Please describe.**
No..]

**Describe the solution you'd like**
Move the "Name:" above "Chart File:" in Chart Key hover-over display. It is less confusing to read the name than the  (often very long) file path and name.

Hakansv, 2024-07-06:
On the other hand is the text formatted so the name is always visible even on a limited space.

[image]

svMigration, 2024-07-06:
Referring to the top part:
[image]

svMigration, 2024-07-06:
Also just noticed, in the screen shot above of the FIJI MBTile, the actual chart file name is:

svMigration, 2025-04-24:
This is becoming more and more important as people use MBTiles and have a large number if charts open.

leamas, 2025-04-24:
Looking at the screenshot one could argue the file path should be moved to bottom, it's probably not what you want while sailing. OTOH, important stuff like "Update" deserves a more visibility at top. Perhaps something like

```
Name:
Scale: 
Updated:
Source Edition:
Depth units:
Soundings:
Id: 
Projection:
Chart file:
```

closed 2025-11-16 by svMigration