OpenCPN/OpenCPN #4047: sudo make install differs on ldd check

aviatorhh, 2024-07-06:
**Describe the bug**
When doing sudo make install the linker does not seem to respect /usr/local.
Checking the opencpn binary in the O build directory after a fresh build (cmake, make) it shows references to /usr/local libraries (which is fine).
After doing a ```sudo make install``` the binary of O that is being placed in /usr/local/bin is showing a different linkage to the libs (ldd)

**To Reproduce**
- build wxwidgets with ```configure --with-opengl```and install to /usr/local
- compile O and do a ```sudo make install```

**Expected behavior**
O running against /usr/local wx libs

**Desktop (please complete the following information if applicable):**
 - RPi debian 12

**Additional context**
commit dfca3d4d802f25aff3ee372b5644829eac449a5f

nohal, 2024-07-07:
I suspect you might just misunderstand how this stuff works.

The Linux dynamic linker, while starting an executable, looks for libraries in a set of directories (which ones and in which order you can see in the output of `ld --verbose | grep SEARCH`).
OpenCPN's `make install` does exactly what it is supposed to do, if you want to change the order of library search paths, reconfigure your `ld.so` accordingly in `/etc/ld.so.conf` or set `LD_LIBRARY_PATH` environment variable as you like while staring OpenCPN.

leamas, 2024-07-07:
As  @nohal said.

Also, the set of  plugins loaded differs if the binary is installed in /usr/local/whatever compared to when  sitting in the build directory (plugins are loaded from $ORIGIN/../lib/opencpn, $ORIGIN is the install location). Different plugins might of course mean different loaded libraries depending on their actual deps.

aviatorhh, 2024-07-07:
Indeed my C++ and environmental knowledge is about to develop. Thanks for clarifying this issue. So I will have to dig a bit more into this. I indeed assumed make install is "just" a copy-into-place method somehow.

```
$ ld --verbose | grep SEARCH
SEARCH_DIR("=/usr/local/lib/aarch64-linux-gnu"); SEARCH_DIR("=/lib/aarch64-linux-gnu"); SEARCH_DIR("=/usr/lib/aarch64-linux-gnu"); SEARCH_DIR("=/usr/local/lib"); SEARCH_DIR("=/lib"); SEARCH_DIR("=/usr/lib"); SEARCH_DIR("=/usr/aarch64-linux-gnu/lib");
```

I wonder why I get two different results on ldd of the same binary one sitting in my build directory and the other under /usr/local.
So this issue for sure is not a bug as it has nothing to do with the O install procedure itself.

leamas, 2024-07-07:
>  So this issue for sure is not a bug as it has nothing to do with the O install procedure itself.

Indeed. Converting to discussion