OpenCPN/OpenCPN #4050: 5.9.0 for legacy Intel Macs on MacOS 10.13.6: o-charts API error codes 50 and 57

ThorHei, 2024-07-09:
When trying to install o-charts with the plugin in OpenCPN 5.9.0 for legacy Intel Macs on MacOS 10.13.6 I get API error codes 50 or 57. I was able to install licensed o-charts by off-line installation but when trying to update the chart list I still get the API error code 50. Is that a known issue and is there any fix available?

Steps to reproduce the behavior:
1. Go to o-chart plugin
2. Click on "update chart list"
3. o-chart API error code 50 or 57

MacBookAir with MacOS 10.13.6

[image]

yalla161, 2024-10-14:
Hey!
Im having exactly the same Problem.. I have the same MacOS and the same openCPN.. Have you gathered any information since July?
Cheers,

ThorHei, 2024-10-16:
I am using OpenCPN 5.6.2. There it is working without problems...