OpenCPN/OpenCPN #4054: Show visible sector lights won't turn off

bc38, 2024-07-12:
Doesn't respond to checkbox in Chart Panel Options

[image]

bdbcat, 2024-07-12:
Please read the docs for this option.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chart_panel:chart_panel_options:vector_charts&s[]=sector

Hakansv, 2024-07-12:
"Show visible sectors light"
For me and other the logic is clear.
"Visible" means leading sectors visible from the ship. That's when the ship is in the area of a sector(s) range.

The cursor is not involved here. Whenever you "mouse over" a lighthouse with leading sector(s) it's shown if "Lights" is on.

closed 2024-07-13 by bc38