OpenCPN/OpenCPN #4056: Toolbar, MUI bar does not change background color via GREY3

bc38, 2024-07-14:
Win11

bdbcat, 2024-07-14:
Not sure what the issue is.
Picture?

bc38, 2024-07-15:
I make the background white for GREY3
Where does the gray color for the background come from?
[image]

bdbcat, 2024-07-16:
The background color is now fixed black.
I will revert to GREY3.

bc38, 2024-07-16:
> The background color is now fixed black. I will revert to GREY3.

I would like to somehow customize the background.
Dave, where is the background color described in the code now?

bdbcat, 2024-07-16:
Refactored, both toolbars now controlled by "GREY3".
For example, in MUIBar.cpp:
    m_backcolor = GetGlobalColor("GREY3");

bdbcat, 2024-07-17:
Seems pretty unrelated to this issue.
Ready to close the "GREY3" question?

closed 2024-07-17 by bc38

transmitterdan, 2024-07-18:
> Can't check when building via winConfig.bat --release errors The C compiler identification is unknown The CXX compiler identification is unknown CMake Error in CMakeLists.txt: No CMAKE_C_COMPILER could be found. CMake Error in CMakeLists.txt: No CMAKE_CXX_COMPILER could be found.

This can happen if you try to run winConfig.bat from a regular CMD prompt.  Use the "x86 Native Tools Command Prompt for VS 2022" instead.

I plan to fix this in the next version of winConfig.bat.