OpenCPN/OpenCPN #4058: AisTargetList does not save the docked position

bc38, 2024-07-14:
I dock the window to the bottom of the screen and click close.
When AisTargetList is called again, the window is displayed in an undocked manner at the bottom of the screen
Win11

bdbcat, 2024-07-14:
Cannot reproduce.  Works fine here on W11 and linux.
Pictures?

Anyone else showing this behavior?

Hakansv, 2024-07-14:
On W10
1. The AIS target windows is here opened at same position and size after each "close".
2. @bc38.. What does "dock" mean for the AIS target list? I can "dock" a dashboard but haven't found the same function for this list?

bc38, 2024-07-15:
[VIDEO](https://www.dropbox.com/scl/fi/sd5swm5dy5fmyijri91d0/2024-07-15-122236.mp4?rlkey=8oedbxglztctx86fmdjav9cah&st=oziy8n2m&dl=0)

Hakansv, 2024-07-15:
Ahaa.. That dock function I was not aware of. Thanks.
Now I'm not sure when I personally would want to use it but that's another thing.
If I dock in single canvas mode and close O it's again docked on next O start when I open AIS target list.
The docked position is however not recreated in dual canvas mode. This may be a request for after v5.10?

bc38, 2024-07-15:
> Ага.. Я не знал об этой функции стыковки. Спасибо. Теперь я не уверен, когда лично я хотел бы ее использовать, но это уже другое дело. Если я стыкую в режиме одного холста и закрываю O, он снова стыкуется при следующем запуске O, когда я открываю список целей AIS. Однако стыковочная позиция не воссоздается в режиме двух холстов. Это может быть запросом для версии после v5.10?

The function is useful so that the windows do not dangle in the middle of the map screen. I wish I could add docking to the windows of the route and mark manager, route properties, and tracks in the future!

There was the same problem in single canvas mode.
After several attempts to dock, the position was still maintained. 

Made a backup copy of the AisTargetList location parameters
"AISTargetListPerspective=name=AISTargetList;caption=AIS Target;state=2103264;dir=3;layer=0;row=1;pos=0;prop=100000;bestw=700;besth=150;minw=-1;minh =-1;maxw=-1;maxh=-1;floatx=551;floaty=1103;floatw=770;floath=209"

It's possible that some work needs to be done somewhere.
So far I'm happy with this solution
closing....

closed 2024-07-15 by bc38