OpenCPN/OpenCPN #4063: For Windows versions, use updated redistributables

TwoCanPlugIn, 2024-07-20:
OpenCPN installs out of date Windows redistributables, in particular msvcp140.dll.

Perhaps I'm the only plugin developer using mutexes, but the following issue bit me in the bum.

Refer to the Microsoft release Notes at [https://github.com/microsoft/STL/releases/tag/vs-2022-17.10](https://github.com/microsoft/STL/releases/tag/vs-2022-17.10)

> Fixed mutex's constructor to be constexpr. https://github.com/microsoft/STL/pull/3824 https://github.com/microsoft/STL/pull/4000 https://github.com/microsoft/STL/pull/4339
Note: Programs that aren't following the documented [restrictions on binary compatibility](https://learn.microsoft.com/en-us/cpp/porting/binary-compat-2015-2017?view=msvc-170) may encounter null dereferences in mutex machinery. You must follow this rule:
When you mix binaries built by different supported versions of the toolset, the Redistributable version must be at least as new as the latest toolset used by any app component.

> You can define _DISABLE_CONSTEXPR_MUTEX_CONSTRUCTOR as an escape hatch.

OK, I can use the compile flag as "an escape hatch", but given that OpenCPN and plugins are compiled for Windows using Visual Studio 2022, shouldn't we be distributing the correct verions of the redistributables?

Hakansv, 2024-07-24:
Do you know if an update of msvcp140.dll would create trouble for e.g. Win7 users?

TwoCanPlugIn, 2024-07-25:
I doubt it. It has nothing to do with Windows Operating System versions.

Runtimes that are distributed with an app are normally installed into that app's executable directory, which is the first path the app will search in, for any required libraries.  

That directory is not added to the system's path environment variable so it is unlikely that it will interfere with the operation of other apps.

Eg. in the case of OpenCPN, the runtimes are installed into C:\Program Files (x86)\OpenCPN. No other apps would be looking into this directory for additional libraries/runtimes.

The problem at the moment is that OpenCPN 5.8x and newer, is compiled with Visual Studio 2022 but the packaging uses older versions of the Windows runtimes. Aside from my experience with std::mutex, there may be other, as yet undiscovered problems arising from the use of mismatched runtimes.

Remember we do this today with the wxWidgets runtimes (dll's). We package the wxWidgets 3.2..x libraries instead of older versions. The major difference however was that  wxWidgets was not ABI compatible between wxWidgets v3.1 and 3.2, so the mismatch was quite obvious !

leamas, 2024-07-25:
> I doubt it. It has nothing to do with Windows Operating System versions.

The possible exception would be if the msvc libs requires something not available in win 7, I guess. This should be easy to check if necessary.

leamas, 2026-02-11:
Windows 7 is no longer part of the game. 

@TwoCanPlugIn : Status? Is the beta dll versionm still outdated?

TwoCanPlugIn, 2026-02-12:
I don't think it's just me.

In core OpenCPN, https://github.com/OpenCPN/OpenCPN/blob/64d492805dd191aa0be5424271b5086a56f7085a/CMakeLists.txt#L431

Hakansv, 2026-02-13:
For info: This is what was installed locally by a package built by VS2026 yesterday.
If official installations use the same I haven't checked.

[image]