OpenCPN/OpenCPN #4064: Failure to build from master on docker image - googletest-src dubious owner

givethemheller, 2024-07-22:
**Describe the bug**
Basic error message bellow 

`[ 22%] Performing download step (git clone) for 'googletest-populate'
Cloning into 'googletest-src'...
fatal: detected dubious ownership in repository at '/src/build/_deps/googletest-src'
To add an exception for this directory, call:

	git config --global --add safe.directory /src/build/_deps/googletest-src
CMake Error at googletest-subbuild/googletest-populate-prefix/tmp/googletest-populate-gitclone.cmake:49 (message):
  Failed to checkout tag: 'release-1.12.1'

gmake[2]: *** [CMakeFiles/googletest-populate.dir/build.make:102: googletest-populate-prefix/src/googletest-populate-stamp/googletest-populate-download] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/googletest-populate.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2

CMake Error at /usr/share/cmake-3.25/Modules/FetchContent.cmake:1616 (message):
  Build step for googletest failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FetchContent.cmake:1756:EVAL:2 (__FetchContent_directPopulate)
  /usr/share/cmake-3.25/Modules/FetchContent.cmake:1756 (cmake_language)
  /usr/share/cmake-3.25/Modules/FetchContent.cmake:1970 (FetchContent_Populate)
  libs/gtest/CMakeLists.txt:41 (FetchContent_MakeAvailable)`

**To Reproduce**
1. Fresh clone of the repo on OSX M1
2. Follow instructions to pull the docker build image and enter it
3. Follow Linux build instructions. 
4. Cmake error occurs as noted above.

**Expected behavior**
well... ya know.

**Desktop (please complete the following information if applicable):**
OSX 14.2 2020M1

**Additional context**
Completely fresh installation.

Figured I'd post an issue before opening up a safe.directory exception. Please let me know if this is a known issue.

nohal, 2024-07-22:
This is a "problem" with your environment, nothing we can do about it.

bdbcat, 2024-07-22:
You can of course build without google test.  There is a cmake  variable, so do something like:

`$ cmake -DOCPN_BUILD_TEST=OFF ..`

or so.

leamas, 2024-07-24:
> Figured I'd post an issue before opening up a safe.directory exception. Please let me know if this is a known issue.

As @nohal  said, this is basically about your environment. If you have a proposal how to modify the instructions, please use the "Edit this page" button in the actual manual pahe to make a PR.  Besides this I think this should be closed.

@givethemheller:  Ok to close?

givethemheller, 2024-07-24:
> use the "Edit this page" button in the actual manual page to make a PR.

Close!

closed 2024-07-24 by leamas