OpenCPN/OpenCPN #4067: Route: send to peer : peer not found, crash

leamas, 2024-07-25:
**Describe the bug**
Trying to use the route context menu option "Send to peer" fails to look up the peer, crashes if peer is set manually

**To Reproduce**
Steps to reproduce the behavior involves two nodes. First a raspberry:
1. Configure the raspberry to work as a wireless AP handling DHCP requests from wifi clients, something like this
```
    # nmcli con add type wifi ifname wlan0 con-name Hotspot autoconnect yes ssid testnet1
    # nmcli con modify Hotspot 802-11-wireless.mode ap 802-11-wireless.band bg ipv4.method shared
    # nmcli con modify Hotspot wifi-sec.key-mgmt wpa-psk
    # nmcli con modify Hotspot wifi-sec.psk "veryhardpassword1234"
    # nmcli con up Hotspot

```
2. Verify that clients can connect to the wlan provided by the RPi
3. Start opencpn

On a PC
1. Start opencpn in safe mode using -s
3. Create a route
4. Right-click on route, select Send to Peer
5. The PC fails to locate the RPi on  address 10.42.0.1
6. If the correct address 10.42.0.1 is  entered manually in the drop down, the PC crashes.

**Expected behavior**
Route transferred from PC to RPi

**Desktop (please complete the following information if applicable):**
 - OS on RPi: Debian 12.
 - OS on PS: Fedora 38.

Both PC and RPi (actually an orange pi) runs Flatpak beta 3

**Additional context**
Network setup seems ok, I can access the RPi from the PC in all sorts of ways, and the RPi sends data to the PC in a network connection.
Stack trace:  [gdb.txt](https://github.com/user-attachments/files/16374133/gdb.txt)

The mDNS logging which is supposed to be disabled on normal runs is once again active, stdout: 
```
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
Read 0 records
Closed sockets
Opened 1 sockets for mDNS query
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
Read 0 records
Closed sockets

```

Hakansv, 2024-07-25:
I haven't tried send-to-peer for long so I made a test as well.
Current Git master:
None of my two RPis are never answering or seeing answers. I've once seen a answer from another Win. My Android (5.8.4) does answer.
In WS I can see the request all time from all devices.
Now I've a router that has not been reliable so I think others would test as well.

There's a for me strange query sent all time. See the addresses of first of the below catches. (192.168.1.67 is my Win)
```
6985	139.543328	2001:2042:7535:bd00:48e7:d33a:798b:d042	ff02::fb	MDNS	110	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question

6986	139.543706	192.168.1.67	224.0.0.251	MDNS	90	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question
```

bdbcat, 2024-07-25:
The first one is ipv6, I guess.

Hakansv, 2024-07-25:
Ohh, yes of course. (Me, would think before talk!) Thanks.

leamas, 2024-07-25:
I could live with the mdns failure. The really annoying thing is the crash when giving the correct address in the drop down edit window.

BTW: is there a way to specify the port as well? The obvious try 10.42.0.2:8443 is not accepted

leamas, 2024-07-25:
This is code I have worked with, I should be able to debug and fix. However, sailing over the weekend

Hakansv, 2024-07-25:
One Q
Using WS on my Win @ 192.168.1.67
If I see this to port 5353, wouldn't my Win immediately reply and view it in WS? Independent of any router?
`25509	506.660967	192.168.1.180	224.0.0.251	MDNS	90	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question`

Three times in a row now. No answer from my 192.168.1.67
Is this working for you?

TwoCanPlugIn, 2024-07-26:
Could it be a flatpak problem ?

Sending to a peer works fine for me in either direction between Windows & Ubuntu 22.04 (running under WSL)

leamas, 2024-07-26:
Possibly, but this has worked earlier in the 5.9 cycle IIRC

TwoCanPlugIn, 2024-07-26:
I wasn't going to say this before, but why on earth are you using a non-standard SRV RR?

The service name should be prepended with the underscore character and followed by the protocol

Eg. _opencpn._tcp or _opencpn-replication._tcp

Not surprising that the command 
`avahi-browse -v -k opencpn-object-control-service`
barfs with the following error message: 
`avahi_service_browser_new() failed: Invalid service type`

If you're going to use mDNS, don't you think it should be done properly ? Not suggesting that this is the cause of the originally reported problem, but it does make it impossible to diagnose with readily available tools.

Hakansv, 2024-07-27:
Though still a hijack of Alec's bug but: @TwoCanPlugIn:
Excuse my ignorance of these network details. If you think it's worth to try could you provide a possible code change and I can test on my Win/Rpi systems here? 
Whatever change in this opencpn-object code parts I think we will jeopardize any send-to-peer interactions with other OCPN versions. So it really need to be worth it.

TwoCanPlugIn, 2024-07-27:
Hi Hakan,

Other than perhaps placating 3rd party tools, clearly the library that O uses doesn't care about non standard SRV RR's. As OpenCPN plays in its own "sandbox", ie no other app is going to query for O's SRV RR, or connect to O using information it obtained from a mDNS query, it doesn't really matter.

Another "oddity" which again is inconsequential is that replication uses port 8443 yet the SRV RR is configured with port 8000. Inconsequential in that replication clearly uses hard coded port number rather than relying on mDNS to determine the port number. 

I doubt that any of this is causing the crashes that were originally reported. As to why it's not working across your network, either your router isn't allowing multicast UDP traffic (maybe unlikely) or a firewall or flatpak problem.

Hakansv, 2024-07-27:
Understood. It works sometimes but randomly. Once detected it always work to transfer, so Alec's main issue here to be able to enter an IP-address may be a solution?

bdbcat, 2024-07-28:
I pushed a commit that may help, especially when there are multiple machines on the net with the same "name", e.g.
"raspberrypi".
Tested in my current network setup of two rPI 5(bookworm), linux AMD64, and Windows 10.  All but the AMD64 on Wifi.  In my tests, everyone can see everyone, generally always.  Occasionally the Windows machine does not respond.

On linux based system, you may watch the console printf results to get some further understanding of the process.

```
Opened 11 sockets for mDNS query
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
192.168.37.97:5353 : answer opencpn-object-control-service. PTR debian.opencpn-object-control-service. rclass 0x1 ttl 10 length 9
192.168.37.97:5353 : answer opencpn-object-control-service. PTR debian.opencpn-object-control-service. rclass 0x1 ttl 10 length 9
192.168.37.144:5353 : answer opencpn-object-control-service. PTR raspberrypi.opencpn-object-control-service. rclass 0x1 ttl 10 length 14
Read 40 records
Closed sockets
```

Hakansv, 2024-07-28:
Yes, although no name clash I'd say it's better now. I have to "Scan again". Sometimes several times before detection but eventually it works. 

Some WS catches if of interest. Both delayed and "strange" answers may lead to no detection?
One may wonder who is causing the delay that sometimes occurs; OCPN, Rpi or the router?

```
Send to from HAS-WIN 192.168.1.67
rpi5 192.168.1.87 answer in due time but message: cashe flush...  No detection
Correct reply after time out. No detection.
2461	27.390755	192.168.1.67	224.0.0.251	MDNS	90	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question
2601	28.768269	192.168.1.87	224.0.0.251	MDNS	133	Standard query response 0x0000 PTR rpi5.opencpn-object-control-service SRV, cache flush 0 0 8000 rpi5.local
4920	57.541858	192.168.1.87	192.168.1.67	MDNS	139	Standard query response 0x0000 PTR opencpn-object-control-service, "QU" question PTR rpi5.opencpn-object-control-service SRV 0 0 8000 rpi5.local

Scan again: Late answer but in due time (9 sec) Detection!
7043	83.076367	192.168.1.67	224.0.0.251	MDNS	90	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question
7482	92.155593	192.168.1.87	192.168.1.67	MDNS	139	Standard query response 0x0000 PTR opencpn-object-control-service, "QU" question PTR rpi5.opencpn-object-control-service SRV 0 0 8000 rpi5.local

Send to from rpi5 192.168.1.87
Answer from rpi5 but then "Port unreachable"  Not detected.
36404	426.683788	192.168.1.87	224.0.0.251	MDNS	90	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question
36405	426.684152	192.168.1.67	192.168.1.87	MDNS	145	Standard query response 0x0000 PTR opencpn-object-control-service, "QU" question PTR HAS-WIN.opencpn-object-control-service SRV 0 0 8000 HAS-WIN.local
36406	426.689324	192.168.1.87	192.168.1.67	ICMP	173	Destination unreachable (Port unreachable)

Scan again. Now detected.
41368	483.104624	192.168.1.87	224.0.0.251	MDNS	90	Standard query 0x0000 PTR opencpn-object-control-service, "QU" question
41369	483.105045	192.168.1.67	192.168.1.87	MDNS	145	Standard query response 0x0000 PTR opencpn-object-control-service, "QU" question PTR HAS-WIN.opencpn-object-control-service SRV 0 0 8000 HAS-WIN.local

```

TwoCanPlugIn, 2024-07-28:
With respect to the duplicate name issue, let's call it the "Pi problem", I didn't realise that OpenCPN doesn't use mDNS in the way it was fully intended. 

Perhaps what I said previously "If you're going to use mDNS, don't you think it should be done properly ?" goes further than my observation regarding the use of non standard service names.

At startup OpenCPN should send a MDNS query with a query type of ANY to detect there if any other devices offering the same service  "opencpn-object-control-service" with the same host name and it should also send an unsolicited mDNS answer.  I'm not going to re-read the RFC but from what I remember from aeons ago there is a section describing startup, probing and  announcements.

At present from what I can tell, OpenCPN only issues a query for the PTR RR's when you are actually instantiating the "Send to Peer" dialog. There seems to be nothing that detects nor alerts the user to duplicate host names.

As networks & Pi's are becoming more prevalent on boat networks and the likelihood of duplicate host names becomes higher, perhaps OpenCPN should follow the standard.

However this doesn't address nor solve the network latency/reliability problem that Hakan is observing.

bdbcat, 2024-07-28:
As a clue to tracking this down:
Can we say fairly that the latency/reliability issues always involve Windows, in one way or another?

bdbcat, 2024-07-28:
I have tested the "direct ip entry into the dropbox" use case.  Simply type an ip ("e.g. 192.168.37.70") into dropdown.
For not-existing hosts, no crash, but a useful error message.
For known existing host, normal transfer occurs.

Anyone else duplicate?

Hakansv, 2024-07-28:
> As a clue to tracking this down: Can we say fairly that the latency/reliability issues always involve Windows, in one way or another?

No.  It's the same between two Rpi's. Just now my Rpi400 detected a WIN but not the Rpi5.
I could though enter the IP address for the Rpi5 and the transfer occurred nicely.

Hakansv, 2024-07-28:
> I have tested the "direct ip entry into the dropbox" use case. Simply type an ip ("e.g. 192.168.37.70") into dropdown. For not-existing hosts, no crash, but a useful error message. For known existing host, normal transfer occurs.
> 
> Anyone else duplicate?

Yes, I see the same. I can enter IP address on both win and Rpi and if it exists the transfer works. A new Pin though even if the same server was used before. Probably because the lack of host name so then it's seen as a new server.

Entering a non existing IP on Win gives the error below.
Interesting: If I answer OK on the Curl message it's return again and again whatever I do on the "Send to..." message box. 
If I answer Cancel on the Curl message both message boxes are closed.
So I'd say to enter an IP address manually is working. Maybe tweak the Curl message OK-logic? (Why OK-Cancel and not only OK?)
[image]

bdbcat, 2024-07-28:
@Hakansv 
When you try to send-to-peer from one rPI to another:
1.  Do both rPI have the latest github code?
2. In the CLI printf on the sender, do you see the "missing" rPI listed in the mDNS query replies?

Hakansv, 2024-07-28:
@bdbcat 
1. Yes all my devices have OCPN built from current master code.
2. No. Below copy is the result of three scans from the RPi400 named "Openplotter". One WIN and one RPi5 are alive.

```
Opened 2 sockets for mDNS query
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
192.168.1.154:5353 : answer opencpn-object-control-service. PTR openplotter.opencpn-object-control-service. rclass 0x1 ttl 10 length 14
Read 4 records
Closed sockets
Opened 2 sockets for mDNS query
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
192.168.1.154:5353 : answer opencpn-object-control-service. PTR openplotter.opencpn-object-control-service. rclass 0x1 ttl 10 length 14
Read 4 records
Closed sockets
Opened 2 sockets for mDNS query
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
192.168.1.154:5353 : answer opencpn-object-control-service. PTR openplotter.opencpn-object-control-service. rclass 0x1 ttl 10 length 14
Read 4 records
Closed sockets
```
Strange?
After having pinged Rpi5 from the WIN PC I got this on next "Scan again" where both WIN and Rpi5 was detected.:

```
Opened 2 sockets for mDNS query
Sending mDNS query : opencpn-object-control-service PTR
Reading mDNS query replies
192.168.1.154:5353 : answer opencpn-object-control-service. PTR openplotter.opencpn-object-control-service. rclass 0x1 ttl 10 length 14
192.168.1.87:5353 : answer opencpn-object-control-service. PTR rpi5.opencpn-object-control-service. rclass 0x1 ttl 10 length 7
192.168.1.67:5353 : answer opencpn-object-control-service. PTR HAS-WIN.opencpn-object-control-service. rclass 0x1 ttl 10 length 10
Read 12 records
Closed sockets
```

leamas, 2024-07-29:
Back from sea.  Some findings at home using a Fedora development machine + the same OrPi  with wired connections:

1. The network setup seems to be part of the problem. When I move the OrPi from the wifi  it supplies to a wired network   at home the client finds the server.
2. The flatpak client still crashes when trying to transfer after finding the server ("server" and "client" are actually not the correct terms here, but I'm using them anyway)
3. Flatpak is another part of the problem -- a native Fedora build finds the peer "server" and transfers the route successfully. This is against the same remote peer running under Flatpak on the OrPi. 

Too tired to be clever at this point. Will try tomorrow.

leamas, 2024-07-29:
more pieces: the crash occurs on exit from `static int ApiGet()` in _peer_client.cpp_  at line 164. Also at this  point the `*** stack smashing detected ***: terminated`  occurs.

Although the explanation once found tends to be simple, this one actually looks hard. Sigh.

leamas, 2024-07-30:
Pushed 25802740e which solves the crash problem for me.

Hakansv, 2024-07-30:
No difference here on Win but I didn't get the crash earlier either.
The Curl error message behaving like an infinitive loop when answering OK is troubling though.  Described above.

leamas, 2024-07-30:
Status after rebuilding client from master @ 25802740e and testing with PC on wifi network provided by the OrPi:
  - The server is not found by mDNS.
  - After entering the address manually the transfer completes

I see the following problems:
1. To repeat, the server is not found. Might be about Flatpak, but "should" not since opencpn has full access to the network stack. Let's focus on Håkan's testcase without Flatpak for now.
2. There is  seemingly no way to manually specify the port if it's  something else than 8443 (?)
3. The typical use case, at least for me, is to use the same "server" repeatedly. In this context it's annoying to wait for the mDNS scan before each transfer. We should IMHO cache the last scan and only re-scan after user request.
4. While handy here and now, the debug output to stdout is a bug. There should be no such logging unless `--loglevel debug` is  set.

Hakansv, 2024-07-30:
1. My mDNS discovery problem can be left out. It's probably due the router. Also ping can sometimes report unreachable for a while. We have seen four unreachable and then eventually replies for the same ping.
2. Maybe but the IP address wo port works .
3. Here the last mDNS discovery is cached and shown next time Send-to-peer is used. But not the manually entered IP.
4. Maybe. But starting O from cmd is seldom used I suppose.

leamas, 2024-07-30:
1. If we leave your problem out, my is still valid. But I can live with it now when manually entering the address works.
2. Seems that we agree on this
3. It's cached for 5 minutes, not refreshed when used during this timespan, and manual entries are not cached. It's just not as it should IMO.
4. We had problems with 5.8.4 when users reported bugs and was trying to correlate it with debug output. It actually adds confusion, and wreaks havoc on the CLI interface we have. _But starting O from cmd is seldom used I suppose_. Definitely true on Windows, not so much on Linux.

We need @bdbcat 's input here

leamas, 2024-07-30:
1. This might be about my multicast setup. I need to review this, until done the ball is on my side.

bdbcat, 2024-07-30:
1.  Stable detection of peers here.  Occasionally missing Windows reply. I think there is some intermittent trouble with Hakan's router on multicast.  Conclusion: leave it, and monitor real user results.
2.  I don't see the use case to need a different port.  If the peer is OCPN, then 8443 is the port.  Or?  On the other hand, it should be trivial to check a manual entry for port following ":".  So why not do it?
3. The caching happens down inside the mDNS receive thread.  It caches received replies.  Thus, it should cache manual entries as is, if there is a reply.  This needs checking.  5 minutes too short?  Suggestions for longer?
4. debug logging.  I agree with leamas.

@leamas:
Which parts of this will you own, and which should I investigate?

leamas, 2024-07-30:
4. Pushed 919dc64f3. 

As I see it, the reasonable assumption is that my name resolving problem is about the network setup.

1, 3. Almost blocking: that I need to wait for a non-working scan and write the ip address for each and every transfer. I have some code proposal for this, but need a day  or to to test.  Personally, I can add this on top of the final release, so not really blocking unless we think there are others with similar problems.

2. is definitely for 5.12, let's leave it for now.

bdbcat, 2024-07-31:
OK, waiting Beta-4 for a day or so.

leamas, 2024-07-31:
I'll take a look into 2. later, it's not important now.

1. seems to be about my network setup. This might need some work in the manual, but otherwise there is so far no sign of bugs in the code w r t this. 

3. and 4.  are fixed by recent commits.

Closing. Please request for reopening if  need be.

closed 2024-07-31 by leamas