OpenCPN/OpenCPN #4070: Streamline  debian/ubuntu packaging

leamas, 2024-07-30:
**Is your feature request related to a problem?**
In 5.10 we are moving to create release Debian packages using the the same mechanisms as have been used in the official packaging. 

This means that we have duplicated debian packaging setups. The original in master branch is about the CMake package, including the debian/ directory in the master  branch. The new packaging lives in packaging  branches like ubuntu/focal which basically uses nothing beyond `make install` in the cmake  setup.

The duplicated setups means more code to maintain, messes up the new Debian packaging and introduces risks that the two packaging chains produces different results.

**Describe the solution you'd like**
We should move the CI builds to use the new system, remove the debian/ dir in master and clean up CMakeLists.txt.

**Describe alternatives you've considered**
None.

bdbcat, 2024-07-30:
AFAICT, we are not using the CI builds targeting ubuntu for anything other than "does it build" validation.
While valuable input, this does consume CCI resources.  We do not yet pay for CCI builds, but I see a time when we will be asked to get a plan, pay for it, and monitor build cycles more closely.
thoughts?

leamas, 2024-08-14:
hm...

Looking at the new builds, we get two things:
1. Packages which can be downloaded for testing, the usual "nightly builds" concept (we could do a better job with making them visible).
2. Contrary to the "old" builds, the new one performs some quite useful tests.

OTOH, you arguments are of course valid. How long is CircleCI's current business model sustainable?

I'll go into this with the idea to have a simple switch which can be used to enable/disable package generation. To be useful it needs to be specific for each branch.

closed 2024-08-15 by leamas

leamas, 2024-08-15:
As pushed the bookworm branch builds a debian package, other linux branches does not build any package at all.  Enabling/disabling the package build for a build is a oneliner in _.circleci/config.yml_

commit 3e1b777c (2024-08-15): First stab on package generation (#4070)

The ci/generic-build-debian.sh now is capable of either making a plain
build or to create a debian package using the setup from packaging
branches

The used path depends on circleci/config.yml. If there is a
PACKAGE_BRANCH environment variable defined for the build a Debian
package will be built

As of this commit, the bookworm builds does build a Debian
package, all others just make a plain build.

Closes: #4070