OpenCPN/OpenCPN #4074: Track recording start timestamp in UTC only

bc38, 2024-08-03:
Local time adjustment required, creating confusion
Route Mark Manager

sebastien-rosset, 2024-08-09:
Does it make sense to address this as part of #4052?

bc38, 2024-08-10:
> Does it make sense to address this as part of #4052? The idea would be to display date/time in a consistent manner across all UI widgets, and formatting date/time according to the global display settings (UTC or local time), using the exact same format everywhere.

yes of course!
Global parameters are required by СPN for integrity

sebastien-rosset, 2024-08-11:
I've made a change such that the track time honors the global date/time parameters. For example:

[image]

leamas, 2025-05-02:
@bc38: does this satisfy your needs i. e., can we close this bug?

bc38, 2025-05-02:
Yes, the time is local BUT it has a 12 hour format!
Time to introduce the 12/24 option for time

[image]

bc38, 2025-05-02:
The track tooltip displays the time in UTC!

[image]

bc38, 2025-06-08:
Yes, in the track properties the problem with the timestamp in local time is solved!
BUT in the tooltip the time is in UTC
In Global settings Local time

[image]