OpenCPN/OpenCPN #4075: Add visual feedback for inactive connections

leamas, 2024-08-03:
**Is your feature request related to a problem?**
Although drivers in many cases detects situations when a connection is missing there is no visual feedback in the Options | Connections tab. This is includes for example missing USB devices and missing network  hosts.

**Describe the solution you'd like**
Some kind of visual feedback in the actual connection setup. Probably based on on more EventVars which are fed from drivers and the GUI listens to.

bc38, 2024-08-03:
Would you add a visual indication of Red, Yellow? Green for connection

leamas, 2024-08-03:
Something like that, yes. Perhaps grey (disabled), red, green

TwoCanPlugIn, 2024-08-04:
> Some kind of visual feedback in the actual connection setup

If you mean during the creation of connections, at least for USB devices, given you know the VID/PID, you can detect the presence of the device via the Registry on Windows, IOReg on MacOSX and dbus and/or uevent files on Linux.

In fact I was kind of surprised O didn't adopt this initially when implementing support for NMEA 2000 USB devices. O could have automagically™ configured the com port/tty device, baud rate etc without the user having to configure anything or determine which port is used. This was the approach I used for TwoCan and an unreleased plugin for the Actisense NGT-1

For NMEA 2000 you could go further and maintain a list of sources (GPS, depth, speed, wind, rudder, AIS etc), their CAN bus address and device details (unique number, manufacturer, model etc)

I'm not sure how the setup wizard scans for network connections (for SignalK presumably mDNS), but once a network connection has been configured, at least for a UDP connection where O may listen on all interfaces (eg. 0.0.0.0 , aka INADDR ANY), O would need to record the remote end point and the local interface over which the data is received in order to later detect the presence of that host. If not, how will O know that the remote host is unavailable, other than not receiving data?

leamas, 2024-08-04:
> If you mean during the creation of connections, at least for USB devices, given you know the VID/PID, you can detect the presence of the device via the Registry on Windows, IOReg on MacOSX and dbus and/or uevent files on Linux.

My initial thinking has not been so much about creation of devices, this might need more attention.

It has more been about after the device is created. Things like USB devices and network hosts which existed when the device was created might go away later.  As it is now, you need to open the nmea debug window to figure this out, and also decipher that information. A simple red/green light in the plugin panel might give a simpler overview about connection state.

The next step would be the traditional device info about bytes transferred, perhaps per bytes per second. However, this would clutter the interface so I'm not sure it's a good idea.

TwoCanPlugIn, 2024-08-04:
Even after the OpenCPN USB connection has been created, you can always use the Windows Registry, MacOSX IOReg and Linux ttyusb*/uevent file to determine whether the USB device is still inserted.

The next step for O would be to detect insertion/removal events and respond accordingly. Perhaps a 5.14 feature?

leamas, 2024-08-04:
There is already linux code for USB devices insert/remove. This could be expanded also to to Windows events (not sure about events in MacOS).

If that code actually works is another question, here might be some regressions. Plan to check later today.

leamas, 2024-08-04:
> Or a message box in the middle of the screen at startup is sure to get the user's attention!

Already in place for missing USB devices.

leamas, 2024-08-04:
> If that code actually works is another question, here might be some regressions. Plan to check later today.

It does. When creating a new connection the list of available serial devices is updated as soon as a device is inserted or removed. This uses dbus signals.  For someone who is familiar with windows (not me) it's probably easy to add a windows events backend.-

So, we already have a beginning.

leamas, 2024-08-04:
> Even after the OpenCPN USB connection has been created, you can always use the Windows Registry, MacOSX IOReg and Linux ttyusb*/uevent file to determine whether the USB device is still inserted.

Yes. But the trick is to know when to check to avoid polling these database, we don't want to do that.

The current Linux logic for list of available serial devices is to re-evaluate what can be found as soon a  dbus  device added/removed signal is received. This is IMHO the correct logic, it creates a responsive UI without polling.

leamas, 2024-08-04:
Filed #4078 to add context

TwoCanPlugIn, 2024-08-05:
Seems like you have Linux sorted. When initially creating a connection on Windows you can use the Registry, on MacOSX IOReg to automagically™ detect and configure the USB device.

Similarly upon OpenCPN startup, you can use the same technique to verify the device is still present.

If the intent is for OpenCPN to dynamically handle USB insertion/removal events, which is an admirable goal, then yes I fully agree with you that polling is not appropriate,. 

In which case during runtime, at least on Windows, you have a myriad of methods that can be used to detect USB insertion/removal events, low level such as RegisterDeviceNotification through to higher level using WMI using something like a ManagementEventWatcher. I'm not familiar enough with OSX to know what the equivalent APIS's are.

However I guess that libusb might provide the appropriate cross platform support, But then again maybe not, just did a little test and at least on Windows, libusb does not support hotplug. No idea what it does on OSX, presumably as OSX has its antecedents with BSD, perhaps libusb does support hotplug.

Might be time to bite the bullet and use platform dependent code?

leamas, 2024-08-05:
> Seems like you have Linux sorted. When initially creating a connection  [...]

Partly, for creating a device. But it still fails miserably if the device does not exist at a later point. That is, even if there is a warning message about the missing device, you can still see the nmea log window writing data to this missing device. This is worth filing  a separate issue.

>  When initially creating a connection on Windows you can use the Registry, on MacOSX IOReg to automagically™ 

This is the `EnumerateSerialPorts()`  method in _model/include/model/ser_ports.h_ + implementation

> If the intent is for OpenCPN to dynamically handle USB insertion/removal events, which is an admirable goal

It is. Furthermore we should handle the situations where the driver discovers that the device or peer to communicate with is gone.

> Might be time to bite the bullet and use platform dependent code?

Since we are using dbus on linux we are already on this path. The trick is to have an overall architecture which minimizes and creates simple, clean interfaces to this code. Hopefully this will stabilize in the next cycle,

It should already be possible to create a windows sibling to the `LinuxUsbWatchDaemon`. This listens for dbus events and notifies some `EventVar`s. Upper layers listens to these vars and acts appropriately.

> However I guess that libusb might provide the appropriate cross platform support

Thanks, good input! If we use libusb instead of dbus to listen to USB devices insert/eject we will cover macos as well. Will try to in next cycle.

>  But then again maybe not, just did a little test and at least on Windows

https://github.com/libusb/hidapi/issues/238. 

Which boils down to that using a modified `LinuxUsbWatchDaemon` + a Windows implementation doing the same thing should cover all platforms besides Android: https://github.com/libusb/libusb/pull/1168#issuecomment-1222133151

TwoCanPlugIn, 2024-08-12:
Had a little bit of time waiting for some wind, Found some old Windows code which I've modified to use wxWidgets. It dates back about 15-20 years ago when I was playing with my old Garmin GPSMap76s and a Prolific USB Serial Port adapter.

Not up to your coding standards, nor obfuscated as to be unreadable but might be useful as a POC or at least give you some indication as to how to detect Windows PnP events.

BTW, as far as I know, three ways to get USB events from Windows.
1. RegisterDeviceNotification which probably goes as far back as Windows NT 3.1 or perhaps when PnP was fully supported in 3.51 (but my memory is too far gone). My POC is based on this set of API's
2. Configuration Manager (I think it is really called Device and Driver Installation)
3. Windows Management Instrumentation (WMI). This is Microsoft's implementation of Web Based Enterprise Management (WBEM) an open standard supported by HP OpenView, IBM Tivoli, BMC Patrol, CA Unicentre. (Well it did at the time, I have no idea whether these products still exist!). 

Option 1. is available on all versions of Windows. Only downside is you have to override the WindowsProc.
Option 2. uses a callback function instead of a Windows proc, but is only supported on Windows 8 and beyond.
Option 3. is, if my memory serves correctly, supported on Windows 2000 and beyond. Requires COM to be initialised, but as I discovered when writing the Windows Sensor plugin, wxWidgets initializes COM. Plenty of examples on the Internet, I also have some sample code. Look for something like WBEM Query Language (WQL, a subset of SQL); "SELECT * FROM __InstanceCreationEvent WHERE TargetInstance ISA Win32_PnPEntity"

With all three, you essentially use the same code to retrieve friendly names, port names from the registry.

I've included the simple POC based on Option 1.
To use a similar methodology with Option 2. just requires these three changes:
`// Register for notifications`
`CM_NOTIFY_FILTER notificationFilter = { 0 };`
`notificationFilter.cbSize = sizeof(notificationFilter);`
`notificationFilter.FilterType = CM_NOTIFY_FILTER_TYPE_DEVICEINTERFACE;`
`notificationFilter.u.DeviceInterface.ClassGuid = GUID_DEVINTERFACE_COMPORT;`

`CONFIGRET returnCode = CM_Register_Notification(&notificationFilter, (PVOID)NULL, (PCM_NOTIFY_CALLBACK)&MyFrame::NotificationCallback, &notificationHandle);`

`// Notification Handler`
`DWORD MyFrame::NotificationCallback(HCMNOTIFICATION notifyHandle, PVOID contextHandle, CM_NOTIFY_ACTION ``notifyAction,`
 `   PCM_NOTIFY_EVENT_DATA eventData, DWORD dataSize) {`
`
 `   wxString deviceName = eventData->u.DeviceInterface.SymbolicLink;`
`
`   switch (notifyAction) {`
`    case CM_NOTIFY_ACTION_DEVICEINTERFACEARRIVAL:`
`	// Need to replace '#' with '\' as the API's use different delimiters`
 `	deviceName.Replace('#', '\\');`
`	// Get the list of all devices, so we can enumerate them to find the matching device`
 `	HDEVINFO devInfoSet = SetupDiGetClassDevs(&GUID_DEVINTERFACE_COMPORT, NULL, NULL, 	DIGCF_PRESENT | ``DIGCF_DEVICEINTERFACE);`
 `       .....`
 `   case CM_NOTIFY_ACTION_DEVICEINTERFACEREMOVAL:`
`	.....`
`// Deregister`
`CONFIGRET returnCode = CM_Unregister_Notification(notificationHandle);`

Not sure if this is of any help to you.

[PlugnPray.zip](https://github.com/user-attachments/files/16580674/PlugnPray.zip)

Oh, BTW, was going to suggest that perhaps OpenCPN should maintain a list of all of the VID/PID's for the supported USB devices (Actisense, YachtDevices, perhaps even USB Serial Port adapters, USB GPS receivers etc).

leamas, 2024-08-14:
@TwoCanPlugIn :  
It's actually simpler than that. The only thing we need to propagate is that "something happened" i. e., an USB device was plugged in or out. The actual device does not matter,  upper layers will just rescan the list,  this is existing working code. 

I'm thinking about making a skeleton for a windows service similar to the existing linux one. Perhaps you could the fill in the hairy details? Please stay tuned.

leamas, 2024-08-14:
@TwoCanPlugIn 

hm... My memory is good, albeit somewhat short. It turns out that we already have such a daemon in place.
Please have a look at _model/include/model/win_usb_watch.h_ and _model/src/win_usb_watch.cpp_

leamas, 2025-01-16:
Fixed in #4307. Closing

closed 2025-01-16 by leamas