OpenCPN/OpenCPN #4076: Activation of next waypoint failing

douwefokkema, 2024-08-03:
Runnning a certain route (attached) activation of next waypoint fails when advancing from waypoint 022 to 021. 
On reaching 022 an advance to 021 is done. Reaching the arrival circle of 021 there is no advance to 020. I did some additional logging, ActivateNextPoint() is called OK from DoAdvance(). In ActivateNextPoint index active point=110, and is not advanced to 111. However the index of the next wp (020) is 112, it seems there is no wp with index 111, mystery??.
This issue is fully reproducible with Shipdriver plugin (and when sailing). The issue is also present in older versions of O.
An identical issue occurs at waypoint wp-WG-L in the same route.
[Lauwersoog-Vlieland.gpx.txt](https://github.com/user-attachments/files/16481091/Lauwersoog-Vlieland.gpx.txt)

douwefokkema, 2024-08-04:
Forget about the route attached above. Here is a simple way to reproduce. Make a 4 waypoint route, waypoints 001, 002, 003, 004. Drag 002 on top of 003. O asks: Replace this RoutePoint by the nearby Waypoint? Answer Yes. You now have a 3 point route, 001, 003, 004. But when 003 is active the context menu command Activate Next Waypoint does not work.
This is a 3 waypoint route where waypoint 3 occurs twice in the same location. Remedy: Context menu, on 003, Remove from route. 003 is not removed from the route, only the second occurrence is removed. Route behaves fine again.
Is this still a problem? Yes. People have many large routes. And you don't know if this error is in somewhere. If it is, and you follow the route with AutoTrackRaymarine plugin, the boat will make a sudden U turn at such a double waypoint. I have had several reports of sudden U turns from users of the plugin and I could not locate the cause in the plugin.

douwefokkema, 2024-08-04:
The essence of the issue is that for such a "twin waypoint" the index of the active point is the same as the index of the next point. I could not make a formal PR, bad internet connections on the way. But I made a new ActivateNextPoint(Route *pr, bool skipped) that works fine and handles twin waypoint properly. Of course the creation of such twins should be prevented, but as these waypoints are around, better handle them properly.
[ActivateNextPoint.cpp.txt](https://github.com/user-attachments/files/16487159/ActivateNextPoint.cpp.txt)

douwefokkema, 2024-08-04:
Apart from the above, when dragging a wp of a route onto the next twice (create triple wp), O will crash on Remove waypoint from route.

bdbcat, 2024-08-04:
"The essence of the issue is that for such a "twin waypoint" the index of the active point is the same as the index of the next point".
Yes, I agree, that is the key point.
Your patch to ActivateNextPoint.cpp looks good.  I'll apply that now.
Moving forward, we want to build a "route scrubber", that detects and removes duplicate points.  Easy enough to build.  But where execute the scrubber?  I see several places:
1.  After any route edit activity.
2. On initial route load for navobj.xml.  (fixes any orphans in user's system)
3. On activate of a route, just to be sure...
Others?

leamas, 2024-08-04:
After receiving a "send to peer" route?

douwefokkema, 2024-08-04:
After any route edit activity.
You never know when an edit is ready, this means continuous checking. System may become slow.

    On initial route load for navobj.xml. (fixes any orphans in user's system)
But if edited afterwards inconsistencies may be introduced again. Unless sources of inconsistencies are largely removed (see below)

    On activate of a route, just to be sure...
Seems best to me. One may edit active route, but nothing is completely safe.

   After receiving a "send to peer" route
I myself never use this function.

An important source of errors seems the (accidental) dragging of a wpt onto a wpt of the same route. It should not be difficult to inhibit replacement in this case (no pop up question) and prevent these errors.

bdbcat, 2024-08-04:
"You never know when an edit is ready, this means continuous checking. System may become slow."
In this case, exactly one route is being scrubbed.Unlikely to be so slow as to affect GUI.

bdbcat, 2024-08-04:
Anyway, other than the "activate next" patch, I propose to defer, and Milestone this issue for O5.12.

douwefokkema, 2024-08-05:
"Unlikely to be so slow as to affect GUI."
Then scrubbing after edit seems the best solution.

"other than the "activate next" patch, I propose to defer, and Milestone this issue for O5.12"
Fine!

douwefokkema, 2024-08-17:
I have submitted PR  Make ActivateNextpoint pass multiple identical points #4100 as a solution for this issue. This PR does not repair routes by removing "twin" waypoints.

leamas, 2025-05-03:
@douwefokkema 

I cannot reproduce this any more, assuming that it is fixed by #4100 (?)  Can we close?

douwefokkema, 2025-05-03:
This issue was solved in commit 8951ab9d7aa0, Make ActivateNextpoint pass multiple identical points, 2024-08-17.

closed 2025-05-03 by douwefokkema