OpenCPN/OpenCPN #4082: Invalid sunrise/sunset times are displayed in the dashboard widgets

sebastien-rosset, 2024-08-07:
**Describe the bug**
The displayed sunrise/sunset calculation is incorrect, probably after a GPS watchdog timeout.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Dashboard preferences.
2. Add widgets: Sunrise/Sunset UTC, Sunrise/Sunset Local, Location, time.
3. Notice sometimes the displayed sunrise/sunset times are wrong. For example, around San Francisco August 6th, 2024, the sunset is displayed as 18:01:00 Local time, which is wrong. The actual sunset is around 8:07PM.

**Expected behavior**
The sunrise/sunset times should be displayed correctly. If the program is unable to calculate the time, it should not display an invalid date.

**Screenshots**

[image]

I've added some debugging to troubleshoot the problem. It looks like after a GPS watchdog timeout, the `m_dt` variable is set to January 1st 1970, and that causes sunrise/sunset calculation errors.

```bash
19:00:56.450 MESSAGE clock.cpp:542 CurrentDate: 2024-08-07T02:00:55, DayOfYear: 220, Sunrise: 2024-08-06T13:18:00 Sunset: 2024-08-06T03:09:00
19:00:56.450 MESSAGE clock.cpp:542 CurrentDate: 2024-08-07T02:00:55, DayOfYear: 220, Sunrise: 2024-08-06T13:18:00 Sunset: 2024-08-06T03:09:00
19:01:05.661 MESSAGE comm_bridge.cpp:188    ***GPS Watchdog timeout at Lat:37.4885   Lon: -122.181
19:01:06.449 MESSAGE clock.cpp:542 CurrentDate: 1970-01-01T00:00:00, DayOfYear: 1, Sunrise: 2024-08-06T15:23:00 Sunset: 2024-08-06T01:01:00
19:01:06.450 MESSAGE clock.cpp:542 CurrentDate: 1970-01-01T00:00:00, DayOfYear: 1, Sunrise: 2024-08-06T15:23:00 Sunset: 2024-08-06T01:01:00
```

**Desktop (please complete the following information if applicable):**
 - OS: macOS
 - Version 5.9

sebastien-rosset, 2024-08-07:
In addition, the "Local GNSS Clock" is wrong after the GPS watchdog timemout. In the attached screenshot, the time is stuck at "18:00:00 LCL". It should display something like "---" or string showing GNSS date is not available.

Arne2024, 2024-10-16:
the same issue in Dashboard shows  wrong data for Local ansd UTC  sunrise / sunset 
Version 5.10.00 
OS MacOS

closed 2025-01-06 by sebastien-rosset