OpenCPN/OpenCPN #4084: fedora  fc40   make  failure

oystercatcher, 2024-08-07:
**Describe the bug**

new install  fedora  fc40   testing build opencpn from git  clone
cmake  ok
make   fails

**To Reproduce**
Steps to reproduce the behavior:
1.  git clone    opencpn
2.  cmake
3.  make
4. See error attached

Expected behavior**
 make step to complete  no errors

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

fedora   fc40

**Additional context**
[opencpn.fc40.make.txt](https://github.com/user-attachments/files/16532804/opencpn.fc40.make.txt)

leamas, 2024-08-07:
We basically don't support native Fedora builds, and there are no plugins available. The way to run and build OpenCPN on Fedora  is Flatpak.

That said, if you really want to make a native build you can work around this issue ny installing the rapidjson-devel package.

oystercatcher, 2024-08-07:
thanks  for  the  work around.   probably  had the same error in  fc41

patrick
oxnard, calif

closed 2024-08-07 by oystercatcher