OpenCPN/OpenCPN #4086: Modifying track name does not get saved after  OCPN crash

antipole2, 2024-08-09:
**Describe the bug**
OCPN v5.9+ saves changes to objects so that, after an untidy end, they are preserved when OCPN restarts.
However, a change to a track name is not saved.

**To Reproduce**
Steps to reproduce the behavior:
1. Modify a track name
2. Force quit OCPN and restart
3. The track name change has been lost.

**Expected behaviour**
The changed name should be preserved

 - OCPN v5.9.4

leamas, 2025-05-02:
This is addressed in the same way as #4345. Major efforts are under way to move the navobjects database to Sqlite which will make the risks for losing data when OpenCPN  crashes much, much lower.

leamas, 2025-05-03:
Supposedly fixed by 84f2b70bf.  Marking as done pending verification

leamas, 2025-07-18:
Closing this for now, the new database backend functions in such a different way that if the condition persists we should file a new bug

closed 2025-07-18 by leamas