OpenCPN/OpenCPN #4087: Problem with recording the active track

bc38, 2024-08-10:
the problem with recording the active track persists
stops marking points and drags one line behind it. I tried to enable speed filtering, lowered the time and distance parameters in the configuration file, changed the GNSS receiver. After rebooting, the dragging line is not saved, it seems that the track does not even write to the file.
[image]

bc38, 2024-08-12:
here is an example where the problem of no track recording appears
NMEA log playback is started, then playback stops and another NMEA log is started. The position jumps a long distance, there is movement but no track recording. This happens to me during long parking and when the signal is bad and the boat position jumps a long distance.
[Video1](https://www.dropbox.com/scl/fi/8fex62c7djg1z4s3q51qj/2024-08-12-184038.mp4?rlkey=dkp08yue61nfqzufrlcjlox1d&st=g2e9ana5&dl=0)
[Video2](https://www.dropbox.com/scl/fi/kk5hsniv1668zvgwf5vid/2024-08-12-184253.mp4?rlkey=tbnobipscz2cz2nkyrdmy18na&st=g2um4c35&dl=0)

bdbcat, 2024-08-19:
I think I understand the problem.
Please verify with this temporary fix.  Add to config file:

`FilterTrackDropLargeJump=5000`

Hakansv, 2024-08-19:
Maybe we should change this function to calculate unreasonable speed instead of distance?

Discussion:
The "FilterTrackDropLargeJump" value and usage is a way to overcome a bad sensor reading caused by a failed or disturbed GNSS receiver. It was introduced to try to filter out irrelevant positions from an ongoing track. Some have seen "jumps" in there tracks.
The basic logic of OCPN is, of course, to trust that incoming data is relevant. That OCPN is blamed by the user when an instrument fails is perhaps simple and to some extent understandable. "O should understand that a movement that involves an unreasonable speed should be neglected."
A function where distance, FilterTrackDropLargeJump, is the only parameter becomes very rough in its predictions. It can also easily go wrong as in this case.

Instead of distance, we should perhaps recalculate a change of position to what speed is needed to execute it. That speed limit could be set based on previous speed data in the current track, but it is impossible because we do not know possible speed changes for each boat. Instead, we will probably have to make use of the unreasonableness limit,102.2 knots, that is already used in OCPN. (The exception for SAR aircraft (b_SarAircraftPosnReport) may be made here as well)

leamas, 2024-08-19:
> That speed limit could be set based on previous speed data in the current track, but it is impossible because we do not know possible speed changes for each boat

OTOH, basic physics says that a boat cannot accelerate in an arbitrary way. That is, a limit on  M/h² should be possible to define.

Hakansv, 2024-08-19:
OTOH_2, We can track everything from a canoe via a water jet to a helicopter.

leamas, 2024-08-19:
Yes, but basic physics applies to also to these --  acceleration is limited.

bdbcat, 2024-08-19:
The OP problem is simpler, I think.
Two cases to consider.
1.  Temporary glitch in GNSS, resulting in reported valid position far from actual position.  Previously seen as lat/lon = 0,0. on son GNSS receivers.  The current logic actually works in that case.  Track points that would lead to track segments greater than "FilterTrackDropLargeJump" are ignored.  Default value is 1000 NM.  When the temp glitch subsides, tracking resumes.
2. The "perverse" cases shown by the video clips.  Start tracking.  Play a VDR from Sweden.  Then, leaving tracking on, play a VDR from south Pacific.  Tracking stops, because this is not a temporary glitch.  This is teleportation.
I call it NOTABUG.

bc38, 2024-08-19:
the problem with recording the track was, so to speak, on a level place, i.e. at anchor. I weighed anchor and the track did not write, intermediate points in the track were not saved, just a straight line was stretching behind me.

bdbcat, 2024-08-19:
Do you think that there could have been a bad position report from GNSS receiver during the time you were anchored?  Are these glitches common on your system?

bc38, 2024-08-19:
> Do you think that there could have been a bad position report from GNSS receiver during the time you were anchored? Are these glitches common on your system?

I used GNSS UBLOX M8 with open sources GPS, Glonass, Galileo, speed 5HZ. But I also used another GNSS receiver with default settings 1HZ, GPS, Glonass. The problem repeated!

bdbcat, 2024-08-19:
In the picture shown, how far way is the previous point, which is off-screen to the bottom?

bc38, 2024-08-20:
+/- 20km

bdbcat, 2024-08-21:
bc38...
No one else can duplicate this problem.
We will need some sort of reproducible situation to understand it, and solve it.
Please try to develop some understanding of what conditions on your system cause this bug, to help us reproduce it.
Thanks