OpenCPN/OpenCPN #4091: Add sglock repo for Debian users

leamas, 2024-08-11:
**Is your feature request related to a problem? Please describe.**
To use the o-charts plugin the sglock library is required. This is a closed source drm library available for amd64 and aarch64

**Describe the solution you'd like**
A simple ubuntu PPA which only is used for this library and thus can be used by Debian users.

**Describe alternatives you've considered**
The long term solution is that o-charts handles this dependency internally.

**Additional context**
The library is available in the Ubuntu opencpn PPA. However, this can not be used by Debian users since it pulls in a large load of Ubuntu packages which potentially might destabilize a Debian system.

bdbcat, 2024-09-18:
Two step process here.
1.  First we need a debian package that is safely installable for debian users.  Currently, the PCN ubuntu PPA contains a package for jammy (22.04) only.  Not installable on bookworm.
2. Step two:  migrate sg-lock functionality to the o-charts plugin itself.

Please advise.

leamas, 2024-09-18:
The debian package at https://launchpad.net/~leamas-alec/+archive/ubuntu/opencpn-sglock should be fine. It has very little dependencies, and enabling the PPA does not mean that anything else will be fed into apt. This is basically the only difference compared to the Ubuntu PPA which contains way to much to be visible in Debian.

However, the process for Debian users cannot be made as automatic as for Ubuntu users. The core issue is that the PPA must be enabled, and this means users needs to be able to follow command line instructions how to add the PPA by editing files, etc.

bdbcat, 2024-09-18:
Can we copy this package to OpenCPN PPA?  Historically, users have known how to load stuff from there, never mind the resulting franken-debian.

leamas, 2024-09-18:
> Can we copy this package to OpenCPN PPA? 

No, there are so many risks related to this. What we could to is create a new fresh copy of the PPA which lives on the more official  https://launchpad.net/~opencpn/+archive/ubuntu/opencpn.  That is, create a new PPA rather than copying into the existing. Shoulld be a nobrainer.

> Or crabdance. after removing the now obsolete OCPN beta builds?

I don't really have the infrastructure for this, crabdance is just an RPi 3 on top of a wardrobe.  It's does work for the short term deployment until I get it into official repos, but I really don't want to take responsibility for this.

leamas, 2024-09-20:
Actually, I don't see the problem here. Can't we just announce that for Debian users the way to use o-charts is:

---

     ①  $  sudo add-apt-repository \
              'deb https://ppa.launchpadcontent.net/leamas-alec/opencpn-sglock/ubuntu/ noble main'
         $ sudo apt update
     ②  $ sudo apt install opencpn 

Notes:
   - ① This adds an Ubuntu PPA repository with very few dependencies which should be fine also on Debian, avoiding the problems described in https://wiki.debian.org/DontBreakDebian
   - ② Installing or re-installing opencpn will pull in the required opencpn-sglock package (there are two different variants for Intel x86_64 and Arm arm64)
---

If required, we could move the PPA to  https://launchpad.net/~opencpn¸ I have really no opinion on this. But besides, why not just announce this and be done with it?

leamas, 2024-09-20:
@bdbcat ^

leamas, 2025-01-13:
There si no need for this, the plugin handles it on it's own. Closing

closed 2025-01-13 by leamas