OpenCPN/OpenCPN #4096: No "Navigate To" option in layer waypoint context menu

rp42, 2024-08-13:
**Describe the bug**
I have some local racing marks imported as a layer of waypoints. It is tricky to navigate to one of these as their context menu (right click / long press) doesn't seem to have a navigate to option. There is such an option in the context menu for an empty spot or a dropped mark.

**To Reproduce**
1. Load a waypoints layer
2. Right click on a layer waypoint
3. Look for an option to navigate to the waypoint.

**Expected behavior**
Context menu contains a navigate to this/here option

**Screenshots**
-

**Desktop (please complete the following information if applicable):**
Ubuntu 22.05
OpenCPN 5.10.0

**Smartphone/Tablet (please complete the following information if applicable):**
Device: Samsung S23 Ultra
Android version: 14
OpenCPN 5.8.4

**Additional context**
I'm a C++ dev, so happy to have a go a looking at the code. Thanks for developing such a great app!

Hakansv, 2024-08-13:
@rp42 
I wouldn't call this a bug. Maybe a future enhancement. 
I bug is when the program for any unintentional cause don't work as expected. Like a typo or bad coding.
In this case the program does as it is designed. 
If you think the program should work in a different way, that's a future request or similar.

leamas, 2024-08-14:
> I wouldn't call this a bug. Maybe a future enhancement.

Indeed. Fixing tags

leamas, 2024-08-14:
> I'm a C++ dev, so happy to have a go a looking at the code. Thanks for developing such a great app!

Welcome on board! Hopefully you have found the docs such as they are. The developer manual [here]( https://opencpn-manuals.github.io/main/ocpn-dev-manual/0.1/intro-AboutThisManual.html)

If you have more informal questions, chit chat etc: welcome to the zulip instance, see [the community](https://opencpn-manuals.github.io/main/ocpn-dev-manual/0.1/intro-Community.html#_the_core_opencpn_project)

Hakansv, 2024-08-14:
@rp42 
About layers.
Layers are mainly intended for informational purposes. Maybe radio beacons, special buoys and the like. It is not intended for navigation purposes such as routes or waypoints. Such a gpx file should instead be imported, via Route & Mark Manager", and then all marks can be used as normal. To start "navigate to" a certain position, it is easiest to hover there and via the right-click menu "Navigate to here".
The Navigate to here function will also start sending autopilot messages to any output connection if not filtered out.

rp42, 2024-08-14:
Hi all. Thanks for the prompt responses and suggestions. I don't mind how this issue is classified, but I do wonder if I perhaps didn't explain the issue very well. What I was trying to do was to use the right-click menu "Navigate to here". This works very well for me, except when there happens to be a layer waypoint in the vicinity. In that case the entry is missing from the "Main menu" submenu.

[image]

I find using layers for these marks really handy. I have a number of externally provided gpx files with such marks, each having a large number of marks. The layers allow switching between the different sets easily and keeps the standard marks clear for ones I create myself.

bdbcat, 2025-05-06:
Closed by https://github.com/OpenCPN/OpenCPN/commit/c403cfc3c968f51936befe758bea960d2297a6a4

closed 2025-05-06 by bdbcat