OpenCPN/OpenCPN #4098: Add Plugin API getUsrDepthUnit_Plugin()

TwoCanPlugIn, 2024-08-14:
While there are a few plugin API's to get the user's speed, distance and temperature units, there isn't one for depth units.

Also one could contemplate using the correct ISO or IHO abbreviations for knots and nautical miles, but I guess what OpenCPN has adopted has become a norm.

TwoCanPlugIn, 2024-08-17:
I would fix this myself, however I am unclear about the relationship between cli/api_shim.cpp and include/ocpn_plugin.h (which is used by the plugins).

gui/src/pluginmanager.cpp has a definition for these conversions and it merely calls methods defined in model/src/navutil_base,cpp 

Bizarrely navutil_base.cpp already definitions for getUsrDepthUnit, toUsrDepth, fromUsrDepth, just that the corresponding methods ( getUsrDepthUnit_Plugin, toUsrDepth_Plugin, fromUsrDepth_Plugin )are missing in pluginmanager.cpp

Also noticed that the windSpeed conversions are similarly not declared in ocpn_plugin.h.

Also there is an opportunity to rationalize and make consistent the enums. Is there really a need to have an enum for both windspeed and speed when both have the same units of measurment (Knots, metres/second, miles per hour, kilometres per hour) ? Similary for distance and depth, both are a lineal measure ?

leamas, 2024-08-19:
> however I am unclear about the relationship between cli/api_shim.cpp and include/ocpn_plugin.h 

We compile two type of targets, both uses _ocpn_plugin.h_. For the opencpn target, _ocpn_plugin.h_ is implemented in too many places. 

For other CLI targets (tests, opencpn-cmd, etc)  _ocpn_plugin.h_ is implemented by _api_shim.cpp_ which basically is empty definitions to satisfy the compiler and plugins.

> Bizarrely navutil_base.cpp already definitions for getUsrDepthUnit, toUsrDepth, fromUsrDepth, it's just that the corresponding methods ( getUsrDepthUnit_Plugin, toUsrDepth_Plugin, fromUsrDepth_Plugin )are missing in pluginmanager.cpp and then in ocpn_pluhin.h

pluginmanager is an evil jack of all trades, if we want to fix this it should not be done there. The best candidate is _ocpn_plugin.cpp_ . That said, exporting things which exists in _navutil_base.cpp_ is trivial.

We should also move stuff from pluginmanager to other places, leaving the GUI dialogs and only those thus cleaning up this mess. This is an ongoing process, for example current plugin_loader was once part of an even more messy pluginmanager. But this is another story.

You might want to check out https://opencpn-manuals.github.io/main/opencpn-dev/gui-model.html. pluginmanager is part of the GUI, navutil_base is in model.