OpenCPN/OpenCPN #4103: wrong direction to steer in RMB and XTE NMEA message in aktiv route when changing to next Waypoint

BerndCirotzki, 2024-08-19:
When the Boat is entering the arrivalcycle OpenCPN is steering to the next Waypoint.
But when the steering direction has to be changed in RMB and XTE Message it is not done.
effect : the Autopilot where the sentence is send to is starting to make a big turn into the wrong direction.

after the arivalcycle from the entered waypoint is lost, the direction in RMB and XTE sentence is chaned to the right direction.

... see in video ... I make the test with OPENCPN 5.10.0.

https://github.com/user-attachments/assets/e2727e5f-30a1-4543-8498-2c19ae24ec99

bdbcat, 2024-08-19:
Looking closely at the RMB sentence in your video.
I see a bug, maybe.  At second 13 of the video, the RMB shows "arrival circle entered".  But, the "course to waypoint" seems to be not yet advanced to the next point.

bdbcat, 2024-08-20:
If we understand the problem correctly, then what is needed is to immediately send a new RMB sentence when the next waypoint has been activated.
This will result in two RMB sentences sent in quick succession.  The first will indicate "arrival", for those A/Ps that might be monitoring this flag.  The second will describe the next waypoint in the route, if present.
Note that the first RMB sent will contain the "bearing" to the waypoint just arrived.  Might be an an unexpected direction, since we cannot know from what direction the arrival circle was entered.  But, the second will follow quickly, so that the A/P has little or no opportunity to follow the first RMB.

This is a little complicated, and would need some real at-sea testing before introduction.
I think we must defer this solution until OCPN 5.12, where some detailed testing might be conducted.

BerndCirotzki, 2024-08-20:
I think it's all quite simple. It doesn't matter where the arrival circle was approached from.
The new waypoint will then be approached immediately in the new RMB message with the correct new bearing. but wrong steering direction.

while steering to waypoint 001 :
$ECRMB,A,0.0045,L,,001, ...... V,A*XX     ....this is OK

arrive the arrival circle ...
One RMB :
$ECRMB,A,0.0045,L,,001, ...... A,A*XX    ....this is OK
(This makes my autopilot "Raymarine S1" to confirm the next waypoint.)

now steering to next waypoint ....
OPENCPN sends :
with the correct new Bearing of 277 deg but the wrong steering direction because the new waypoint is now not to steer to Left it is to steer to "right"
$ECRMB,A,0.0045,L,001,002, ...... V,A*XX 
___________________^
		    	    This is worng ... must be ,R, in this video example that's all.

and because of this, the autopilot turns to the next waypoint, to 277 deg, but not by turning right, ... the vessel turns left and makes a big U-turn. not so good.

$ECRMB,A,0.0045,R,001,002, ...... V,A*XX .... this would be correct.

bdbcat, 2024-08-20:
```
$ECRMB,A,0.0045,L,001,002, ...... V,A*XX
This is worng ... must be ,R, in this video example that's all.
```

Well, not exactly our idea of how RMB should work.
The "direction to steer" is derived from the "cross track error".  On the new course, at this point in time, the vessel is to the RIGHT of the direct course from last waypoint to next waypoint.  So, the pilot should turn to LEFT to regain the course line.

This seems funny behavior of the A/P.  From the RMB, the first order should be to steer toward the new waypoint.  A second priority should be to reduce the XTE. 

I suppose it might be possible to somehow "reset" the XTE calculation by creating a "phantom" course line from the point where arrival circle is entered to the next waypoint.  But this seems a little dangerous in the general case, since one must assume that the course lines and waypoints have been layed out with some care to avoid obstructions.

BerndCirotzki, 2024-08-20:
I think this works fine, when the vessel is already on a good way to the next waypoint an has only do do little changes.
But in this case when switching to the next WP and the difference between the new bearing to WP002 and the vessel heading or COG is too big.
Why ist the second waypoint already activ when the first is not reached yet ? It's only the area that is reached.

Hakansv, 2024-08-20:
Isn't this a simulating problem of some kind.
In real when within the arrival circle of the nearest WP O route is switched to next WP with a changed course to steer and zero XTE.
Then a "normal" AP is leaving Route/Nav/PC-mode and is set to Auto using last course.
You have to again press Route/Nav/PC-mode and if there are accepted Nav-data on the line the AP read the course to steer and use that as a set point for the steering regulator. The course to steer is read only once on this leg and further XTE is used to adjust the course when necessary.
All is working very good when navigating with my boat.
Also are there no complains from those using O Nav-mode to control other APs like "AutoTrackRaymarine". (Except from the WP-equal-name trouble discussed in another issue.)   
In you simulated example I'd manually switched to next WP instead of heading to a (start)WP "out of course" and then reloaded the Nav-mode on the AP.

BerndCirotzki, 2024-08-20:
Hi Hakansv ... the problem is in real live ... My Raymarine S1 ist doing it this way.
The simulation I only made to find out where the problem is.

But I discribed how it could be fixed. could be a little complicated but not insoluble.
I made some picture to my Idea. the second could be a problem, but not really. The course could be changed, when arriving the route line from WP2 to WP3, but it's outside the arrival cycle.

[image]
[image]