OpenCPN/OpenCPN #4105: opencpn 5.10.0 initialization issue with multiple screens

alannz42, 2024-08-19:
I have compiled opencpn from the github source code in OpenCPN-Release_5.10.0.tar.gz
on Almalinux x86_64 version 9.4.

It works - thank you (with raster charts, S63 charts, AIS input).

I have an odd initialization problem.

I run opencpn from the command line - it works. I exit from the Navigate/exit menu.

When I run it again I cannot see any charts and the scale in the bottom right hand corner shows 1:0.
This happens every time I exit from the menu or click on X on the window.

I kill opencpn with CTRL-C.

When I run it again I get the "The last opencpn run seems to have failed" Do you want to run in safe mode popup.
If I answer NO (run normally with all plugins) it runs perfectly displaying my charts.

Opencpn will only run properly if the safe mode popup displays (you can answer both yes or no).

If it runs without the safe mode popup I get scale 1:0 and no charts (the zoom +/- menu does not work).

So why?

I tested my new rpm on a near identical machine with one display and it works fine.
I tested running it on my machine but via VNC with a single display and it works fine.

This is a complete guess but the only major difference on my machine is that I have 3 monitors.

opencpn clearly initializes differently if the safe mode popup is displayed.

bdbcat, 2024-08-20:
Check your logfile.
You will see that one or more of your displays have detected dimensions of "0x0".
There is a patch to fix that in github now

nohal, 2024-08-20:
Try with the current master, which has a fix for headless X11/VNC scenario, where the system reports even non -existent monitors as having 0x0 resolution and  0x0mm size.
If it does not resolve your problem, run opencpn with `-l debug` and look into the logfile, we are interested in the part at the very beginning of the run, where it reports the monitors. Should look something like this (just showing 3 monitors for you, not 2 like on my system):

```
19:37:36.128   DEBUG displays.cpp:133 GDK Monitors: 2
19:37:36.128   DEBUG displays.cpp:154 Monitor 1:
19:37:36.128   DEBUG displays.cpp:155   Name: HDMI-1
19:37:36.128   DEBUG displays.cpp:156   Connection: 0
19:37:36.128   DEBUG displays.cpp:157   Physical Size (mm): 600 x 340
19:37:36.128   DEBUG displays.cpp:158   Resolution: 1920 x 1080
19:37:36.128   DEBUG displays.cpp:160   Scale: 100
19:37:36.128   DEBUG displays.cpp:154 Monitor 2:
19:37:36.128   DEBUG displays.cpp:155   Name: eDP-1
19:37:36.128   DEBUG displays.cpp:156   Connection: 0
19:37:36.128   DEBUG displays.cpp:157   Physical Size (mm): 340 x 190
19:37:36.128   DEBUG displays.cpp:158   Resolution: 1920 x 1080
19:37:36.128   DEBUG displays.cpp:160   Scale: 100
19:37:36.128   DEBUG displays.cpp:258 Number of monitors: 2
19:37:36.128   DEBUG displays.cpp:259 Monitor info:
19:37:36.128   DEBUG displays.cpp:261 Monitor: HDMI-1 600x340mm 1920x1080DIP 1920x1080px 100%
19:37:36.128   DEBUG displays.cpp:261 Monitor: eDP-1 340x190mm 1920x1080DIP 1920x1080px 100%
```

leamas, 2024-08-20:
That patch is BTW applied to the Flatpak builds which should be fine to run on Alma.

alannz42, 2024-08-21:
Thank you. Compiling from master works as expected.
I attach the following debug as I am concerned 2/3 connected monitors have a resolution of 0x0.

16:13:28.740   DEBUG displays.cpp:133 GDK Monitors: 3
16:13:28.741   DEBUG displays.cpp:154 Monitor 1:
16:13:28.741   DEBUG displays.cpp:155   Name: HDMI-1
16:13:28.741   DEBUG displays.cpp:156   Connection: Disconnected/Unknown
16:13:28.741   DEBUG displays.cpp:160   Physical Size (mm): 444 x 277
16:13:28.741   DEBUG displays.cpp:161   Resolution: 1680 x 1050
16:13:28.741   DEBUG displays.cpp:163   Scale: 100
16:13:28.741   DEBUG displays.cpp:154 Monitor 2:
16:13:28.741   DEBUG displays.cpp:155   Name: HDMI-2
16:13:28.741   DEBUG displays.cpp:156   Connection: Disconnected/Unknown
16:13:28.741   DEBUG displays.cpp:160   Physical Size (mm): 0 x 0
16:13:28.741   DEBUG displays.cpp:161   Resolution: 0 x 0
16:13:28.741   DEBUG displays.cpp:163   Scale: 100
16:13:28.741   DEBUG displays.cpp:154 Monitor 3:
16:13:28.741   DEBUG displays.cpp:155   Name: DP-1
16:13:28.741   DEBUG displays.cpp:156   Connection: Disconnected/Unknown
16:13:28.741   DEBUG displays.cpp:160   Physical Size (mm): 0 x 0
16:13:28.741   DEBUG displays.cpp:161   Resolution: 0 x 0
16:13:28.741   DEBUG displays.cpp:163   Scale: 100
16:13:28.745   DEBUG displays.cpp:154 Monitor 4:
16:13:28.745   DEBUG displays.cpp:155   Name: HDMI-3
16:13:28.745   DEBUG displays.cpp:156   Connection: Connected
16:13:28.745   DEBUG displays.cpp:160   Physical Size (mm): 459 x 296
16:13:28.745   DEBUG displays.cpp:161   Resolution: 1680 x 1050
16:13:28.745   DEBUG displays.cpp:163   Scale: 100
16:13:28.745   DEBUG displays.cpp:154 Monitor 5:
16:13:28.745   DEBUG displays.cpp:155   Name: VGA-1-1
16:13:28.745   DEBUG displays.cpp:156   Connection: Disconnected/Unknown
16:13:28.745   DEBUG displays.cpp:160   Physical Size (mm): 444 x 277
16:13:28.745   DEBUG displays.cpp:161   Resolution: 1680 x 1050
16:13:28.745   DEBUG displays.cpp:163   Scale: 100
16:13:28.745   DEBUG displays.cpp:154 Monitor 6:
16:13:28.745   DEBUG displays.cpp:155   Name: DVI-D-1-1
16:13:28.745   DEBUG displays.cpp:156   Connection: Connected
16:13:28.745   DEBUG displays.cpp:160   Physical Size (mm): 474 x 296
16:13:28.745   DEBUG displays.cpp:161   Resolution: 0 x 0
16:13:28.745   DEBUG displays.cpp:163   Scale: 100
16:13:28.745   DEBUG displays.cpp:154 Monitor 7:
16:13:28.745   DEBUG displays.cpp:155   Name: DVI-D-1-2
16:13:28.745   DEBUG displays.cpp:156   Connection: Connected
16:13:28.745   DEBUG displays.cpp:160   Physical Size (mm): 430 x 270
16:13:28.745   DEBUG displays.cpp:161   Resolution: 0 x 0
16:13:28.745   DEBUG displays.cpp:163   Scale: 100
16:13:28.745   DEBUG displays.cpp:264 Number of monitors: 1
16:13:28.746   DEBUG displays.cpp:265 Monitor info:
16:13:28.746   DEBUG displays.cpp:267 Monitor: HDMI-3 459x296mm 1680x1050DIP 1680x1050px 100%

alannz42, 2026-02-14:
Now historic

leamas, 2026-02-14:
Ok, let's close it them Thanks for reporting!

closed 2026-02-14 by leamas