OpenCPN/OpenCPN #4106: OpenCPN freezes with o-charts for Belgium and Netherlands

mgrouch, 2024-08-19:
Rpi 5, Debian version from crabdance. 

Thanks

bdbcat, 2024-08-20:
I have loaded the charts, using rPI5 with bookworm and crabdance repo for OpenCPN.
Initial test seems OK.
Any particular action that causes freezing?

leamas, 2024-08-22:
> When using OpenCPN on a laptop I sometime get a freeze where OpenCPN executes a backlog of commands. When sailing in England I might suddenly view the world small. In those cases I have to zoom in again, look up the area I want too see, etc..

Happens to me as well. In my case it's related to the virtual scroll wheel in the touchpad, I make some moves which turns the wheel very fast and I find myself looking at a world map. Very frustrating.

bdbcat, 2024-08-23:
This is Interesting...
Using the charts referenced above, on rPI5/bookworm.
I see significant performance difference between OCPN native and OCPN flatpak.  So much so that this chartset is basically not usable for  effective navigation on flatpak, even with modern rPI5.
I am beginning to suspect the OpenGL drivers bundled in flatpak are somehow stalling the rendering, at least with these charts.
More to come.

bdbcat, 2024-08-24:
BTW, unless I misunderstand, this issue does NOT relate to o-charts.
If that is correct, will OP please edit the Issue title?
Thanks
Dave

transmitterdan, 2024-08-24:
> This is Interesting... Using the charts referenced above, on rPI5/bookworm. I see significant performance difference between OCPN native and OCPN flatpak. So much so that this chartset is basically [...]

I seem something similar on WSL-Ubuntu Jammy.  O is extremely slow to the point it cannot handle all the incoming NMEA data.  When I use gdb to break it is always here:
[image]

It's as if the gtk rendering is just so slow it can never catch up and is doing nothing but handling OnPaint() events.

transmitterdan, 2024-08-24:
Well, it's the opengl test failing under WSL so O reverts back to non-opengl mode and just cannot keep up with OnPaint() events.

bdbcat, 2024-08-24:
Not too terribly concerned with problems on WSL.  This has never been a design target for OCPN.
But still, I would like to understand why the GL test is failing here.
What does the log and gl_caps.json file (if present) say?

transmitterdan, 2024-08-24:
I fails because install target does not install the opengl utility app.  Manually moved it and now GL works.

nohal, 2024-08-24:
I have invested some science into this mess.

RPi5, 8GB RAM, NVMe disk, VNC connection over a gigabit network with 192x1080 resolution

OpenCPN with default configuration
Only https://github.com/chartcatalogs/catalogs/blob/master/NL_IENC_Catalog.xml charts installed

I start OpenCPN in a way that it shows the following immediately, so the problem manifests itself right after the launch:
[image]
I quit OpenCPN as soon as it displays the whole UI, starts to be responsive and reacts to the the close event.

I collect data using `perf record -a -g ./opencpn`, all results attached in [perf_data.tar.gz](https://github.com/user-attachments/files/16736955/perf_data.tar.gz)

The attached data can be viewed and analyzed using `perf report -i <perf data file.data>`

It seems clear that the problem is in the VC4 GPU driver - as soon as it is used most of the time is spent in the driver when it is called from `s52plib::RenderToGLAC_GLSL(_ObjRazRules*, _Rules*)`

Performance of both OpenGL disabled and Software OpenGL is quite satisfactory.

bdbcat, 2024-08-24:
I suspect that the VC4 driver does not handle large VBO very well.  The GPU core does not have dedicated memory, and it may be thrashing somehow.
Rebuilding now with VBO disabled for GLAC rendering.  Takes a while to build, but we shall see.

mgrouch, 2024-08-24:
I've just tried with same 

Catalog for Dutch charts:

https://raw.githubusercontent.com/chartcatalogs/catalogs/master/NL_IENC_Catalog.xml

on pi5 with Debian opencpn_5.10.0+dfsg-1.bpo12+3_arm64.deb installed.

I've prepared all ENC charts.

Then I soon as I zoom in to 10nm OpenCPN locks-up and eventually crashes.

This is Debian opencpn_5.10.0+dfsg-1.bpo12+3_arm64.deb installed and NOT flatpak.

Thanks

mgrouch, 2024-08-24:
The same lock up and crash happens with opencpn_5.10.0 installed from Ubuntu PPA (Jammy) arm64 on pi5 on those Dutch charts.

Thanks

mgrouch, 2024-08-24:
OpenCPN 5.8.4 seems handles those charts much better but still sometimes locks up. And shows chart filled with depth soundings text. Too much of text for depth sounding and chart is almost unreadable.

Thanks

mgrouch, 2024-08-24:
BTW are you sing mmap or read when opening/reading chart files for reading?
Using mmap is faster in Linux on read-only access.

Thanks

mgrouch, 2024-08-24:
Just to verify that nothing is fundamentally wrong with those charts I've installed QTVLM on the same SD card and run
in the same pi5. QTVLM handles those charts just fine and fast.

Thanks

transmitterdan, 2024-08-24:
@bdbcat A possible hint.  See around line 15K.
[opencpn.log](https://github.com/user-attachments/files/16738365/opencpn.log)

bdbcat, 2024-08-27:
Here is a little more info on this chart set.
1.  All of the charts for Waddenzee and Zeeland charts are scaled at 1:2000.  This is a very large scale chart, and is commonly used for inland rivers and lakes.
2. Coverage of large geographical areas is done by tiling the 1:2000 scale charts.  This by itself is not unusual, and indeed is the direction that many HOs are shifting for their ENC sets.  However, on the Zeeland and Waddenzee IENCs, there are no intermediate smaller scale charts provided in the set.  As a result, displaying quilted charts at small viewing scale involves the loading and processing of a very large number of individual large scale cells.
3. The combination of many cells and poor OpenGL drivers is not handled well in OpenCPN at smaller view scales, particularly on low-spec machines like 2GB rPI/4.
4. There are two possible methods to avoid these problems:
a)  Use the OpenCPN Software GL Renderer.  It is counterintuitive, but performance is in this case better than can be provided by the embedded GPU core in the ARM chipset.
b)  Use the o-charts chartset for active navigation in Zeeland/Waddenzee.  The o-chart set contains a number of intermediate scale cells to provide better performance at smaller viewing scales.

Further research on this issue is deferred to OpenCPN 5.12 or later.

mgrouch, 2024-08-27:
It doesn’t work even on 8Gb pi5. 
There seems to be nothing wrong with the charts as the same charts work fine on the same machine in QTVLM. Only OpenCPN seems not able to handle them. 

Thanks

bdbcat, 2024-08-27:
If you require these specific charts today, then I suggest that you use qtVlm.

mgrouch, 2024-08-27:
> If you require these specific charts today, then I suggest that you use qtVlm.

Thanks I’ll suggest it to the users.

leamas, 2024-08-27:
> It doesn’t work even on 8Gb pi5.
>> If you require these specific charts today, then I suggest that you use qtVlm.

Or different hw. Made a try with my [Orange Pi 5](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5.html) (Opi5), and it works just fine. 

The Opi5 has a better GPU, the Mali-G610.  It costs a few bucks more, but then you get a nmve socket as well.

I'm currently on the custom kernel from the OrangePi vendor, but the RK3588 Mali drivers has recently been upstreamed to the mainline kernel. Intend to switch later this autumn.