OpenCPN/OpenCPN #4108: Calling the Chart Panel Options menu changes the map orientation to North

bc38, 2024-08-20:
Calling the Chart Panel Options menu changes the map orientation to North!
Win11

Hakansv, 2024-08-20:
Please use the bug template to describe the steps to reproduce. What's "Chart Panel Options"?
I can't get any menu to change the Navigation mode unintentionally.

bc38, 2024-08-20:
after rebooting CPN the bug disappeared!
devilry!

Hakansv, 2024-08-20:
Sorry, "Chart Panel Options" is the official name for the "Hamburger" menu.
But I can't reproduce.

Hakansv, 2024-08-20:
bc38..
I actually found a way to reproduce. I'll be back to describe.

Hakansv, 2024-08-20:
@bdbcat 
Steps to reproduce:
Use single or double canvas mode.
1. Use the Hamburger menu to set a Navigation mode, e.g. "Heading up". Close the menu.
2. Change to e.g. North up using the upper right icon next to Compass/GPS status icon.
3. Open the Hamburger menu. The navigation mode is now changed to last used Hamburger menu option, here "Heading up".
4. For more tests: Use the Hamburger menu to set a Navigation mode, e.g. "North up". Close the menu.
5. Use the upper right icon to change to "Course up"
6. Open the Hamburger menu. The navigation mode is now changed to last used Hamburger menu option, here "North up".

Conclusion:
The navigation mode set by the upper icon is reset to last used Hamburger menu option when the menu is opened.

I'd call this a 5.10.2 fix?
Håkan

bdbcat, 2024-08-20:
Hmmm...
I can reproduce on Windows, but not on linux
Hmmm.

Hakansv, 2024-08-24:
Tested the above correction on Win and it's now working  as expected. I suggest closing.

closed 2024-08-25 by bc38