OpenCPN/OpenCPN #4111: Missing plugins search path with 5.10.0 version (I had a working workaround for 5.8.4)

filochard, 2024-08-21:
Hi everyone (particularly bdbcat, Nohal, Alec and Sean de Panier)
I continue to use a legacy install of OpenCPN and its plugins (rpm based)
I am actually testing OpenCPN 5.10.0 (built without any patch and installed as a rpm)
It can't find the plugins previously built and  installed as rpms that are found by 5.8.4 by my patched version

The problem to find the plugins started with 5.8.2 and 5.8.4 version
https://github.com/OpenCPN/OpenCPN/issues/3334
I finally found a workaround  by patching 
- plugin_paths.cpp
- plugin_loader.cpp
and that worked : 
https://github.com/OpenCPN/OpenCPN/issues/3334#issuecomment-1601726695

I adapted these patches for OpenCPN 5.10.0
[opencpn-5.10.0-plugin_loader.txt](https://github.com/user-attachments/files/16692151/opencpn-5.10.0-plugin_loader.txt)
[opencpn-5.10.0-plugin_paths.txt](https://github.com/user-attachments/files/16692154/opencpn-5.10.0-plugin_paths.txt)
Unfortunately when building I get a problem : OCPN_LINUX_LOAD_PATH is not defined

This may mean that since now I won't  be able anymore to build rpms for OpenCPN and tell the users to use flatpak
Which means losing all their data created by the legacy installation and having to regularly download Go of updates for Flatpak itself

bdbcat, 2024-08-21:
Without patches, there is an Environment variable that modifies the plugin load path.
I think it is "OPENCPN_PLUGIN_DIRS", semicolon separated list of search locations.

filochard, 2024-08-21:
Hi Dave
Thanks for your quick reply
I don't know where to modify this Environment variable

filochard, 2024-08-21:
Hi Alec
I rewrote my opencpn-plugin_paths.patch this way : 

--- a/model/src/plugin_paths.cpp	2024-08-10 03:54:37.000000000 +0200
+++ b/model/src/plugin_paths.cpp	2024-08-21 16:19:40.228882068 +0200
@@ -133,7 +133,7 @@
 #endif
 
   const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS");
-  string dirlist = envdirs ? envdirs : "~/.local/lib/opencpn";
+  string dirlist = envdirs ? envdirs : "/usr/lib/opencpn:/usr/lib64/opencpn";
   m_libdirs = split(dirlist, ':');
   for (auto& dir : m_libdirs) {
     dir = expand(dir);

I did two test : 
1) after having renamed my previous /home/myusername/.opencpn -> /home/myusername/.opencpnbak
I have updated  OpenCPN 5.8.4 to OpenCPN 5.10.0
I launch it which creates a new  /home/myusername/.opencpn
the only plugins I can see are chartdldr wmm grib and dashboard 
(even if the others are installed in the same /usr/lib64/opencpn/ directory

2) I stop OpenCPN 5.10.0
I erase the newly created  /home/myusername/.opencpn
I copy the /home/myusername/.opencpnbak to a /home/myusername/.opencpn
I launch again OpenCPN 5.10.0
It can use all the plugins, use the data that were created by OpenCPN 5.8.4

To conclude : 
1) using for the first time OpenCPN 5.10.0  installed with the plugins that I already packaged and that are recommended and installed with it , it can't use the plugins that are nevertheless in the same directory as chartdldr wmm grib and dashboard 

2) Updating an existing and used version (5.8.4) to the new (5.10.0) is painless : it uses the previous  /home/myusername/.opencpn and all the configs and data are preserved, and the plugins are found and OK

filochard, 2024-08-21:
Hi Alec
I rewrote my opencpn-plugin_paths.patch this way : 

--- a/model/src/plugin_paths.cpp	2024-08-10 03:54:37.000000000 +0200
+++ b/model/src/plugin_paths.cpp	2024-08-21 16:19:40.228882068 +0200
@@ -133,7 +133,7 @@
 #endif
 
       const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS"); 
    string dirlist = envdirs ? envdirs : "~/.local/lib/opencpn"; 
    string dirlist = envdirs ? envdirs : "/usr/lib/opencpn:/usr/lib64/opencpn";
       m_libdirs = split(dirlist, ':');
       for (auto& dir : m_libdirs) {
       dir = expand(dir);

NB I can't write - or + before the 2 string dirlist  lines  : this changes the preview

I did two test : 
1) after having renamed my previous /home/myusername/.opencpn -> /home/myusername/.opencpnbak
I have updated  OpenCPN 5.8.4 to OpenCPN 5.10.0
I launch it which creates a new  /home/myusername/.opencpn
the only plugins I can see are chartdldr wmm grib and dashboard 
(even if the others are installed in the same /usr/lib64/opencpn/ directory

2) I stop OpenCPN 5.10.0
I erase the newly created  /home/myusername/.opencpn
I copy the /home/myusername/.opencpnbak to a /home/myusername/.opencpn
I launch again OpenCPN 5.10.0
It can use all the plugins, use the data that were created by OpenCPN 5.8.4

To conclude : 
1) using for the first time OpenCPN 5.10.0  installed with the plugins that I already packaged and that are recommended and installed with it , it can't use the plugins that are nevertheless in the same directory as chartdldr wmm grib and dashboard 

2) Updating an existing and used version (5.8.4) to the new (5.10.0) is painless : it uses the previous  /home/myusername/.opencpn and all the configs and data are preserved, and the plugins are found and OK

filochard, 2024-08-21:
There something strange in the fact 
that a previous /home/myusername/.opencpn  works well with a simple update of OpenCPN and can use the installed plugins
 but 
that a fresh install creates a new /home/myusername/.opencpn  unable to use the same  plugins installed in the same directory as the bundled ones

filochard, 2024-08-21:
I only need a simple patch to allow OpenCPN 5.10.0 to find its plugins (the same way as I did for OpenCPN 5.8.4)
It must be userfriendly for the rpms users when they update OpenCPN...

filochard, 2024-08-21:
As I said months ago : using flatpak instead of legacy rpms is as irrational as carrying 12 eggs with a truck instead of a basket : 
Since I test the flatpak versions of OpenCPN I have had to download several Go to only keep flatpak updated...

filochard, 2024-08-23:
> You have been on thin ice since some time, and sooner or later it will break.
> 
> That said, you can define environment variables if you move the installed /usr/bin/opencpn to say /usr/libexec/opencpn and installs a shellscript /usr/bin/opencpn which sets up the environment and then does "exec /usr/libexec/opencpn" or so.

Hi Alec and thanks

I have tried your simple recipe (with a non patched built OpenCPN 5.10.0 installed as a rpm) this doesn't work : 

1) with a fresh install OpenCPN 5.10.0 doesn't find the plugins that are installed in the same directory as the 4 bundled ones 
(/usr/lib64/opencpn ; /usr/share/opencpn/plugins ;  /usr/share/locale)
2) if I use the /home/myusername/.opencpn/ directory created by a previous installed OpnCPN 5.8.4 , then OpenCPN 5.10.0 can find all the plugins

filochard, 2024-08-23:
To Dave and Alec
Maybe you know that I use (and package) OpenCPN since 2015 
I have contributed to correct some bugs or improve some plugins
I have found some incompatibility with some versions of wxgtk

I have been testing the flatpak versions since they exist and brought some feedbacks for them

I understand that you promote these flatpak versions for Linux distributions as they allow to use the Plugin Manager

But Flatpak is a very heavy container, needing lots of heavy updates (several Go since I began to test it)
Flatpak needs a workaround that is not user friendly to get access to Charts that are not in the /home/myusername/ directory
here is the argument I have to use to allow OpenCPN to find my own charts
run --filesystem=/mnt/Seagate/CD_ISOS/cartes-marines/ --branch=stable --arch=x86_64 --command=opencpn.sh org.opencpn.OpenCPN

with Flatpak, a user loses all the configs, data, that he had created with OpenCPN and plugins installed as packages : 
it's not simple for a simple user to look inside /home/hisusername/.opencpn and copy what is useful 
to /home/hisusername/.var/org.opencpn.OpenCPN/

That's the reason why I continue to package OpenCPN and its plugins, so that updating this stuff is painless for old users...

With OpenCPN 5.8.2 and 5.8.4 I needed 2 patches for plugin_paths.cpp and plugin_loader.cpp that were suggested (or found by me) that allowed to continue to package and use OpenCPN and preserve all the previous configs and data

Now I am stuck and don't know where the source code needs to be modified so that OpenCPN can find the plugins installed by rpms besides the bundled plugins (chartdldr grib wmm dashboard)

You know the code better than me and know what changes have been done since OpenCPN5.8.4 and I would be grateful if you lead me to a solution

I only found this inside plugin_handler.h : 

 However, plugins are loaded from multiple locations basically
 corresponding to new and old lacations:
   Windows : GetSharedDataDirPtr():GetWinPluginBaseDir()
  linux : As defined  by XDG_DATA_DIRS, defaulting to
    /.local/lib/opencpn:/usr/local/lib/opencpn:/usr/lib/opencpn
  flatpak :
   /app/lib/opencpn:/app/extensions/lib/opencpn:~/.var/app/opencpn/lib
 
My problem perhaps comes from the fact that XDG_DATA_DIRS doesn't define /usr/lib64/opencpn as a location by default

nohal, 2024-08-23:
You do not need any patch with OpenCPN 5.10, to search for plugins in a non-standard location you need to set the `OPENCPN_PLUGIN_DIRS` environment variable.
Run OpenCPN 5.10 from command line using `OPENCPN_PLUGIN_DIRS=</wherever/plugin/libs/are> opencpn -l debug` and look into your logfile, you should see that they are found.

filochard, 2024-08-23:
Thank you very much Pavel

Indeed you are right the log shows they are all found, but indeed they are not displayed in the gui...
The problem leans here for each of them : 
17:19:06.765 MESSAGE plugin_loader.cpp:446 Checking plugin candidate: /usr/lib64/opencpn/libpolar_pi.so
17:19:06.765   DEBUG plugin_loader.cpp:498 Skipping plugin /usr/lib64/opencpn/libpolar_pi.so in /usr/lib64/opencpn

OpenCPN 5.10.0 doesn't like legacy plugins
Indeed for OpenCPN 5.8.4 I already had to patch plugin_loader.cpp   to suppress these lines : 

// Avoid loading/testing legacy plugins installed in base plugin path.
 wxFileName fn_plugin_file(file_name);
wxString plugin_file_path = fn_plugin_file.GetPath(wxPATH_GET_VOLUME | wxPATH_GET_SEPARATOR);
wxString base_plugin_path = g_BasePlatform->GetPluginDir();
if (!base_plugin_path.EndsWith(wxFileName::GetPathSeparator()))
   base_plugin_path += wxFileName::GetPathSeparator();
 if (!g_bportable){
   if (base_plugin_path.IsSameAs(plugin_file_path)){
    if (!IsSystemPlugin(file_name.ToStdString())){       
DEBUG_LOG << "Skipping plugin " <<  file_name << " in " << g_BasePlatform->GetPluginDir();
       return false;
      }
   }
 }

I am gonna try to patch it for OpenCPN 5.10.0 and verify if that changes something

filochard, 2024-08-23:
OK
The problem is worse than I thought
Now that the patch allows not to skip plugins sitting in base_plugin_path

They are all of them blacklisted (except the bundled ones)
for instance
17:59:54.300   DEBUG plugin_loader.cpp:1474 blacklist: Get status for o-charts 2 0
17:59:54.301   DEBUG plugin_handler.cpp:352 Cannot parse API version ""

filochard, 2024-08-23:
Two last tests
1) erase  the last  /home/myusername/.opencpn created by OpenCPN 5.10.0
I have launched OpenCPN 5.10.0 without  setting the OPENCPN_PLUGIN_DIRS environment variable
only used opencpn -l debug in the command line
the debug log shows that the plugins are found in /usr/lib64/opencpn
but they are all blacklisted 
and they don't appear in the GUI

2) erase the last  /home/myusername/.opencpn created by OpenCPN 5.10.0
copy /home/myusername/.opencpnbak (created by OpenCPN 5.8.4) to  /home/myusername/.opencpn
I launch OpenCPN 5.10.0
Everything is OK !!! all my installed plugins are present and working

There's something strange about parsing API version

filochard, 2024-08-23:
when OpenCPN 5.10.0 is launched as a fresh install with no previous  /home/myusername/.opencpn   

None of the installed plugins can be used (they are immediately  found in the right path since I have patched plugin_paths.cpp (defining the environment variable in line 136 as string dirlist = envdirs ? envdirs : "/usr/lib/opencpn:usr/lib64/opencpn";)

here is an extract of the debug log about o_charts plugin 

22:49:11.988 MESSAGE plugin_loader.cpp:446 Checking plugin candidate: /usr/lib64/opencpn/libo-charts_pi.so
22:49:11.988 MESSAGE plugin_loader.cpp:493 Checking plugin compatibility: /usr/lib64/opencpn/libo-charts_pi.so
22:49:12.002 MESSAGE plugin_loader.cpp:1357 Plugin is compatible: true
22:49:12.002 MESSAGE plugin_loader.cpp:1373 PluginLoader: Loading PlugIn: /usr/lib64/opencpn/libo-charts_pi.so
22:49:12.004 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /home/philippe/.local/share/opencpn/plugins;/home/philippe/.local/share/flatpak/exports/share/opencpn/plugins;/var/lib/flatpak/exports/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/lib64/opencpn
22:49:12.004 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /usr/share/opencpn/plugins/o-charts_pi
22:49:12.005   DEBUG plugin_loader.cpp:1474 blacklist: Get status for o-charts 2 0
22:49:12.006   DEBUG plugin_handler.cpp:352 Cannot parse API version ""

What does mean      Cannot parse API version ""   ?  something is missing but where ?

bdbcat, 2024-08-24:
If I understand your definition of "legacy"....
Looks to me like the logic is looking for the plugin metadata, which  does not exist for a legacy plugin install location.
Your prior comment is correct: "OpenCPN 5.10.0 doesn't like legacy plugins"

We shifted to user-space managed plugins some versions ago for good reasons.
Of course, O could be made to understand legacy plugins, if they were treated like the 4 "standard" plugins.
But that has never been implemented, and there are no plans to do so, at least in the OCPN mainline.

filochard, 2024-08-24:
Hi Dave and thanks again

I always install the plugins inside the same path as the bundled ones (dashboard grib chartdldr wmm)

here is the list of the plugins that I package as rpms and regularly update (with the last version I provide) : 

ais-radar-1.4.11.5
celestial-navigation-2.4.41.0
climatology-21.6.31
dashboardsk-0.2.0 (git version 2024-01-02)
iacfleet-0.32.0 (git version 2023-12-06)
logbookkonni-1.4.31.0
o-charts-2.0.6.0
objsearch-0.26.0
polar-1.2.31.0
radar-5.5.0
s63-1.30.0.11
sar-4.0.0
squiddio-1.3.91.0
statusbar1.1.12.0
watchdog-2.4.110
weather-routing-1.15.18.6
weatherfax-1.9.111.1

With a little patch inside plugin_paths.cpp and plugin_loader.cpp OpenCPN 5.8.4 can use them 
(creating /home/mysuername/.opencpn ) 

If I simply update OpenCPN 5.8.4 to OpenCPN 5.10.0 (with same kind of patches) OpenCPN 5.10.0 can use the same previously created 
/home/mysuername/.opencpn  
and all the plugins are still available and the configs and data are preserved (paths to charts, logbooks etc.)

BUT
If I do a fresh install of OpenCPN 5.10.0 with all the 17 plugins that are recommended the bug log shows that OpenCPN find them in the right place but can't use them : they are blacklisted !

If I understand your advice I should add my plugins names in the list inside plugin_loader.cpp so that OpenCPN 5.10.0 accept them

/** Return true if name is a valid system plugin name. */
static bool IsSystemPluginName(const std::string& name) {
  static const std::vector<std::string> kPlugins = {
      "chartdownloader", "wmm", "dashboard", "grib", "demo"};
  auto found = std::find(kPlugins.begin(), kPlugins.end(), ocpn::tolower(name));
  return found != kPlugins.end();

Am I right ?

NB
before this I only had to suppress this lines inside plugin_loader.cpp for OpenCPN 5.8.4 
(I did this for OpenCPN 5.10.0 too but that seems not to be enough)

if (loaded) return true;
 // Avoid loading/testing legacy plugins installed in base plugin path.
 wxFileName fn_plugin_file(file_name);
 wxString plugin_file_path =
     fn_plugin_file.GetPath(wxPATH_GET_VOLUME | wxPATH_GET_SEPARATOR);
  wxString base_plugin_path = g_BasePlatform->GetPluginDir();
 if (!base_plugin_path.EndsWith(wxFileName::GetPathSeparator()))
    base_plugin_path += wxFileName::GetPathSeparator();
  if (!g_bportable) {
    if (base_plugin_path.IsSameAs(plugin_file_path)) {
      if (!IsSystemPluginPath(file_name.ToStdString())) {
       DEBUG_LOG << "Skipping plugin " << file_name << " in "
                 << g_BasePlatform->GetPluginDir();
        return false;
      }
   }
  }

Beside adding my plugins to the list of valid system plugins do I need to suppress this lines too ?

filochard, 2024-08-24:
Thanks a lot Dave

These tricks functioned 

2 patches : 

1) for plugin_paths.cpp : 
replacing line 136
string dirlist = envdirs ? envdirs : "~/.local/lib/opencpn";
by
string dirlist = envdirs ? envdirs : "/usr/lib/opencpn:usr/lib64/opencpn";

2) for plugin_loader.cpp : 
line 112
adding my own plugins names to the list
and
suppressing lines from  486 to 504

I may now continue to build rpms of OpenCPN and its plugins, install them, use them,
And propose users to update smoothly the previous version or to do a fresh install

filochard, 2024-08-24:
PS
I perhaps found a little bug that will  need an other issue
I'm gonna verify inside flatpak if it exists too

filochard, 2024-08-24:
I can close this issue as resolved for me ...

for the ones interested the solution is here : 
https://github.com/OpenCPN/OpenCPN/issues/4111#issuecomment-2308384310

closed 2024-08-24 by filochard