OpenCPN/OpenCPN #4112: Chart downloader: fails on Peruvian charts

leamas, 2024-08-21:
**Describe the bug**
In the chart downloader tab, attempts to download  Peruvian charts fails with a dialog box saying "Error, the URL to the chart data https:/... seems wrong"

**To Reproduce**
Steps to reproduce the behavior:
1. In Options | Plugins enable the chart downloader
2. In Options | Charts click on the Char Downloader tab
3. Click Add Catalog
4. Select Chart Catalog | Peru RNC Nautical charts, click OK
5. In new window, click Update
6. Click "Download selected charts".
7. See dialog box

**Expected behavior**
The charts are downloaded

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 Discovered on Fedora 40, likely affects all platforms

**Additional context**
This is old code  which have been working. The problem is that the Peruvian charts URLs have a prefix like `https:/www.dhn.mil.pe` instead of the correct `https://www.dhn.mil.pe`.  Perhaps lately wxWidgets updates have become more picky and fails to recognize the ill-formed URL.

To make it more interesting the _bool wxURI::IsReference()_ method is [documented](https://docs.wxwidgets.org/3.2.5/classwx_u_r_i.html#af763650bde87684c07a981721567f2a4) as returning true if the url is an absolute URL, else false. The truth seem to be the opposite, this is what existing code seems to assume. In any case, I cannot understant the actual meaning of "Referenc" in this context

leamas, 2024-08-21:
OK. But will existing catalogs linger by users?

Sidenote: I note that the catalogs are stored un ~/Documents also on Linux. Seems strange, it's a Windows-style path (and outdated also there)

nohal, 2024-08-21:
I do not understand. Next time the user presses the Update button, he gets the new fixed catalog.

leamas, 2024-08-21:
Ok, verified. Closing

closed 2024-08-21 by leamas