OpenCPN/OpenCPN #4117: libdnet cmake build broken with Ninja generator

nohal, 2024-08-22:
After `cmake -G Ninja ..` it is impossible to build with `ninja` because of

```
ninja: error: 'libdnet-src/installed/lib/libdnet.a', needed by 'opencpn', missing and no known rule to make it
```

This can't be easily fixed with "We just support Makefiles" as Ninja is the default for C++/Cmake in VisualStudio Code.

bdbcat, 2024-08-22:
Yes, me too.  Exactly the same message. 
I have had to switch to makefiles to proceed in CLion.
Any idea what is going on?

leamas, 2024-08-22:
Not really, but I'm on it

leamas, 2024-08-22:
@nohal : I assume you got that message on Linux, right?

nohal, 2024-08-22:
The `_libdnet` target must be built before the rest, but for some reason it is not when the Ninja generator is used. If you build it explicitly with `ninja _libdnet`, stuff starts working.

nohal, 2024-08-22:
> @nohal : I assume you got that message on Linux, right?

Yes, on Linux

leamas, 2024-08-22:
A dependency is missing. Give me some time

PR #4118 (2024-08-22): libdnet: handle ninja FTBFS (#4117)
Closes: #4117
closed 2024-08-22 by leamas