OpenCPN/OpenCPN #4127: chartdldr: State stored in $HOME/Documents on Linux

leamas, 2024-08-30:
**Describe the bug**
The chart downloader stores state in *$HOME/Documents* on Linux and corresponding paths on other platforms. This is a non-standard location and breaks the general contract that opencpn stores it's state in the data directory, *$HOME/.opencpn* on Linux.

**To Reproduce**
Steps to reproduce the behavior:
1. Options | Charts | Chart Downloader
2. Download some charts
3. Find state files in *$HOME/Documents*

**Expected behaviour**
All state should be stored in the opencpn data directory, *$HOME/.opencpn* on Linux

**Additional context**
The reason is that chartdldr uses `GetWritableDocumentsDir()` to get a state directory. This method is just not meaningful for any OpenCPN purpose, `GetPrivateDataDir()` should be used instead.
 
`GetWritableDocumentsDir()`  is probably  a left over from Windows 7 which had a habit of storing state in Documents.  chartdldr  is the only usage of this method in core opencpn, but some plugin could of course use it.  For this reason the method should probably be marked as deprecated in *ocpn_plugin.h*.

nohal, 2024-08-30:
We have already discussed this a couple of days ago. No, I do not want to move the downloaded catalogs and `chartdldr_pi.dat` files away from the charts as it brings nothing and only will break the usecase where the charts are copied between machines.

nohal, 2024-08-30:
And no, I do not want to move the charts into the hidden data directory neither.

leamas, 2024-09-01:
Have I mentioned the charts directory here?  Cannot see that.

So indeed, the expected behaviour would be to store all state in  ~/.OpenCPN and the chart directory by default being ~/Charts on Linux.

But this sort of misses the point which is that vhartdldr stores state In ~/Documentsm

nohal, 2024-09-01:
What do you call "state"? If `chartdldr_pi.dat` and the downloaded catalogs, then they are exactly where designed and they are there for a reason.

Charts are also where designed and I see absolutely no reason to move them from the documents directory. Whoever does not want them there and prefers for example `~/Charts` can do so while adding the catalog.

As far as I'm concerned this is not a bug and I do not see anything that would justify the changes you ask for.

leamas, 2025-04-22:
I'll close this for 5.12. Reluctantly, but here are other things to do

closed 2025-04-22 by leamas