OpenCPN/OpenCPN #4128: "make package" is broken on Debian Bookworm (X86)

hreuver0183, 2024-09-01:
**Describe the bug**
When building OpenCPN from source most works up to "cmake .. && make -j8"
Building the package with "make package" fails.  

Last messages:
```
/home/user/opencpn-beta/OpenCPN-test/build/cmake_install.cmake:540 (include)

/usr/bin/strip: /usr/local/lib/opencpn/libchartdldr_pi.so: could not create temporary file to hold stripped copy: cause of error unknown
CMake Error at /home/hja/opencpn-beta/OpenCPN-test/build/plugins/dashboard_pi/cmake_install.cmake:60 (file):
  file INSTALL cannot copy file
  "/home/user/opencpn-beta/OpenCPN-test/build/plugins/dashboard_pi/libdashboard_pi.so"
  to "/usr/local/lib/opencpn/libdashboard_pi.so": Permission denied.
```

The commands which show the error:

```
git reset --hard  924f72c45ec772c7cbd5298c55c164ebd9beed0e
cd build && make -j 8 package 
```

The commits to introduce the behaviour are narrowed down to:
commit 924f72c45ec772c7cbd5298c55c164ebd9beed0e
commit a905126d03ad3c3b9ede759fdacfb680cff95ac5
commit 8fe5e2796ea7053050db41b0f85c5fccdbdd7a9d

**Expected behavior**
make package should end with an installeble opencpn package.
it also should not change files in /usr locations.
The reverting to an older commit still yields a working package:

```
git  reset --hard  9185b732aa99b45d2684c41d4aaa5597983a9fbd)
cd build && make -j 8 package
``` 

**Desktop (please complete the following information if applicable):**
 - OS: Debian GNU/Linux 12 (bookworm)
 - System: Linux 6.6.48 #1 x86_64 GNU/Linux (Laptop, i7)

hreuver0183, 2024-09-01:
Proposed fix:
Revert commit 924f72c45ec772c7cbd5298c55c164ebd9beed0e

If needed change the patch to mark the package as an unmaintained copy (see comment on the above commit).

leamas, 2024-09-01:
Well, since this change is deliberate it:s not an unexpected behaviour, it's working as designed. Thus this is not really a bug.

Also, OpenCPN certainly has no need of unmaintained code. On the contrary, to say the least.

I can see two solutions for your needs. One could be that you maintain a personal patch. If this is organized as a separate file in the cmake/ directory the only patch needed to CMakeLists.txt would be a single Include() line. This is maintainable.

The other would be that you build a package according to the manual.. Sorry, on phone, no link but look for Buildimg Debian Packages in core development. This is used in CI builds and is thus considered maintained

hreuver0183, 2024-09-01:
Then there might be an issue, since:
https://opencpn-manuals.github.io/main/ocpn-dev-manual/index.html
Returns "Page not found".

Please add a correct link to the developer page in that case (when you are able to, no real hurry).

leamas, 2024-09-01:
The debian packaging instructions are [here](https://opencpn-manuals.github.io/main/opencpn-dev/create-deb-package.html).

The overall start page is https://opencpn-manuals.github.io/main/manuals/index.html

hreuver0183, 2024-09-01:
Found it. Will test it.
The first local release was opencpn_5.8.10~alpha2+dfsg-1 which seems wrong (with current 5.10+). 
I probably have to reread the changelog part, perhaps change the checkout since I normally want the current github status instead of the latest official build.

In general a package was built, so it should work.

I'll close this issue.

closed 2024-09-01 by hreuver0183

leamas, 2024-09-01:
> I probably have to reread the changelog part, p

dch(1) is your friend

leamas, 2024-09-02:
@hreuver0183 

hm... in hindsight, the manual has been missing a section on your use case, we have been pretty busy with the stuff targeting actual releases so far.

Added a chapter [here](https://opencpn-manuals.github.io/main/opencpn-dev/create-deb-package.html#master-build). Would appreciate if you have time to review it. For anything found, just press the Edit This Page button top-right which puts you in an editor. When you save here, it will create a PR against the manual.

You can put also questions or remarks, not just proposed changes in the text if you want, I'll try to handle it.