OpenCPN/OpenCPN #4132: OpenCPN needs to support (generate) PGN's 129283 & 129284 when route is activated.

dagnall53, 2024-09-06:
The new N2000 ‘compatibiity’ has added the ability to ‘receive’ N2K in a number of ‘raw’ protocols.
However the program is currently horribly inconsistent in how it sends Autopilot Control messages when a route is activated:
The 0183 elements happily send XTE etc messages with the course to the WP on the 0183 communications channels.
But the N2000 (OUT) channels are silent!  This is inconsistent and dissapointing. They need to send 129283 129284 etc .. 
ALSO: When this is fixed, please make sure that the N2000 (OUT) are displayed in the NMEA debug window, in the same way that the 0183 messages OUT are displayed. This is essential for debugging!

bdbcat, 2025-03-23:
dag...
Can you give us some specific use-cases, including notes on your system configuration, that you would like to see in the next OpenCPN release?
Also, take a look at https://github.com/OpenCPN/OpenCPN/discussions/4414

dagnall53, 2025-03-23:
The ‘etc’ that confused you was 129285 that completes the set  192283,129284,129285 that send waypoint and nav data. 
If open cpn wishes to provide a routing using function to N2000 systems, then it will need to send this data so that the autopilots can do the necessary XTE, waypoint display etc. 

The much more important issue is the question of exactly what method OpenCPN will use to accept and transmit N2000 datagrams, as they need to be converted to one of the multitude of ‘raw’ formats in order to pass via WIfi or even serial. 
I think you have selected to use the ‘actisense raw’ format that is supported by the Timo Lappinen N2000 library, and if so, i think it is very important that you make this very clear in the OCPN documentation. 
If you ‘send N2000’, it is ONLY ever over CANbus. 
But if you are ‘sending/recieving N2000’ over wifi /serial, you are sending an ‘encoded’ / ‘raw’ version compatible with that transport medium.

My personal preference, as commented in previous N2000 comments would be to use one of the  ASCII readable RAW formats, as they are easier to read for debug and would be compatible wth other 0183 data streams.  
 
This means little to the user with no knowledge of the hardware, but for developers and supporters of wifi multiplexers such as myself, it is crucial to helping users. 

The original query, if I recall, was a summary of a user’s problem trying to send OCPN waypoints to a N2000 autiopilot. (Probably raymarine). When he switched to 0183, all worked. 

Purely FYI. My personal preference is always 0183, and my own older autopilot (raymarine) refuses to accept N2000 data, as I have the ST1 ap control box.

The other ‘major’ need in Open CPN N2000 support, is to identify in the debug monitor when OCPN is sending any “N2000” messages. This would be consistent with the 0183 behaviour, where output messages are clearly displayed. - Again, this will greatly assist debugging at a distance!