OpenCPN/OpenCPN #4134: AutoAnchorDrop should not delete other anchoring waypoints

svMigration, 2024-09-07:
**Describe the bug**
When you have AutoAnchorDrop set to 1 in the INI file, if there is another anchoring waypoint close by, it is automatically deleted.

**Expected behavior**
I know this is not a 'bug'. But I requested this be changed years ago as an enhancement.

Please remove this feature. It is wrong to have OCPN delete waypoints without confirmation. Just leave the nearby waypoints where they are. The user can delete them if they want.

Any chance this can be added to 10.2?

Thank you!
Bruce

svMigration, 2024-10-25:
Just to add an example of why this should be changed: 
We were recently anchored in an area surrounded by reef -- the only anchorage protected from nearly all directions in the Ha'apai group of Tonga. A gale was forecast. 3 other boats there. As the wind backed from the SSE to the NW, we found one of the boats has anchored too close to us and we werenow  directly upwind of them by a boat length. We chose to move. However, I wanted to come back to my same spot if there was a change in the wind. Unfortunately, our new spot was only .18 NM away and I forgot to turn off the GPS feed to OpenCPN before shutting down. So lost my previous spot.

svMigration, 2025-02-05:
A simpler solution would also be to just remove the autodelete function completely and _never_ delete nearby waypoints.

svMigration, 2025-02-06:
> Aqua Map Anchor Alarm Increases safety at anchor!
>
> The basic Anchor Alarm function provides real-time monitoring of your GPS and anchor positions, triggering an acoustic and visual alert when your position exceeds the predefined range.
>
> With the advanced AnchorLink™ feature, you can stay updated on the status of your anchorage, even while away from the boat, through mirroring and Telegram/Email notifications.
>
> https://www.youtube.com/watch?v=VT_s4hxUcq0
> "Revolutionary Smart Anchor integration.... Traditional anchor watch apps often suffer from issues like losing GPS signals and false alarms, but our integration eliminates these problems by utilizing the boat's high-quality GPS and 24/7 Smart Boat system. With this setup, you can monitor your anchored position onboard or remotely, and even trigger alarms through indoor sirens or the ship's horn, ensuring maximum safety and reliability.

HI Rick,

This is something I’m not interested in. Way more functionality than I want.

Thanks,
Bruce

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

rgleason, 2025-02-07:
Bruce, you might not want it, but others might want the ability to monitor their boat while ashore. I will remove it from the Issue

svMigration, 2025-02-07:
Hi Rick. You are absolutely correct. I do think there would be users happy with features like this. However, that functionality is completely separate from the issue in this thread (which addresses solving a problem in current functionality). 
Thanks!
B

bdbcat, 2025-02-07:
I think we can plan on implementing some sort of control for "nearby" waypoint deletion in the next Release.  
But I do not anticipate additional functionality, at least as response to this Issue.
But RFE's are always logged, and perhaps some new developer would take this on as a first project...

svMigration, 2025-02-08:
Thanks, Dave. That would be so great.
Bruce

rgleason, 2025-02-08:
Bruce,  you could hide or delete the following to clean this up.
https://github.com/OpenCPN/OpenCPN/issues/4134#issuecomment-2641355404
https://github.com/OpenCPN/OpenCPN/issues/4134#issuecomment-2641787377
https://github.com/OpenCPN/OpenCPN/issues/4134#issuecomment-2641787379
plus delete this message. 
I can't do it.

svMigration, 2025-02-09:
Hi Rick, I deleted the others. Can't delete your last post.
Cheers,
Bruce

svMigration, 2025-04-24:
### I think we can plan on implementing some sort of control for "nearby" waypoint deletion in the next Release

Thanks, Dave. I hope you can. Easy fix is just get rid of the autodelete.

svMigration, 2025-11-16:
Hi Dave, Did this ever get added? If so, how was it implemented? Thanks, Bruce