OpenCPN/OpenCPN #4135: 5.10 on macOS - crash when changing language

bcn2, 2024-09-08:
**Describe the bug**
OCPN will crash when changing the UI language
**To Reproduce**
Steps to reproduce the behavior:
Normal procedure:
Options|UserInterface --> Change language --> "ok"

We have seen that in the beta stage, thought that it had been solved.....

**Screenshots**
The dump:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenCPN [6993]
Path:                  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.10.0-universal+860222c02 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-08 15:05:33.5405 +0200
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        24089CE9-363A-CCBA-FEA0-0B62E0E783C4

Time Awake Since Boot: 93000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000794
Exception Codes:       0x0000000000000001, 0x0000000000000794

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [6993]

VM Region Info: 0x794 is not in any region.  Bytes before following region: 4307564652
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100c04000-101498000    [ 8784K] r-x/r-x SM=COW  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libocpn_draw_pi.dylib         	       0x12f558228 ODicons::initialize_images() + 32
1   libocpn_draw_pi.dylib         	       0x12f515ad4 ocpn_draw_pi::Init() + 1100
2   OpenCPN                       	       0x1010ed528 PluginLoader::UpdatePlugIns() + 844
3   OpenCPN                       	       0x100de1efc ApplyLocale() + 816
4   OpenCPN                       	       0x100dd78a8 MyFrame::DoOptionsDialog() + 3312
5   OpenCPN                       	       0x100dd3e08 MyFrame::DoSettings() + 48
6   OpenCPN                       	       0x100dc641c MyFrame::OnToolLeftClick(wxCommandEvent&) + 228
7   libwx_baseu-3.2.0.2.3.dylib   	       0x10264977c wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 376
8   libwx_baseu-3.2.0.2.3.dylib   	       0x10264a46c wxEvtHandler::ProcessEvent(wxEvent&) + 260
9   libwx_baseu-3.2.0.2.3.dylib   	       0x10264a59c wxEvtHandler::ProcessEventLocally(wxEvent&) + 212
10  libwx_baseu-3.2.0.2.3.dylib   	       0x10264a3a0 wxEvtHandler::ProcessEvent(wxEvent&) + 56
11  libwx_baseu-3.2.0.2.3.dylib   	       0x10264a188 wxEvtHandler::ProcessPendingEvents() + 240
12  libwx_baseu-3.2.0.2.3.dylib   	       0x102576c18 wxAppConsoleBase::ProcessPendingEvents() + 116
13  libwx_baseu-3.2.0.2.3.dylib   	       0x10262eed0 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 84
14  CoreFoundation                	       0x18ed4887c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
15  CoreFoundation                	       0x18ed48768 __CFRunLoopDoObservers + 536
16  CoreFoundation                	       0x18ed47d94 __CFRunLoopRun + 776
17  CoreFoundation                	       0x18ed47434 CFRunLoopRunSpecific + 608
18  HIToolbox                     	       0x1994f119c RunCurrentEventLoopInMode + 292
19  HIToolbox                     	       0x1994f0e2c ReceiveNextEventCommon + 220
20  HIToolbox                     	       0x1994f0d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
21  AppKit                        	       0x1925a6cc8 _DPSNextEvent + 660
22  AppKit                        	       0x192d9d4d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
23  AppKit                        	       0x192599ffc -[NSApplication run] + 476
24  libwx_osx_cocoau_core-3.2.0.2.3.dylib	       0x10304a758 wxGUIEventLoop::OSXDoRun() + 140
25  libwx_baseu-3.2.0.2.3.dylib   	       0x10262f54c wxCFEventLoop::DoRun() + 40
26  libwx_baseu-3.2.0.2.3.dylib   	       0x10259db70 wxEventLoopBase::Run() + 120
27  libwx_baseu-3.2.0.2.3.dylib   	       0x1025764bc wxAppConsoleBase::MainLoop() + 120
28  libwx_baseu-3.2.0.2.3.dylib   	       0x1025c8d34 wxEntry(int&, wchar_t**) + 104
29  OpenCPN                       	       0x100dac540 main + 24
30  dyld                          	       0x18e8df154 start + 2476

**Desktop (please complete the following information if applicable):**
 - OS: macOS 14.6.1

nohal, 2024-09-08:
It is the Draw plugin what crashes, not the main application.

bcn2, 2024-09-08:
Sorry for the noise...
Confirmed ok when disabling the Draw-PI

closed 2024-09-08 by bcn2