OpenCPN/OpenCPN #4143: Why is there no more content in /data/doc of master tree of OpenCPN 5.10.* except help_web.html

filochard, 2024-09-12:
Hi
You may now that I continue to package OpenCPN and its plugins as rpms (with some patches to access to these plugins)
(I am now testing 5.10.0 and 5.10.2 before providing them to the distribution)

When I created rpm for OpenCPN 5.8.4 ,  I  usually copied the whole content of master/data/doc/ so that it can be found inside : 
/usr/share/opencpn/doc/
This way we could access to the user's manual by simply clicking on "help" or on the F1 keyboard touch... that leaded to help_en_US.html and displayed the manual inside Firefox ... no need of an internet access !

Now for OpenCPN 5.10.* there's only help_web.html inside master/data/doc/   and no possibility to read the manual "onboard"  when we are without internet access...

Are you waiting for the manual been updated before providing it again (it had been written for OpenCPN 5.0) ?

leamas, 2024-09-12:
Since you are on a linux dist you cannot really bundle the manual even if it existed. The manual is generated stuff without usable sources, hence not open-source.   Toolchains like Fedora and Debian complains loudly if bundling html which is not generated as part of the build.

We are working on solutions like #3855 in this cycle.

filochard, 2024-09-12:
Thanks fo your quick answer
My bad
I didn't know this restriction for the manual and simply used it since it was present in the previous master tree

Nevertheless o-charts-plugin provides binaries that are also not open-source ....

For my distro we use a special nonfree repository were we provide this o-chats-plugin besides s63-plugin   
and inside this repo we provide lot of closed source stuff : nvidia drivers, microcodes, radeon-firmware, rlink-firmware, iwlwifi-firmware, and some kernel-nonfree-firmwares  that are necessary for lots of computers

Maybe I should have packaged OpenCPN in the free opensource repo without the bundled manual and in nonfree with this bundled manual

Post Scriptum
My two pence

 I wonder why to be so pernickety about html :  after all lot of stuff is not generated as part of the build (icons for instance, even if they have been created previously by an opensource program they are not generated as part of the build)
And each .html file of the manual can be explored in the package or in its source if you want to look for a pernicious line

Lots of softwares provide a bundled pdf guide that is not generated as part of the build (sometimes even created with Adobe !!!)

If the html manual is copyleft (and it can be created by an opensource program) I don't understand clearly why it should not be bundled  
(I personally use an opensource program to create my html site of photos, panoramas, and travel diaries but this needs to know a little html coding)

Hakansv, 2024-09-12:
filochard
Well formulated and it reflects my thoughts I have not been able to express.
To my understanding a HTML file is not a built binary, but an open text file that can be viewed and edited by anyone who wishes. Comparable to other text files within OpenCPN, such as CMakeLists.txt. The content of these HTML files is also created by members of our community and not copyrighted.

leamas, 2024-09-13:
> I wonder why to be so pernickety about html : after all lot of stuff is not generated as part of the build (icons for instance, even if they have been created previously by an opensource program they are not generated as part of the build)
And each .html file of the manual can be explored in the package or in its source if you want to look for a pernicious line

This is basically a discussion you need to pursue within your distribution community. That said, if I tried within Debian or Fedora I think I would meet arguments along these lines:
  - All open source licenses grants the user the right to obtain the sources, modify them and generate a new version of the licensed program.
  - Html is in this context handled more or less as compiled code. That is, user has a right to obtain the sources used to create the html, modify them and generate a new version of the docs.
  - To this end, linux distros generates html documentation as part of the build process from some kind of sources.
  - In opencpn, we generate some docs either from asciidoc using Antora or directly from the C++ source files  using Doxygen.   We don't distribute this, but we could without major problems.
  - However, the mediawiki instance used to generate the user manual don't even has the concept of sources. It's technically impossible to obtain the sources and generate a new version. The only way is to use the current mediawiki instance which is not controlled by the user. Case closed.

OTOH, the manual adds some weight to the package (10-20Mb IIRC). More important, we want to bundle also the plugin manual which is about 120Mb due to many images. We cannot handle this in the package, it's just too much.

So in this perspective it's a much better solution to make the manual(s) a separate download. This would also make it possible to have it more actual than now in the same way as other plugins.

> For my distro we use a special nonfree repository 

Indeed, there usually is something like that. But in most cases, including Debian, non-free program are sort of second class citizens. For example, non-free repositories are not always enabled by default for new users, etc.

Now, back to work with #3855.

filochard, 2024-09-13:
I know that Fedora or Debian are strict and don't provide non-free programs and that someone needs to use external repo for those needed programs
For instance Debian can't provide o-charts-plugin and a simple user couldn't use it (Fedora is no more concerned since they haven't build rpms anymore for years )

but Flatpak allows to use this plugin on a Debian or Fedora system !!!

Nevertheless I understand that a separate plugin for the manuals would be a good way to provide it or to easily update it

I disagree a little about an html manual not being open source even if it has not been generated by build : 
It should be considered as a source code that you can explore and modify  with an opensource program (such as Bluefish for instance if you know html coding) or Dokuwiki itself :  the only limit would be an existing copyright

a pdf manual wouldn't be considered as opensource anymore ...

Out of topic about Flatpak : 
 I have updated OpenCPN from 5.10.0 to 5.10.2 inside Flatpak (40Mo) 
but I have had to update Flatpak itself too : 800Mo to download : Freedesktop Platform, Mesa, Mesa(extra), org.freedesktop.Platform.GL.nvidia
Flatpak needs to be regularly updated (4 or 5 times a year) : it's a very heavy way to use OpenCPN

That's the reason why I continue to build rpms ! (using our system libraries that are regularly updated when a CVE appears for them) and test each new build (5.10.0 and then 5.10.2)

nohal, 2024-09-13:
Every system with a sane user has to be regularly updated, much more often than 4-5 times a year actually. How big the RPM update will be if you leave your system without updates for 3 or 4 months?

filochard, 2024-09-13:
> Every system with a sane user has to be regularly updated, much more often than 4-5 times a year actually. How big the RPM update will be if you leave your system without updates for 3 or 4 months?

Hi Pavel
The problem is I have already to update my distribution ... and that's what I do regularly, then updating OpenCPN installed with rpm needs only 40Mo to download...
For Flatpak, besides the updates of my distro that I do regularly, I have to regularly update Flatpak itself too : a useless duplicate of lots of things that I wouldn't need if I didn't test Flatpak version of OpenCPN and only use my rpm version...

Flatpak is like using a truck to carry a basket of eggs

nohal, 2024-09-13:
If you wouldn't use a computer, you would not have to download anything. These arguments are simply complete nonsense.

leamas, 2024-09-13:
>  (Fedora is no more concerned since they haven't build rpms anymore for years )

 This is new, fascinating info for me who actually maintain some Fedora rpms?

> I disagree a little about an html manual not being open source even if it has not been generated by build :
It should be considered as a source code that you can explore and modify with an opensource program (such as Bluefish for instance if you know html coding) or Dokuwiki itself : the only limit would be an existing copyright

As I said, this is a discussion you need to pursue within your distro community.  What **you** think is not really the issue here, it's a question of distro policies.

>  a pdf manual wouldn't be considered as opensource anymore ...

Could be just fine, the are many source  formats from which it is possible to generate pdf. But without source, how could it be open source?

> If you wouldn't use a computer, you would not have to download anything. These arguments are simply complete nonsense.

+1

filochard, 2024-09-13:
> > (Fedora is no more concerned since they haven't build rpms anymore for years )
> 
> This is new, fascinating info for me who actually maintain some Fedora rpms?
> 

Sorry I was not clear enough I meant Fedora doesn't build OpenCPN rpms anymore for years 

> > If you wouldn't use a computer, you would not have to download anything. These arguments are simply complete nonsense.
> 
> +1
I have been using  a GNU-Linux computer for more than 35 years, building rpms for 20 years...
I don't mind downloading updates or upgrades for my distro (I use iso copies to upgrade several computers limitating this way the weight of downloads)
What I meant is that updating Flatpak for each computer consume lot of downloading that don't bring anything more that what I have already in my distribution : these heavy duplicates are only necessary to use a program inside Flatpak, when the same program would be used  after a simple package installation
That's here where I find a nonsense : needing more and more memory, more and more useless GigaOctets downloaded
That's why I continue to build packages for OpenCPN and its plugins for my distro

Post Scriptum
if you are interested I tried and succeed to build rpms of OpenCPN for Fedora with mock... the spec file and patches are available (adapted for Fedora that needs more BuildRequires than my distro where some of them are implicit !)

filochard, 2024-09-13:
> As I said, this is a discussion you need to pursue within your distro community. What **you** think is not really the issue here, it's a question of distro policies.

No problem since we have a nonfree repo I can create 2 rpms of OpenCPN 5.8.4 : 
one free without manual
one nonfree with the manual

But for now OpenCPN 5.10.* will be packaged without manual
you may consider this as closed until an OpenCPN-user-manual-plugin exist

filochard, 2024-09-13:
This may continue on
https://github.com/OpenCPN/OpenCPN/issues/3855

closed 2024-09-13 by filochard