OpenCPN/OpenCPN #4144: Viewport incorrectly displays land coverage when panning South Pacific with OpenGL disabled"

sebastien-rosset, 2024-09-14:
**Describe the bug**
The viewport displays 100% land when OpenGL is disabled, charts are zoomed out over the Pacific and the user performs panning towards the South Pacific.

**To Reproduce**
Steps to reproduce the behavior:

1. Disable OpenGL in advanced settings
2. Center the chart around 30N, 105W. Set the scale to about 1:32.5M
3. Slowly start panning South
4. When the chart is centered around 45S, 105W, **the entire viewport displays an area as if it were entirely covered by land, even though it's mostly the South Pacific.**
5. Keep panning South.
6. When the chart is centered around 75S, 105W, the view port shows expected sea and land areas.

Repeat the same steps, but now with OpenGL enabled: no problem occurs.

**Expected behavior**
When panning south, the viewport should not suddenly render the Pacific Ocean as if it were entirely covered by land.

**Screenshots**
See attached screen recording.

**Desktop (please complete the following information if applicable):**
 - OS: macOS.
 - Version: OpenCPN version 5.10

**Additional context**

leamas, 2024-09-24:
@sebastien-rosset : Which charts are you using here, and how did you acquire them?

nohal, 2024-09-24:
It is the basemap, you don't need any charts.

sebastien-rosset, 2024-09-24:
Yes, I confirm this is just with the base map.

leamas, 2025-05-03:
I can reproduce this on Linux.

There are other ways to make this happen. One is to center on Australia, around 30° S, 128°W. Then just zoom out. Checking around, it seems like it happens when centered south of the equator, otherwise not.