OpenCPN/OpenCPN #4150: 10.2 MBTiles/Chart Groups/Scrolling "Unhandled unkown exception; terminating the thread in which it happened"

svMigration, 2024-09-25:
**Describe the bug**
While scrolling over large areas with MBTiles displayed in a chart group, sometimes a **"Unhandled unkown exception; terminating the thread in which it happened**" error is displayed. 

[image]

After this happens, the MBTiles in the chart group currently selected will no longer update. Video uploaded below. (Sorry for the poor quality but had to reduce due to poor bandwidth.) Other charts in different chart groups display fine although in the video example, the ArcGIS MBtiles would not display for some reason.

OpenCPN does not crash. However, it does have a tendency to give the same error again during that session.

This has happened several times but the exact steps to reproduce are difficult to determine. 

https://github.com/user-attachments/assets/5edb910c-cc7e-4b87-b120-8678c6f82e0f

**Desktop (please complete the following information if applicable):**
Edition	Windows 11 Pro
Version	22H2
Installed on	‎4/‎6/‎2023
OS build	22621.4169

Lenovo Laptop Slim 7 Pro
Processor	AMD Ryzen 7 6800HS Creator Edition   3.20 GHz
Installed RAM	32.0 GB (27.7 GB usable)
System type	64-bit operating system, x64-based processor

svMigration, 2024-09-25:
It seems that OpenCPN may not have closed down properly as I found this crash report later in the day after the error message above had been received. 
[a2dbeda1-85c4-46d3-a42c-6f5ecc02b6cb.zip](https://github.com/user-attachments/files/17138034/a2dbeda1-85c4-46d3-a42c-6f5ecc02b6cb.zip)

leamas, 2025-05-03:
@svMigration : the mbtiles code has been rewritten in the 5.12 cycle. Could you possibly make a new test using current master to check if problem still persists?

svMigration, 2025-05-05:
Hi @leamas, Excuse my ignorance, but I'm not a developer and don't know how to download the current master. 
I just know how to report bugs on Github!
Thanks,
Bruce

leamas, 2025-07-18:
@svMigration 

This is supposed to be fixed in 5.12. Could you please report back here after updating, hopefully so we can close this?

svMigration, 2025-07-18:
Hi @leamas, I just downloaded 5.12. Definitely appears more stable when scrolling across large areas of MBTiles. However, it did crash during large area scrolling when displaying the CM93 chart group. I didn't get the unhandled exception... just a crash with a crash report.

[02fa980a-986c-462f-84e0-de99bcabf1e2.zip](https://github.com/user-attachments/files/21323253/02fa980a-986c-462f-84e0-de99bcabf1e2.zip)

svMigration, 2025-07-18:
Another report. This time with an MBTile. But it took a long time to get it to crash and a lot of switching between chart groups (4 MBTiles groups are set up and 1 CM93 group)

[77b2d56c-617c-4d29-8813-4a86310c42e4.zip](https://github.com/user-attachments/files/21323273/77b2d56c-617c-4d29-8813-4a86310c42e4.zip)

closed 2025-11-16 by svMigration