OpenCPN/OpenCPN #4151: 10.2 Crash when panning CM93 (possibily related to #4150 and #4124)

svMigration, 2024-09-25:
Crash when panning from the south pacific to the Caribbean with CM93 charts visible. This happened after doing some panning in the South Pacific with MBtiles in a chart group.
May be related to issues #4150 & #4124.

**To Reproduce**
Cannot reproduce on demand

CRASH REPORT:
[1b452f61-8ee2-4d3e-842a-0377ff6d1a7f.zip](https://github.com/user-attachments/files/17138398/1b452f61-8ee2-4d3e-842a-0377ff6d1a7f.zip)

svMigration, 2024-09-25:
Just happened again. Same scenario. 
[fd6450d4-3e75-425a-99ec-2a49543bcc90.zip](https://github.com/user-attachments/files/17139011/fd6450d4-3e75-425a-99ec-2a49543bcc90.zip)
And again:
[37005377-8781-467e-98c7-cca8210c96b5.zip](https://github.com/user-attachments/files/17139580/37005377-8781-467e-98c7-cca8210c96b5.zip)

svMigration, 2024-09-25:
And again. 
[e588fd5b-fa39-457a-b759-5dc0b7abdb6f.zip](https://github.com/user-attachments/files/17139621/e588fd5b-fa39-457a-b759-5dc0b7abdb6f.zip)

svMigration, 2024-09-25:
Almost reproducible. Same configuration at it #4124. If I switch between chart groups and zoom out to load the larger Pacific MBTiles file from different sources, and then switch to CM93 and pan around, I get a crash or the message in #4150 .
[df79e72b-b009-417e-9eef-cee376c9222b.zip](https://github.com/user-attachments/files/17139658/df79e72b-b009-417e-9eef-cee376c9222b.zip)

svMigration, 2024-09-26:
Reproducible. Video and 2 crash reports below.
[ff0ba739-cbc4-4c79-bf1c-43397c078e7d.zip](https://github.com/user-attachments/files/17145381/ff0ba739-cbc4-4c79-bf1c-43397c078e7d.zip)

https://github.com/user-attachments/assets/bffe7ed0-fac8-4df9-89fd-f70585297a50

bc38, 2024-09-26:
You probably have a miracle of graphics from Intel, try updating the driver directly through the Intel utility.
The video driver that Microsoft installs is kind of buggy, I've seen this on Intel and AMD.

svMigration, 2024-09-26:
No Intel Processors:
AMD Ryzen 7 6800HS Creator Edition 3.20 GHz 
AMD Radeon
NVIDIA GeForce RTX 3050 Graphics Cards

svMigration, 2024-09-28:
>Does only happen with some MBTiles displayed in the chart group?
How about with just cm93 displayed?

Hi Dave,
The video was done with ONLY CM93 charts displayed. Never selected the chart group that had the MBTiles in it.
B

bdbcat, 2024-10-01:
Analyzed the crash dumps.
All indicated the same trouble, some fault in the ATI Graphics driver for OpenGL.  Looks like something inside "atioglxx.dll".
Does this driver file look at all familiar?
Anyway, unlikely to be related to previous MBTile faults.  Completely different signature.

svMigration, 2024-10-01:
Hi Dave,

Glad to hear that it has nothing to do with the MBTiles problem.

That driver is in my AMD drivers folder. 

C:\DRIVER\AMD\source\Packages\Drivers\Display\WT6A_INF\B382442

That’s all I know. Happy to test something if necessary.

B

rgleason, 2024-10-01:
BCN wrote:  try updating the driver directly through the.. [manufacturer].
The video driver that Microsoft installs is kind of buggy, I've seen this on Intel and AMD.

svMigration, 2024-10-01:
Hi Rick,
The laptop has a AMD Ryzen processor and a NVIDIA® GeForce RTX™ 3050 graphics chip. The dll in question is from AMD. I don't know enough about OpenGL to know if it is handled by the graphics chip or the main processor. I'm a bit gun-shy to update drivers right now as we are in Tonga with our passage to New Zealand coming up soon and I don't want to mess up my computer if things go wrong. 
There's an Adrenalin software download from AMD that is supposed to check the drivers and I could try that though it makes me a bit nervous.
Any thoughts?
Thanks,
Bruce

rgleason, 2024-10-03:
I don't know AMD Ryzan but I am familiar with older Nvidia Geforce and our current version
NVIDIA GeForce RTX 2060 with Max-Q Design which is on a Dell Computer

Do you have the old drivers saved somewhere?  If so, it should be possible to revert back, but don't do anything you aren't confident about.

svMigration, 2024-10-05:
Thanks, Rick. I'll have to wait until NZ to update drivers. But happy to do any other testing while I have bandwidth.
B

svMigration, 2024-10-05:
Would adjusting 
[Settings] .....
GPUMemorySize=nnn
affect this at all?
[image]

rgleason, 2024-10-07:
Bruce, I wonder if testing your opengl would give some perspective?
I've used https://www.realtech-vr.com/home/?page_id=142  GLview a number of times to resolve questions and problems.

svMigration, 2024-10-13:
Rick, Will test as directed. 

Dave,
A couple more crashes tonight with CM93. Not even panning very far... just within one island group of Tonga and not zooming out. Same problem? Reports below.

Thanks,
Bruce
[ea50f6b1-8625-4cbe-972d-257428c6cbbe.zip](https://github.com/user-attachments/files/17352804/ea50f6b1-8625-4cbe-972d-257428c6cbbe.zip)

svMigration, 2024-10-13:
Another crash. This time with MBTiles showing after doing a lot of panning and close-in zooming with several groups of MBTiles and CM93. 
[329dd7ec-0bb4-4c36-a24d-d99ed38b2826.zip](https://github.com/user-attachments/files/17357126/329dd7ec-0bb4-4c36-a24d-d99ed38b2826.zip)