OpenCPN/OpenCPN #4158: Tracks rendered over AIS targets

nohal, 2024-10-01:
When there are many tracks, vessel can be completely hidden by them. Rendering order should be swapped. 
[image]

Hakansv, 2024-10-04:
I could reproduce this on Win for persistent tracks.

sebastien-rosset, 2024-12-01:
On MacOS, I see the vessels are on top of the tracks, at least with the settings I have. I've used a NMEA simulator for this experiment. I don't think it should matter for the ordering.

[image]

sebastien-rosset, 2024-12-04:
On my  Linux system I also see the AIS targets rendered on top of the tracks. 
[image]