OpenCPN/OpenCPN #4159: Option to calculate COG & SOG from position changes does not work.

TwoCanPlugIn, 2024-10-02:
Seems like another regression (or a feature that never worked).

Selecting the option to calculate COG & SOG when position is changing, does not work. (Only tested on Windows with OpenCPN 5.10)

Discovered whilst researching the following cruisers forum [post](https://www.cruisersforum.com/forums/f134/sog-and-cog-289676.html)

Some sample data (note the RMC sentence omits the COG & SOG fields)

$GPGGA,144533,3830.00,N,14448.00,E,1,10,0.98,1113.0,M,-21.3,M,25,10*6A
$GPGLL,3830.00,N,14448.00,E ,144533.00,A,A*48
$GPRMC,144533.68,A,3830.00,N,14448.00,E,,,021024,8.5,E,A*32
$GPGGA,144534,3830.00,S,14448.01,E,1,10,0.98,1113.0,M,-21.3,M,25,10*71
$GPGLL,3830.00,S,14448.01,E ,144534.00,A,A*53
$GPRMC,144534.68,A,3830.00,S,14448.01,E,,,021024,8.5,E,A*29
$GPGGA,144535,3829.99,S,14448.01,E,1,10,0.98,1113.0,M,-21.3,M,25,10*78
$GPGLL,3829.99,S,14448.01,E ,144535.00,A,A*5A
$GPRMC,144535.69,A,3829.99,S,14448.01,E,,,021024,8.5,E,A*21
$GPGGA,144536,3829.99,S,14448.02,E,1,10,0.98,1113.0,M,-21.3,M,25,10*78
$GPGLL,3829.99,S,14448.02,E ,144536.00,A,A*5A
$GPRMC,144536.70,A,3829.99,S,14448.02,E,,,021024,8.5,E,A*29
$GPGGA,144537,3829.98,S,14448.02,E,1,10,0.98,1113.0,M,-21.3,M,25,10*78
$GPGLL,3829.98,S,14448.02,E ,144537.00,A,A*5A
$GPRMC,144537.70,A,3829.98,S,14448.02,E,,,021024,8.5,E,A*29
$GPGGA,144538,3829.98,S,14448.03,E,1,10,0.98,1113.0,M,-21.3,M,25,10*76
$GPGLL,3829.98,S,14448.03,E ,144538.00,A,A*54
$GPRMC,144538.73,A,3829.98,S,14448.03,E,,,021024,8.5,E,A*24
$GPGGA,144539,3829.98,S,14448.03,E,1,10,0.98,1113.0,M,-21.3,M,25,10*77
$GPGLL,3829.98,S,14448.03,E ,144539.00,A,A*55
$GPRMC,144539.74,A,3829.98,S,14448.03,E,,,021024,8.5,E,A*22
$GPGGA,144540,3829.97,S,14448.04,E,1,10,0.98,1113.0,M,-21.3,M,25,10*71
$GPGLL,3829.97,S,14448.04,E ,144540.00,A,A*53
$GPRMC,144540.74,A,3829.97,S,14448.04,E,,,021024,8.5,E,A*24
$GPGGA,144541,3829.97,S,14448.04,E,1,10,0.98,1113.0,M,-21.3,M,25,10*70
$GPGLL,3829.97,S,14448.04,E ,144541.00,A,A*52
$GPRMC,144541.74,A,3829.97,S,14448.04,E,,,021024,8.5,E,A*25
$GPGGA,144542,3829.96,S,14448.05,E,1,10,0.98,1113.0,M,-21.3,M,25,10*73
$GPGLL,3829.96,S,14448.05,E ,144542.00,A,A*51
$GPRMC,144542.76,A,3829.96,S,14448.05,E,,,021024,8.5,E,A*24
$GPGGA,144543,3829.96,S,14448.05,E,1,10,0.98,1113.0,M,-21.3,M,25,10*72
$GPGLL,3829.96,S,14448.05,E ,144543.00,A,A*50
$GPRMC,144543.78,A,3829.96,S,14448.05,E,,,021024,8.5,E,A*2B
$GPGGA,144544,3829.96,S,14448.06,E,1,10,0.98,1113.0,M,-21.3,M,25,10*76
$GPGLL,3829.96,S,14448.06,E ,144544.00,A,A*54
$GPRMC,144544.81,A,3829.96,S,14448.06,E,,,021024,8.5,E,A*29
$GPGGA,144545,3829.95,S,14448.06,E,1,10,0.98,1113.0,M,-21.3,M,25,10*74
$GPGLL,3829.95,S,14448.06,E ,144545.00,A,A*56
$GPRMC,144545.83,A,3829.95,S,14448.06,E,,,021024,8.5,E,A*29
$GPGGA,144546,3829.95,S,14448.07,E,1,10,0.98,1113.0,M,-21.3,M,25,10*76
$GPGLL,3829.95,S,14448.07,E ,144546.00,A,A*54
$GPRMC,144546.83,A,3829.95,S,14448.07,E,,,021024,8.5,E,A*2B
$GPGGA,144547,3829.95,S,14448.07,E,1,10,0.98,1113.0,M,-21.3,M,25,10*77
$GPGLL,3829.95,S,14448.07,E ,144547.00,A,A*55
$GPRMC,144547.86,A,3829.95,S,14448.07,E,,,021024,8.5,E,A*2F
$GPGGA,144548,3829.94,S,14448.08,E,1,10,0.98,1113.0,M,-21.3,M,25,10*76
$GPGLL,3829.94,S,14448.08,E ,144548.00,A,A*54
$GPRMC,144548.88,A,3829.94,S,14448.08,E,,,021024,8.5,E,A*20
$GPGGA,144549,3829.94,S,14448.08,E,1,10,0.98,1113.0,M,-21.3,M,25,10*77
$GPGLL,3829.94,S,14448.08,E ,144549.00,A,A*55
$GPRMC,144549.88,A,3829.94,S,14448.08,E,,,021024,8.5,E,A*21
$GPGGA,144550,3829.93,S,14448.09,E,1,10,0.98,1113.0,M,-21.3,M,25,10*79
$GPGLL,3829.93,S,14448.09,E ,144550.00,A,A*5B
$GPRMC,144550.90,A,3829.93,S,14448.09,E,,,021024,8.5,E,A*26
$GPGGA,144551,3829.93,S,14448.09,E,1,10,0.98,1113.0,M,-21.3,M,25,10*78
$GPGLL,3829.93,S,14448.09,E ,144551.00,A,A*5A
$GPRMC,144551.92,A,3829.93,S,14448.09,E,,,021024,8.5,E,A*25
$GPGGA,144552,3829.93,S,14448.10,E,1,10,0.98,1113.0,M,-21.3,M,25,10*73
$GPGLL,3829.93,S,14448.10,E ,144552.00,A,A*51
$GPRMC,144552.95,A,3829.93,S,14448.10,E,,,021024,8.5,E,A*29
$GPGGA,144553,3829.92,S,14448.10,E,1,10,0.98,1113.0,M,-21.3,M,25,10*73
$GPGLL,3829.92,S,14448.10,E ,144553.00,A,A*51
[... 230 lines cut ...]
$GPGLL,3829.62,S,14448.49,E ,144712.00,A,A*55
$GPRMC,144712.24,A,3829.62,S,14448.49,E,,,021024,8.5,E,A*27
$GPGGA,144713,3829.62,S,14448.49,E,1,10,0.98,1113.0,M,-21.3,M,25,10*76
$GPGLL,3829.62,S,14448.49,E ,144713.00,A,A*54
$GPRMC,144713.25,A,3829.62,S,14448.49,E,,,021024,8.5,E,A*27
$GPGGA,144714,3829.61,S,14448.50,E,1,10,0.98,1113.0,M,-21.3,M,25,10*7A
$GPGLL,3829.61,S,14448.50,E ,144714.00,A,A*58
$GPRMC,144714.25,A,3829.61,S,14448.50,E,,,021024,8.5,E,A*2B
$GPGGA,144715,3829.61,S,14448.50,E,1,10,0.98,1113.0,M,-21.3,M,25,10*7B
$GPGLL,3829.61,S,14448.50,E ,144715.00,A,A*59
$GPRMC,144715.28,A,3829.61,S,14448.50,E,,,021024,8.5,E,A*27
$GPGGA,144716,3829.60,S,14448.51,E,1,10,0.98,1113.0,M,-21.3,M,25,10*78
$GPGLL,3829.60,S,14448.51,E ,144716.00,A,A*5A
$GPRMC,144716.30,A,3829.60,S,14448.51,E,,,021024,8.5,E,A*2D
$GPGGA,144717,3829.60,S,14448.51,E,1,10,0.98,1113.0,M,-21.3,M,25,10*79
$GPGLL,3829.60,S,14448.51,E ,144717.00,A,A*5B
$GPRMC,144717.30,A,3829.60,S,14448.51,E,,,021024,8.5,E,A*2C
$GPGGA,144718,3829.60,S,14448.52,E,1,10,0.98,1113.0,M,-21.3,M,25,10*75
$GPGLL,3829.60,S,14448.52,E ,144718.00,A,A*57
$GPRMC,144718.31,A,3829.60,S,14448.52,E,,,021024,8.5,E,A*21
$GPGGA,144719,3829.59,S,14448.52,E,1,10,0.98,1113.0,M,-21.3,M,25,10*7E
$GPGLL,3829.59,S,14448.52,E ,144719.00,A,A*5C
$GPRMC,144719.31,A,3829.59,S,14448.52,E,,,021024,8.5,E,A*2A

[image]

TwoCanPlugIn, 2024-10-02:
Github screwed up the NMEA data.

Instead, find it in the attached document.
[cogsog-err.txt](https://github.com/user-attachments/files/17229734/cogsog-err.txt)

nohal, 2024-10-02:
Duplicate of https://github.com/OpenCPN/OpenCPN/issues/4157