OpenCPN/OpenCPN #4161: Android. Some Dashboard preferenses are not saved to config.

Hakansv, 2024-10-06:
**Describe the bug**
When changing any preference for a present Dashboard it's not saved to config and consequently not recreated after a O reboot.
Also mentioned on [this CF post](https://www.cruisersforum.com/forums/f134/5-10-02-android-13-and-dashboard-289449.html#post3935311)

**To Reproduce**
Example to reproduce the behavior:
1. In Dashboard open preferences.
2. Create a new dashboard with some instruments and enable "Show this Dashboard"
3. Disable "Show this Dashboard" for the old dashboard.
4. Click OK and the above changes are implemented.
5. Stop OCPN by a double click on the Android Back button. 
6. Start O and open Dashboard. 
7. The old Dashboard is again viewed.
8. Open Dashboard preferences
9. The new Dashboard is actually saved to config and is still there but disabled and the old Dashboard is again enabled. 
    The same for any similar changes to a Dashboard like fonts. 
10. Also if a Dashboard is viewed when O is closed it's not viewed on next start. One has to again enable Tools->Dashboard.

**Expected behavior**
Change of preferences for a present Dashboard would be saved to config.
If Dashboard is used when O is closed it would be present again at next start.

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Tested on several
 - Android version: Tested on A11 and A13
 - Using OCPN 5.10.2[111]

bdbcat, 2024-12-01:
Hakan...
I can see the same effect on linux desktop, but only if you turn OFF the dashboards  (by toolbar click), before exiting OCPN.  In this case, "active" and "inactive" dashboards change places.

If you have the chance, please check this also on Windows.

Hakansv, 2024-12-01:
Dave..
Yes, it's the same on Windows.
But if I change which panels are active using the right-click menu, and not with preferences, then close the Dashboard with tools menu, selected panels are saved for the next O restart.

Hakansv, 2025-05-29:
All solved by adding: "pConf->Flush();" to "SaveConfig()". Most probably also for Android.
Me PR with fix?