OpenCPN/OpenCPN #4166: 5.10.2 - KDE - Wayland - Arch Linux - black window with OpenGL enabled

siborg, 2024-10-12:
**OpenGL under Wayland in Arch producing black application window**
OpenCPN v5.10.2 runs but shows black window (screenshot attached) when run on desktop Arch Linux in KDE 6.2 DE and nvidia proprietary driver. No errors being produced in the console window. Disabling opengl acceleration makes the charts visible again(2nd screenshot). 

When running in X11 session opengl acceleration works with no issues. 

**To Reproduce**
Just start the app in wayland environment 

**Expected behavior**
Charts are visible. 

**Screenshots**
Wayland: [image]
OpenGL Disabled(started with -G):  [image]

**Desktop:**
 - OS: Arch Linux
 - Version Rolling
 - KDE: 6.2
 - Nvidia: 560.35.03

nohal, 2024-10-12:
Does any OpenGL software actually work correctly on your system?

siborg, 2024-10-12:
Yes. all working correctly. Issue limited to OpenCPN

Very Best Regards,

Szymon Kosecki

sa-mendez, 2024-12-13:
Ok, I am glad I am not the only one with this issue.  

I have the same setup:

- Kernel Version: 6.12.4-arch1-1
- KDE Version:  6.2.4
- Nvidia Driver Version: 565.77

I can get around this on Wayland, by disabling OpenGL in OpenCPN, but that makes moving around the charts really slow, though usable.

I did also try switching from the Wayland session to X11 session and using OpenGL worked!

So something about OpenCPN with Xwayland is probably to blame here (or maybe even KDE's Kwin).

Given that there are workarounds, and Xwayland is supposed to be a stop gap since it seems OpenCPN is eventually moving to Wayland:  #4013, I dont know if this is worth digging into very deeply

sa-mendez, 2025-02-14:
FYI, after a recent updates, I am no longer having this issue.  OpenCPN works fine with OpenGL acceleration.  Perhaps it was a recent update to xwayland, or one of my other components.  Here is what I have right now that is working.

- Kernel Version: 6.13.2-arch1-1
- KDE Version: 6.3 
- Nvidia Driver Version: 570.86.16
- xorg-xwayland Version 24.1.5-1 (Arch package version)

Guenael, 2025-03-02:
Quick workaround for Archlinux users (it don't solve this issue, and @siborg already found a alternative), you can restore the X11 compatibility by changing your GDM profile:
Edit `/etc/gdm/custom.conf` and set `WaylandEnable` to `false`.

```
[daemon]
# Uncomment the line below to force the login screen to use Xorg
WaylandEnable=false
```

leamas, 2025-04-29:
@crstmkt : hm... perhaps the nvidia driver is the prime suspect. Their wayland support is quite new, there various reports on it.

Have you tried to unistall the nvidia driver and thus using nouveu?

Sitting on Fedora 41/Wayland/nouveau  which works without obvious problems. Waiting for a new graphics card, my current is too old to have working nvidia support.

transmitterdan, 2025-04-30:
For some time I've been searching in the background for a pernicious bug in OpenGL mode which I see on a new Windows 11 computer. The main parts of the canvas area does not update. When O starts, it writes to the canvas areas ok. But shortly after launch the updates stop. O is still writing to the frame buffers and is oblivious to the fact the canvas is "frozen". The menus work, setting panel works, as does the Windows managed parts of the frame. But the toolbar and chart canvas areas do not update. I have tried many other OpenGL programs that do not have this problem. All GL test programs work. 

Another "weird" factoid is that about once in every 20 runs of O the OpenGL mode will work flawlessly. 

I'm not saying this behavior is definitely related to this issue but would not be surprised if it is. I believe O is not doing something quite right with OpenGL. It could be wxWidgets but I doubt.

I'm waiting until new release before submitting PRs that speed up non-OpenGL mode for some activities. On my Windows 11 computer I cannot tell the performance difference between OpenGL on/off. But for a Pi that's probably not going to be the case.

sa-mendez, 2025-05-07:
@leamas: I am convinced that it is definitely the nvidia driver.  I am using the flatpak version of opencpn, and I have other flatpaks that rely on opengl and the flatpak nvidia driver.  

Sometimes, I would update the other flatpaks causing the flatpak nvidia driver to also be updated.  I would then launch opencpn, and get the black screen forcing me to revert to disabling accelerated opengl.

I finally realized that this was due to me having updated the nvidia driver as a result of updating my other flatpak.  When I reverted the nvidia driver to the previous version, opencpn was working with accelerated opengl again.

The flatpak version of the opengl driver that works for me right now is: org.freedesktop.Platform.GL.nvidia-570-124-04

siborg, 2025-09-03:
I am still having these issues. Does not matter which version of nVidia drivers I would use the OpenCPN window is always black. Only way to use it is to disable OpenGL with -G command line switch. Unfortunately some chart types require OpenGL to be displayed e.g. MBTiles so it is not feasible.  

`{
   "GL_EXT_framebuffer_object" : 1,
   "GL_RENDERER" : "NVIDIA GeForce RTX 4080 SUPER/PCIe/SSE2",
   "GL_ARB_texture_non_power_of_two" : 1,
   "GL_SHADING_LANGUAGE_VERSION" : "4.60 NVIDIA",
   "GL_ARB_texture_rectangle" : 1,
   "GL_EXTENSIONS" : [
      "GL_AMD_multi_draw_indirect",
      "GL_AMD_seamless_cubemap_per_texture",
      "GL_AMD_vertex_shader_viewport_index",
      "GL_AMD_vertex_shader_layer",
      "GL_ARB_arrays_of_arrays",
      "GL_ARB_base_instance",
      "GL_ARB_bindless_texture",
      "GL_ARB_blend_func_extended",
      "GL_ARB_buffer_storage",
      "GL_ARB_clear_buffer_object",
      "GL_ARB_clear_texture",
      "GL_ARB_clip_control",
      "GL_ARB_color_buffer_float",
      "GL_ARB_compatibility",
      "GL_ARB_compressed_texture_pixel_storage",
      "GL_ARB_conservative_depth",
      "GL_ARB_compute_shader",
      "GL_ARB_compute_variable_group_size",
      "GL_ARB_conditional_render_inverted",
      "GL_ARB_copy_buffer",
      "GL_ARB_copy_image",
      "GL_ARB_cull_distance",
      "GL_ARB_debug_output",
      "GL_ARB_depth_buffer_float",
      "GL_ARB_depth_clamp",
      "GL_ARB_depth_texture",
      "GL_ARB_derivative_control",
      "GL_ARB_direct_state_access",
      "GL_ARB_draw_buffers",
      "GL_ARB_draw_buffers_blend",
      "GL_ARB_draw_indirect",
      "GL_ARB_draw_elements_base_vertex",
      "GL_ARB_draw_instanced",
      "GL_ARB_enhanced_layouts",
      "GL_ARB_ES2_compatibility",
      "GL_ARB_ES3_compatibility",
      "GL_ARB_ES3_1_compatibility",
      "GL_ARB_ES3_2_compatibility",
      "GL_ARB_explicit_attrib_location",
      "GL_ARB_explicit_uniform_location",
      "GL_ARB_fragment_coord_conventions",
      "GL_ARB_fragment_layer_viewport",
      "GL_ARB_fragment_program",
      "GL_ARB_fragment_program_shadow",
      "GL_ARB_fragment_shader",
      "GL_ARB_fragment_shader_interlock",
      "GL_ARB_framebuffer_no_attachments",
      "GL_ARB_framebuffer_object",
      "GL_ARB_framebuffer_sRGB",
      "GL_ARB_geometry_shader4",
      "GL_ARB_get_program_binary",
      "GL_ARB_get_texture_sub_image",
      "GL_ARB_gl_spirv",
      "GL_ARB_gpu_shader5",
      "GL_ARB_gpu_shader_fp64",
      "GL_ARB_gpu_shader_int64",
      "GL_ARB_half_float_pixel",
      "GL_ARB_half_float_vertex",
      "GL_ARB_imaging",
      "GL_ARB_indirect_parameters",
      "GL_ARB_instanced_arrays",
[... 660 lines cut ...]
09:39:15.237   DEBUG PortAudioSound.cpp:154 Device: 20: dmix
09:39:15.237   DEBUG PortAudioSound.cpp:154 Device: 21: default
09:39:15.237   DEBUG PortAudioSound.cpp:154 Device: 22: Studio 1810c Pro
09:39:15.237   DEBUG PortAudioSound.cpp:154 Device: 23: Google Chrome
09:39:15.237   DEBUG PortAudioSound.cpp:154 Device: 24: libcanberra
09:39:15.353 MESSAGE ocpn_frame.cpp:4683 OnInitTimer...5
09:39:15.453 MESSAGE ocpn_frame.cpp:4683 OnInitTimer...6
09:39:15.554 MESSAGE ocpn_frame.cpp:4683 OnInitTimer...7
09:39:15.554 MESSAGE ocpn_frame.cpp:4958 OnInitTimer...Last Call
09:39:15.554 MESSAGE ocpn_frame.cpp:4987 OnInitTimer...Finalize Canvases
09:39:15.557 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
09:39:15.558 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1790
09:39:15.561 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
09:39:15.561 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1790
09:39:18.022 MESSAGE plugin_loader.cpp:861 PluginLoader: Deactivating PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
09:39:18.022 MESSAGE chartdldr_pi.cpp:214 chartdldr_pi: DeInit
09:39:18.022 MESSAGE plugin_loader.cpp:861 PluginLoader: Deactivating PlugIn: /usr/lib/opencpn/libwmm_pi.so
09:39:18.022 MESSAGE ocpn_frame.cpp:1802 opencpn::MyFrame exiting cleanly.
09:39:18.025 MESSAGE chartdb.cpp:287 Chart cache PlugIn purge
09:39:18.026 MESSAGE ocpn_frame.cpp:1971 gFrame destroyed.
09:39:18.026 MESSAGE ocpn_app.cpp:1995 opencpn::MyApp starting exit.
09:39:18.026 MESSAGE ocpn_app.cpp:2035 LOGBOOK:  2025-09-03 08:39:18 UTC OFF: Lat   33.35800 Lon  -79.28200
09:39:18.026 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /usr/share/opencpn/tcdata/harmonics-dwf-20210110-free.tcd
09:39:18.027 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
09:39:18.028 MESSAGE ocpn_app.cpp:2063 opencpn::MyApp exiting cleanly...